raspberrypi/rpicam-apps #199: libcamera-still stops sending image data after just a few images

knutegit, 2021-12-15:
pi@raspberrypi4B:~ $ uname -a
Linux raspberrypi4B 5.10.83-v7l+ #1499 SMP Tue Dec 7 14:08:09 GMT 2021 armv7l GNU/Linux

pi@raspberrypi4B:~ $ libcamera-still --version
libcamera-apps build: 7617ec5cdba5-intree 14-12-2021 (16:39:38)
libcamera build: v0.0.0+3368-8ff5a8d5
pi@raspberrypi4B:~ $ 

RPi 4 - latest Bullseye with newer kernel

I was having issues with a color/exposure problem so I checked to see if I had the latest libcamera-apps.  I didn't so I attempted to compile them again.  They wouldn't compile so I downloaded the libcamera source and compiled that and libcamea-apps but libcamera-still would seg fault.  After posting on the RPi forum I learned that there is a kernel issue that requires a newer kernel.  I did an rpi-update for the latest kernel.  I recompiled libcamera, libepoxy and libcamera-apps.  I now have a new problem with libcamera-still.  It will take a few (varies from 1 to a dozen or more) and then locks up and no more data comes out.

The command line I am using is:   libcamera-still --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80 --output -

Below is the end of the output to the terminal:

�GJM�O�@�o�,_c1��˃��z�Ԯ4�
                         *͖�-�0��ɴe��P	X�a��?*��4�n��8��ҝ����m)Z�����"�tc)	��=�/�^��[y�Yc8 �
���n�Z�g�g�RK�$�O���b{-L���?���t�>WΣ~U�k��ɦ�g��@]",�� '�UG\���F����f�{b���>��	em�߿@j�Մ�&�H��k68Ғd��!�7C���*�����zm�~TQ|��9�LM��8�J2[�P�08&��۴������E��c��j��Qz�5]*���	-����};�c�T��a��P-H�i$
b���d#�R�lXe#f�P��ԗrۅ2)P�+��-�Ⱦi������n���<�tA@�[H�=�QA` g�)�8�MG���^���L�U;��$��֜b�G�=3�L
�dv���*p�N6�%X\P?��[0:19:51.702734798] [5180]  INFO Camera camera.cpp:1028 configuring streams: (0) 1296x972-YUV420
[0:19:51.703518154] [5184]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:19:51.723473730] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:19:51.723571993] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:19:51.723602839] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:19:51.723631890] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:19:51.723660923] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:19:51.723691640] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:19:51.723721394] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:19:51.723747335] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:19:51.723777182] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:19:51.723803344] [5225]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:19:51.726206014] [5184]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:19:51.902673763] [5180]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420 (1) 2592x1944-SGBRG10_CSI2P
[0:19:51.903424568] [5184]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 2592x1944-SGBRG10_1X10 - Selected unicam format: 2592x1944-pGAA
[0:19:51.992697679] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:19:51.992828954] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:19:51.992892481] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:19:51.992951157] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:19:51.993005408] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:19:51.993081359] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:19:51.993137054] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:19:51.993187934] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[0:19:51.993246740] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:19:51.993298695] [5228]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 2
[0:19:51.997458259] [5184]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

This time when I ran the command I got about 4 images before it hung.  Like I said it varies but usually around that many before it stops.  Never runs for a minute or more.

I have a Java application that I wrote to do motion detection.  I has been running for a couple of years on Buster with raspistill.  Once you guys fixed the --thumb none and --output - issues it ran fine on Bullseye with libcamera-still except for the occasional color/exposure issue.  See the sample Java program below that demonstrates the method I'm using in my program to capture the images from libcamera-still (and before that raspistill).  It will run with Java 11 or later.  It formats a little funny here.

package com.knutejohnson.pi.camera;

import java.awt.*;
import java.awt.event.*;
import java.awt.image.*;
import java.io.*;
import java.util.*;
import javax.imageio.*;
import javax.swing.*;

public class Camera extends JFrame implements Runnable {
    private static final long serialVersionUID = 1L;
    private static final int WIDTH = 640;
    private static final int HEIGHT = 480;
    private final ImagePanel panel;
    private final Thread thread;
    private volatile ProcessHandle processHandle;
    private volatile boolean stopFlag;

    public Camera() {
        super("Camera");
        thread = new Thread(this);
        addWindowListener(new WindowAdapter() {
            public void windowOpened(WindowEvent we) {
                start();
            }
            public void windowClosing(WindowEvent we) {
                dispose();
            }
            public void windowClosed(WindowEvent we) {
                stop();
            }
        });
        panel = new ImagePanel();
        add(panel,BorderLayout.CENTER);
        pack();
        setVisible(true);
    }

    public void start() {
        if (thread.getState() == Thread.State.NEW) {
            thread.start();
        }
    }

    public void run() {
        try {
            ProcessBuilder builder = new ProcessBuilder(
//             "raspistill",
             "libcamera-still",
             "--nopreview",
             "--thumb","none",
             "--timeout","0",
             "--timelapse","1",  // fast as it will go
             "--width",Integer.toString(WIDTH),
             "--height",Integer.toString(HEIGHT),
             "--quality","80",
             "--output","-");

            builder.redirectError(new File("/dev/null"));

            Process process = builder.start();
            processHandle = process.toHandle();

            InputStream is = process.getInputStream();
            ByteArrayOutputStream baos = new ByteArrayOutputStream();
            while (process.isAlive()) {
                int last = 0;
                int nextToLast = 0;
                int n;

                do {
                    if ((n = is.read()) == -1)
                        throw new IOException("end of stdin stream");
                    baos.write(n);
                    nextToLast = last;
                    last = n;
                } while (!(nextToLast == 0xff && last == 0xd9)) ;

                ByteArrayInputStream bais = new ByteArrayInputStream(
                 baos.toByteArray());
                BufferedImage image = ImageIO.read(bais);
                System.out.println("image read");
                if (image != null)
                    panel.setImage(image);
                baos.reset();
            }
        } catch (IOException ioe) {
            if (!stopFlag)
                System.out.println(ioe);
        }

        dispose();
    }

    public void stop() {
        stopFlag = true;
        processHandle.destroyForcibly();
    }

    public static void main(String[] args) {
        EventQueue.invokeLater(() -> new Camera());
    }

    private class ImagePanel extends JPanel {
        private static final long serialVersionUID = 1L;
        private BufferedImage image;

        public ImagePanel() {
            setPreferredSize(new Dimension(Camera.WIDTH,Camera.HEIGHT));
        }

        public void setImage(BufferedImage image) {
            this.image = image;
            repaint();
        }

        public void paintComponent(Graphics g) {
            if (image != null)
                g.drawImage(image,0,0,getWidth(),getHeight(),null);
        }
    }
}

naushir, 2021-12-16:
Running the following command
```
libcamera-still --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80 --output -
```
seems fine for me on a Pi4 with imx219 running the latest libcamera version on Bullseye.

@knutegit, when you see your crash, are you also running your Java app with the piped output?  If so, can you just run the libcamera-still command on its own?

It might also help if you run with the ``camera-bug-report`` tool that comes in the libcamera-apps repo:

```
utils/camera-bug-report -c "libcamera-still --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80"
```

Press ``Ctrl+C`` when the app crashes, and you will get a crash, and post the generated bug-report.txt file over here.  Note that I've removed the ``--output -`` when generating the bug report, otherwise the tool may not capture the log output correctly.

knutegit, 2021-12-16:
The crash is just a stop.  If I'm running it from the command line libcamera-still just stops sending data.  I have to hit CTRL-C to get the prompt back.  When it stops in the Camera.java program it just blocks on the read.  The output in my original post was from the command line.

I don't know what camera I have.  The only marking on it is Raspberry Pi Camera rev 1.3.

Command line output from the camera-bug-report.  It stopped on its own, no CTRL-C was required:

pi@raspberrypi4B:~/libcamera-apps/utils $ ./camera-bug-report -c "libcamera-still --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80"
[0:18:23.635716672] [4993]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3368-8ff5a8d5
[0:18:23.638649941] [4994] ERROR CameraSensor camera_sensor.cpp:535 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[0:18:23.660976849] [4994]  INFO RPI raspberrypi.cpp:1256 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:18:23.661153810] [4993]  INFO Camera camera.cpp:842 Pipeline handler in use by another process
[0:18:23.661443268] [4994] ERROR V4L2 v4l2_videodevice.cpp:1179 /dev/video16[15:cap]: Unable to request 0 buffers: Device or resource busy
[0:18:23.661657098] [4994] ERROR V4L2 v4l2_videodevice.cpp:1179 /dev/video15[14:cap]: Unable to request 0 buffers: Device or resource busy
[0:18:23.661815503] [4994] ERROR V4L2 v4l2_videodevice.cpp:1179 /dev/video14[13:cap]: Unable to request 0 buffers: Device or resource busy
[0:18:23.661968649] [4994] ERROR V4L2 v4l2_videodevice.cpp:1179 /dev/video13[12:out]: Unable to request 0 buffers: Device or resource busy
[0:18:23.662132405] [4994] ERROR V4L2 v4l2_videodevice.cpp:1179 /dev/video0[11:cap]: Unable to request 0 buffers: Device or resource busy
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/ov5647@36 ***

Bug report generated to bug-report.txt
Please upload this file when you create a new bug report at:
https://github.com/raspberrypi/libcamera-apps/issues/
pi@raspberrypi4B:~/libcamera-apps/utils $ 

[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7728660/bug-report.txt)

So I ran the above from the ~/libcamera-apps/utils directory.  Running it from ~/bin I don't get the failed to acquire camera notice.

pi@raspberrypi4B:~/bin $ ../libcamera-apps/utils/camera-bug-report -c "libcamera-still --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80"
[0:27:16.055986991] [6578]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3368-8ff5a8d5
[0:27:16.059042034] [6579] ERROR CameraSensor camera_sensor.cpp:535 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[0:27:16.113054571] [6579]  INFO RPI raspberrypi.cpp:1256 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:27:16.113762883] [6578]  INFO Camera camera.cpp:1028 configuring streams: (0) 1296x972-YUV420
[0:27:16.115862837] [6579]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[0:27:16.183371151] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:27:16.183522447] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:27:16.183549891] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:27:16.183571336] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:27:16.183680280] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:27:16.183722131] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:27:16.183766205] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:27:16.183802520] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:27:16.183862501] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:27:16.183886594] [6584]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:27:16.185635327] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:27:16.185663235] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:27:16.185679049] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:27:16.185693346] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:27:16.185707234] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:27:16.185720734] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:27:16.185734845] [6579]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0

Bug report generated to bug-report.txt
Please upload this file when you create a new bug report at:
https://github.com/raspberrypi/libcamera-apps/issues/

[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7728698/bug-report.txt)

I'm going to download and recompile again today, maybe I just missed an update somewhere.

naushir, 2021-12-16:
It might also be worth adding the ``-v`` command to your ``libcamera-still`` command line to provide more verbose logging.

knutegit, 2021-12-16:
I found this in syslog.  It only happens when I run libcamera-still from my Java program, not from the command line.

Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.581891] ------------[ cut here ]------------
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.581919] WARNING: CPU: 2 PID: 19849 at drivers/media/common/videobuf2/videobuf2-core.c:1956 __vb2_queue_cancel+0x1e0/0x258 [videobuf2_common]
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.581924] Modules linked in: 8021q garp stp llc ov5647 snd_soc_hdmi_codec brcmfmac brcmutil joydev v3d cfg80211 sg gpu_sched rfkill gpio_keys raspberrypi_hwmon i2c_mux_pinctrl i2c_brcmstb i2c_mux bcm2835_unicam v4l2_dv_timings v4l2_fwnode i2c_bcm2835 bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common vc_sm_cma(C) snd_bcm2835(C) videodev mc vc4 rpivid_mem cec drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops dht11 nvmem_rmem industrialio uio_pdrv_genirq uio nf_log_ipv6 ip6t_REJECT nf_reject_ipv6 xt_hl squashfs ip6_tables ip6t_rt nf_log_ipv4 nf_log_common ipt_REJECT nf_reject_ipv4 xt_LOG nf_conntrack_ftp xt_CT nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nft_counter nf_tables nfnetlink i2c_dev fuse drm drm_panel_orientation_quirks
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582335]  backlight ip_tables x_tables ipv6
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582364] CPU: 2 PID: 19849 Comm: libcamera-still Tainted: G        WC        5.10.83-v7l+ #1499
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582368] Hardware name: BCM2711
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582371] Backtrace: 
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582388] [<c0b86a40>] (dump_backtrace) from [<c0b86dd4>] (show_stack+0x20/0x24)
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582394]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6fd8
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582401] [<c0b86db4>] (show_stack) from [<c0b8b1b4>] (dump_stack+0xcc/0xf8)
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582408] [<c0b8b0e8>] (dump_stack) from [<c0220d0c>] (__warn+0xfc/0x114)
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582414]  r10:bf444474 r9:00000009 r8:bf3876b4 r7:000007a4 r6:00000009 r5:bf3876b4
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582417]  r4:bf38cf90 r3:c1205094
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582423] [<c0220c10>] (__warn) from [<c0b87534>] (warn_slowpath_fmt+0x70/0xd8)
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582427]  r7:000007a4 r6:bf38cf90 r5:c1205048 r4:00000000
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582440] [<c0b874c8>] (warn_slowpath_fmt) from [<bf3876b4>] (__vb2_queue_cancel+0x1e0/0x258 [videobuf2_common])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582445]  r9:bf402118 r8:c33c850c r7:00000001 r6:c33c83ac r5:c5853780 r4:c33c83ac
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582463] [<bf3874d4>] (__vb2_queue_cancel [videobuf2_common]) from [<bf3886b4>] (vb2_core_streamoff+0x28/0xbc [videobuf2_common])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582468]  r10:bf444474 r9:bf402118 r8:c33c850c r7:00000001 r6:c1205048 r5:c5853780
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582471]  r4:c33c83ac r3:00000001
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582488] [<bf38868c>] (vb2_core_streamoff [videobuf2_common]) from [<bf3f30f8>] (vb2_streamoff+0x24/0x64 [videobuf2_v4l2])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582492]  r5:c5853780 r4:00000001
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582505] [<bf3f30d4>] (vb2_streamoff [videobuf2_v4l2]) from [<bf3f3184>] (vb2_ioctl_streamoff+0x4c/0x50 [videobuf2_v4l2])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582546] [<bf3f3138>] (vb2_ioctl_streamoff [videobuf2_v4l2]) from [<bf402144>] (v4l_streamoff+0x2c/0x30 [videodev])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582550]  r5:40045613 r4:bf3f3138
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582602] [<bf402118>] (v4l_streamoff [videodev]) from [<bf4071b4>] (__video_do_ioctl+0x234/0x460 [videodev])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582606]  r5:40045613 r4:c33c8520
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582657] [<bf406f80>] (__video_do_ioctl [videodev]) from [<bf407bd8>] (video_usercopy+0x138/0x5f0 [videodev])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582662]  r10:00000000 r9:c78e3e3c r8:00000000 r7:c1205048 r6:00000000 r5:40045613
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582665]  r4:40045613
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582715] [<bf407aa0>] (video_usercopy [videodev]) from [<bf4080b0>] (video_ioctl2+0x20/0x24 [videodev])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582720]  r10:0000000b r9:c78e2000 r8:00000000 r7:c5853780 r6:c5853781 r5:c1205048
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582723]  r4:bf408090
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582772] [<bf408090>] (video_ioctl2 [videodev]) from [<bf4001f4>] (v4l2_ioctl+0x4c/0x60 [videodev])
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582802] [<bf4001a8>] (v4l2_ioctl [videodev]) from [<c04552dc>] (sys_ioctl+0x1d4/0x8ec)
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582806]  r5:c1205048 r4:40045613
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582813] [<c0455108>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
Dec 16 10:31:25 raspberrypi4B kernel: [ 4144.582817] Exception stack(0xc78e3fa8 to 0xc78e3ff0)
[... cut ...]
Dec 16 10:31:33 raspberrypi4B kernel: [ 4152.920084] 3fe0: 00000000 b22367c8 00000000 b6b958e8 80000010 b1919ac8
Dec 16 10:31:33 raspberrypi4B kernel: [ 4152.920089]  r10:0000019e r9:c78e2000 r8:c0200204 r7:0000019e r6:b1919ac8 r5:00000000
Dec 16 10:31:33 raspberrypi4B kernel: [ 4152.920092]  r4:00000008
Dec 16 10:31:33 raspberrypi4B kernel: [ 4152.920097] ---[ end trace 0435eb8b0700132a ]---
Dec 16 10:31:33 raspberrypi4B kernel: [ 4152.920104] videobuf2_common: driver bug: stop_streaming operation is leaving buf 7737e506 in active state

knutegit, 2021-12-16:
With -v added.

pi@raspberrypi4B:~/bin/com/knutejohnson/pi/camera $ camera-bug-report -c "libcamera-still -v --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80"
Options:
    verbose: 1
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 0
    width: 640
    height: 480
    output: 
    post_process_file: 
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
    encoding: jpg
    quality: 80
    raw: 0
    restart: 0
    timelapse: 1
    framestart: 0
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 0
    thumbnail height: 0
    thumbnail quality: 0
    latest: 
    immediate 0
Running without preview window
Opening camera...
[1:12:56.646765237] [20739]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3368-8ff5a8d5
[1:12:56.649968755] [20740] ERROR CameraSensor camera_sensor.cpp:535 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[1:12:56.713048519] [20740]  INFO RPI raspberrypi.cpp:1256 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring viewfinder...
Viewfinder size chosen is 1296x972
[1:12:56.713724054] [20739]  INFO Camera camera.cpp:1028 configuring streams: (0) 1296x972-YUV420
[1:12:56.714538161] [20740]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
Camera streams configured
Buffers allocated and mapped
Viewfinder setup complete
Requests created
[1:12:56.733538390] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[1:12:56.733627294] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[1:12:56.733654656] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[1:12:56.733677924] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[1:12:56.733707454] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[1:12:56.733731296] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[1:12:56.733777944] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[1:12:56.733800101] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[1:12:56.733828797] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[1:12:56.733849193] [20745]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[1:12:56.735574444] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[1:12:56.735602343] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[1:12:56.735618665] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[1:12:56.735634393] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[1:12:56.735649361] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[1:12:56.735664256] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[1:12:56.735679076] [20740]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
Camera started!
Viewfinder frame 0
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed

Bug report generated to bug-report.txt
Please upload this file when you create a new bug report at:
https://github.com/raspberrypi/libcamera-apps/issues/

[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7728954/bug-report.txt)

naushir, 2021-12-16:
Could you turn on libcamera verbose logging as well?  To do this, you need to set an environment variable before running:
```
export LIBCAMERA_LOG_LEVELS=RPI:0
```

knutegit, 2021-12-16:
Please disrgard what I said about the syslog messages only showing up when I run it from Java.  I hadn't actually stopped libcamera-still with the CTRL-C yet.  They are the same.

pi@raspberrypi4B:~/bin/com/knutejohnson/pi/camera $ export LIBCAMERA_LOG_LEVELS=RPI:0
pi@raspberrypi4B:~/bin/com/knutejohnson/pi/camera $ camera-bug-report -c "libcamera-still -v --nopreview --thumb none --timeout 0 --timelapse 1 --width 640 --height 480 --quality 80"
Options:
    verbose: 1
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 0
    width: 640
    height: 480
    output: 
    post_process_file: 
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
    encoding: jpg
    quality: 80
    raw: 0
    restart: 0
    timelapse: 1
    framestart: 0
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
    thumbnail width: 0
    thumbnail height: 0
    thumbnail quality: 0
    latest: 
    immediate 0
Running without preview window
Opening camera...
[1:18:12.587465425] [21544]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3368-8ff5a8d5
[1:18:12.590457915] [21545] ERROR CameraSensor camera_sensor.cpp:535 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[1:18:12.653051928] [21545]  INFO RPI raspberrypi.cpp:1256 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[1:18:12.653100867] [21545] DEBUG RPI raspberrypi.cpp:1060 Unable to acquire a Unicam instance
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
Configuring viewfinder...
[1:18:12.653543761] [21544] DEBUG RPI raspberrypi.cpp:505 Try color space JPEG
Viewfinder size chosen is 1296x972
[1:18:12.653768253] [21544] DEBUG RPI raspberrypi.cpp:505 Try color space JPEG
[1:18:12.653885499] [21544] DEBUG RPI raspberrypi.cpp:505 Try color space JPEG
[1:18:12.653915885] [21544]  INFO Camera camera.cpp:1028 configuring streams: (0) 1296x972-YUV420
[1:18:12.654032834] [21545] DEBUG RPI raspberrypi.cpp:166 Format: 640x480 fmt SGBRG10 Score: 3296 (best 3296)
[1:18:12.654077699] [21545] DEBUG RPI raspberrypi.cpp:166 Format: 1296x972 fmt SGBRG10 Score: 1000 (best 1000)
[1:18:12.654100752] [21545] DEBUG RPI raspberrypi.cpp:166 Format: 1920x1080 fmt SGBRG10 Score: 1349.67 (best 1000)
[1:18:12.654123620] [21545] DEBUG RPI raspberrypi.cpp:166 Format: 2592x1944 fmt SGBRG10 Score: 1567 (best 1000)
[1:18:12.654148913] [21545] DEBUG RPI raspberrypi.cpp:166 Format: 640x480 fmt SGBRG8 Score: 4296 (best 1000)
[1:18:12.654766732] [21545]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected unicam format: 1296x972-pGAA
[... 169 lines cut ...]
[1:18:13.019661144] [21545] DEBUG RPI raspberrypi.cpp:1943 Could not find matching embedded buffer
[1:18:13.019678049] [21545] DEBUG RPI raspberrypi.cpp:1950 Returning bayer frame without a match
[1:18:13.019713675] [21545] DEBUG RPI raspberrypi.cpp:1887 Signalling signalIspPrepare: Bayer buffer id: 0
[1:18:13.020709042] [21545] DEBUG RPI raspberrypi.cpp:1509 Input re-queue to ISP, buffer id 0, timestamp: 4692996794000
[1:18:13.020749075] [21545] DEBUG RPISTREAM rpi_stream.cpp:237 Queuing buffer -1 for ISP Input
[1:18:13.026937384] [21545] DEBUG RPI raspberrypi.cpp:1620 Stream ISP Input buffer complete, buffer id 0, timestamp: 4692996794000
[1:18:13.026984082] [21545] DEBUG RPISTREAM rpi_stream.cpp:237 Queuing buffer 0 for Unicam Image
[1:18:13.027061036] [21545] DEBUG RPI raspberrypi.cpp:1648 Stream ISP Output0 buffer complete, buffer id 0, timestamp: 4692996794000
[1:18:13.027143879] [21545] DEBUG RPI raspberrypi.cpp:1648 Stream ISP Output1 buffer complete, buffer id 0, timestamp: 4692996794000
[1:18:13.027167709] [21545] DEBUG RPISTREAM rpi_stream.cpp:237 Queuing buffer 0 for ISP Output1
[1:18:13.027243608] [21545] DEBUG RPI raspberrypi.cpp:1648 Stream ISP Stats buffer complete, buffer id 0, timestamp: 4692996794000
[1:18:13.028284896] [21545] DEBUG RPISTREAM rpi_stream.cpp:237 Queuing buffer 0 for ISP Stats
Viewfinder frame 0
Closing Libcamera application(frames displayed 0, dropped 0)
[1:18:13.028681258] [21545] DEBUG RPI raspberrypi.cpp:1015 queueRequestDevice: New request.
[1:18:13.028718050] [21545] DEBUG RPISTREAM rpi_stream.cpp:237 Queuing buffer 0 for ISP Output0
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed

Bug report generated to bug-report.txt
Please upload this file when you create a new bug report at:
https://github.com/raspberrypi/libcamera-apps/issues/

[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7728987/bug-report.txt)

naushir, 2021-12-17:
Thank you for the logs!

I'm afraid I don't see anything obvious going wrong from them, but I will keep looking.  If there are any further details you could provide, please do post them here!

davidplowman, 2021-12-17:
I've just gone through a process as follows:

* Plugged an ov5647 into my Pi 4B (I see this is what you're using).
* Installed a fresh Bullseye using Raspberry Pi Imager tool.
* Let it start up and apply all updates. Rebooted.
* Run `sudo rpi-update`. Rebooted.
* Worked through the build instructions for libcamera, libepoxy and libcamera-apps (choosing the "non-Lite" options where I had a choice). For libcamera I first checked out the revision that you listed with `libcamera-still --version`, ditto for libcamera-apps.
* My kernel, libcamera and libcamera-apps all report the same versions as you.
* Finally I ran the exact command you give at the top.

As Naush did, I have found it runs with no problems. I have left it running for minutes at a time.

@knutegit Are you able to perform this exact experiment starting with a completely clean Bullseye image? If that still fails for you then I wonder if we're going to have to start swapping our cameras or cables, but maybe we should run that experiment first and see if it reveals anything. Thanks!

knutegit, 2021-12-17:
Same Pi 4 2GB, same camera, new bullseye, rpi-update, recompiled libcamera, libepoxy and libcamera-apps.

pi@raspberrypi:~/bin $ uname -a
Linux raspberrypi 5.10.87-v7l+ #1502 SMP Fri Dec 17 15:13:41 GMT 2021 armv7l GNU/Linux

pi@raspberrypi:~/bin $ libcamera-still --version
libcamera-apps build: 7617ec5cdba5-intree 17-12-2021 (10:13:31)
libcamera build: v0.0.0+3368-8ff5a8d5

No change.  The first time I ran the libcamera-still I got no image data.  Subsequently I got a few before it hung.  This is very frustrating for both of us I think :-).

davidplowman, 2021-12-17:
It all seems very mysterious. I'm starting to wonder a little bit about trying to swap out things like cameras/cables/SD cards (depending on whether you have spares). Do you have a particularly long ribbon cable or anything?

knutegit, 2021-12-17:
I don't think it is the camera or the cable.  When I switch to the legacy camera stack it runs fine.  The ribbon cable is short about 6 inches.  I used a different SD card for the previous test.

This version is running continuously on my Pi3 and not hanging.  It has other issues but it doesn't hang.
pi@camerapi:~ $ libcamera-still --version
libcamera-apps build: 19190467dbd2-intree 03-12-2021 (20:23:56)
libcamera build: v0.0.0+3156-f4070274

Something changed between that one and the current one I'm using:
libcamera-apps build: 7617ec5cdba5-intree 16-12-2021 (10:15:05)
libcamera build: v0.0.0+3368-8ff5a8d5

Oh and when I did the previous test with a new bullseye that one had a libcamera-apps dated today I think.  I'll update it and see if that makes any difference.

knutegit, 2021-12-17:
Today's libcamera-apps didn't make any difference.

davidplowman, 2021-12-18:
I'm wondering whether the fresh Pi 4 image you made is clean and free of any private information? If so you could maybe upload it and then I could try it?

knutegit, 2021-12-19:
I would be happy to upload it if you can tell me how.  But I think I have figured out where the problem is.  I moved the fresh image SD card to my Pi3 that has a version 2, I think, camera.  It's been running now just over 6800 images.  That made me try to think what were the differences between my Pi3 and the Pi4 I had been using.  I tried removing all of the dtoverlays and the arm_boost=1 that wasn't in the Pi3.  Still didn't run very long on the Pi4, so I changed Pi4s to an older one without the 3 little chips in a triangle.  It is just now approaching 1000 images.  There is something different in the way libcamera works between those two types of Pi4s.  It runs fine on the Pi3 and the Pi4 1.1 and won't run on the Pi4 1.4.  The only difference between them is the 1.1 is 4GB and 1.4 2GB.

davidplowman, 2021-12-19:
I'm using a Pi 4 4GB revision 1.4 and it seems fine. Setting `total_mem=2048` in my config.txt doesn't change the behaviour either. So it all seems to remain very baffling.

knutegit, 2021-12-20:
Well I guess we'll just have to forget this one for now.  I thought I had it figured out there for a minute.  I'm going to post the original problem I had before I couldn't get libcamera to run on my Pi4.

knutegit, 2022-01-03:
I've pretty well convinced myself that this problem was the specific Pi4 that I was using.  It worked for you on your Pi4 1.4 and it works on the new Pi4 1.4 that I got a few days ago.  It still doesn't work on the old one, who knows why.

naushir, 2022-01-04:
This may be down to a power supply issue.  Could you swap the power supply from the known working board to the not working board and try again?

Given this does not seem software related, feel free to close this down if you are satisfied with the resolution.

knutegit, 2022-01-04:
I tried it with 6 different power supplies and that one Pi4 1.4 has a problem.  My other one works just fine.  I think we should close this issue unless/until something else shows up.

closed 2022-01-04 by knutegit

sgr-fr, 2022-02-06:
This issue is now fixed see #234