raspberrypi/rpicam-apps #200: [BUG] OV9281 not working in libcamera-hello, libcamera-still

manhalt, 2021-12-16:
If running libcamera-still or libcamera-hello (from repo/package on image 2021-10-30-raspios-bullseye-armhf), it freezes while dropping frames:
```
pi@raspberrypi:~ $ libcamera-hello -t 0
[1:51:52.090517826] [1458]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[1:51:52.106587981] [1460]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'ov9281'
[1:51:52.106659091] [1460]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[1:51:52.153444282] [1458]  INFO Camera camera.cpp:937 configuring streams: (0) 640x400-YUV420
[1:51:52.153661778] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 1280x800 fmt Y10  Score: 760 (best 760)
[1:51:52.153766017] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt Y10  Score: 806.667 (best 760)
[1:51:52.153832979] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 640x400 fmt Y10  Score: 500 (best 500)
[1:51:52.153903978] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 1280x800 fmt Y10P Score: 760 (best 500)
[1:51:52.153970439] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt Y10P Score: 806.667 (best 500)
[1:51:52.154034105] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 640x400 fmt Y10P Score: 500 (best 500)
[1:51:52.154096993] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 1280x800 fmt GREY Score: 2760 (best 500)
[1:51:52.154158584] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt GREY Score: 2806.67 (best 500)
[1:51:52.154225417] [1460]  INFO RPI raspberrypi.cpp:122 Mode: 640x400 fmt GREY Score: 2500 (best 500)
[1:51:52.154895165] [1460]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov9281@60 - Selected mode: 640x400-Y10P
[1:51:52.172932267] [1460]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[1:51:52.173044451] [1460]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[1:51:52.173106375] [1460]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[1:51:52.246903314] [1460]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[1:51:52.271553305] [1460]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[1:51:52.306745562] [1460]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[1:51:52.735786183] [1460]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
```
This is a problem with the 640x400-Y10P format. When forced to use GREY (recompiling driver) it works. Also libcamera is loosing somewhere memory and gets locked, but I guess this is another problem. See here https://forums.raspberrypi.com/viewtopic.php?t=325820

Sorry, I couln't find the mentioned ``camera-bug-report`` tool.

manhalt, 2021-12-16:
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/7727222/bug.txt) ok I found the camera-bug-report tool by reading the other open issues - maybe someone can add this to README where to find the tool...

naushir, 2021-12-16:
> When forced to use GREY (recompiling driver) it works

Could you please clarify what changes you made to the kernel driver to get libcamera-still to work?  Perhaps post the patch/diff snippet here?

manhalt, 2021-12-16:
```
diff ov9281.c ov9281_cust.c
581a582,584
> //    case 0:
> //            code->code = MEDIA_BUS_FMT_Y10_1X10;
> //            break;
583,585d585
<               code->code = MEDIA_BUS_FMT_Y10_1X10;
<               break;
<       case 1:
909c909
<       try_fmt->code = MEDIA_BUS_FMT_Y10_1X10;
---
>       try_fmt->code = MEDIA_BUS_FMT_Y8_1X8;
```
I just enforced the different encoding. Not sure why this helps. I will try and have a look at the camera controls, since it works for the first frames. I now had the thought that maybe some algorithm in the pipeline (like AE) is messing up the sensor, since it stops after some frames.

manhalt, 2021-12-16:
OK I can confirm it is the vertical_blanking that messes up the sensor. libcamera sets it higher than 3000, but I it seems like every value higher then 403 locks the sensor. If I stream this: 
```
v4l2-ctl --stream-mmap --stream-count=-1 -d /dev/video0 --stream-to=/dev/null --set-fmt-video=width=640,height=400,pixelformat=Y10P
```
Setting this: 
```
v4l2-ctl --set-ctrl=vertical_blanking=404
```
will freeze it. Not sure why in this mode it needs to be limited.
If I find some time in my holidays I will create a PR to update the driver.

6by9, 2021-12-16:
I tested the driver in isolation earlier today, setting VBLANK to 3778 same as libcamera - it worked fine, although gave 26.7fps instead of 30, so there is something a little off in the numbers.
That was with the latest kernel, but ```dtoverlay=ov9281,media_controller=0``` to disable media controller and allow setting the resolution the easy way.

I'll test again tomorrow.

naushir, 2021-12-17:
> OK I can confirm it is the vertical_blanking that messes up the sensor. libcamera sets it higher than 3000, but I it seems like every value higher then 403 locks the sensor.

Good find, thanks for debugging that!  The libcamera pipeline handler will simply use the control limits advertised by the sensor, so this will likely always go wrong.

@6by9, did you run your test in the ``640x400-Y10P`` mode?  This seems to be the one that triggers the lockup.

6by9, 2021-12-17:
> did you run your test in the 640x400-Y10P mode? This seems to be the one that triggers the lockup.

Yes, I had.
Pi4, Innomaker OV9281.
Default vblank gives 260.33fps.
Vblank of 3778 gives 26.29fps.

naushir, 2022-01-05:
@manhalt any update on this issue?

I get the same result as 6by9, OV9281 frames correctly in the 640x400-Y10P mode, but the framerate is a bit wrong.

6by9, 2022-01-05:
Possibly https://github.com/raspberrypi/linux/pull/4807

The min difference between VTS and exposure was insufficient and did lock up the Y10 640x480 mode. Y10 1280x800 mode gave a visible horizontal line, but kept streaming.

manhalt, 2022-01-11:
> Possibly [raspberrypi/linux#4807](https://github.com/raspberrypi/linux/pull/4807)
> 
> The min difference between VTS and exposure was insufficient and did lock up the Y10 640x480 mode. Y10 1280x800 mode gave a visible horizontal line, but kept streaming.

I can confirm this fixes the issue. If you would run libcamera-still before this patch while closing the lens (so the picture is dark) it would freeze but when running with enough light on the sensor not. I just compiled the updated driver and it works, thank you.

closed 2022-01-11 by manhalt