raspberrypi/rpicam-apps #201: Can't build repo

fire-eggs, 2021-12-20:
**Describe the bug**

Did a `git clone` of the repo as of Dec 19, 2021.

Followed build instructions, except:
```
cmake .. -DENABLE_DRM=0 -DENABLE_X11=1 -DENABLE_QT=0 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0
```
make error:
```
[ 42%] Building CXX object core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o
/home/pi/libcam/work/core/libcamera_app.cpp: In member function ‘void LibcameraApp::setupCapture()’:
/home/pi/libcam/work/core/libcamera_app.cpp:666:54: error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’
  666 |     if (i == buffer->planes().size() - 1 || plane.fd.get() != buffer->planes()[i + 1].fd.get())
      |                                                      ^~~
```

**Bug report**

N/A

**Additional context**
Add any other context about the problem here.

Linux raspberrypi 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux

Raspberry Pi O/S Bullseye, 64 bit, fully updated, on a RPi4B, 2G.

fire-eggs, 2021-12-20:
Fails the same with the "recommended" options:
```
cmake .. -DENABLE_DRM=1 -DENABLE_X11=1 -DENABLE_QT=1 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0
```

fire-eggs, 2021-12-20:
```
sudo apt install libcamera-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libcamera-dev is already the newest version (0~git20211020+f4070274-3).
```

davidplowman, 2021-12-20:
Hi, unfortunately the libcamera API is not stable yet so occasionally it moves on and we have to update libcamera-apps to match. So the solution is that you will have to update and build libcamera as well.

There is one more catch in that libcamera has also been updated since the release of the Bullseye kernel. This means that if you want to rebuild libcamera, you will also have to do `sudo rpi-update` to get a newer kernel. Usual caveats apply here: rpi-update is not part of the "normal" update process, so be sure to back up anything critical first.

So to sum it all up: you'll need to do `sudo rpi-update` and then check out and rebuild both libcamera and libcamera-apps.

There will be an updated OS release "shortly" that will resolve the kernel mismatch problems, and then we'll be able to update the apt repositories regularly too, so this situation will improve.

fire-eggs, 2021-12-20:
Maybe Looks to me like if I use commit https://github.com/raspberrypi/libcamera-apps/commit/11c5203a202777ddc339bba1dd8f85b53b60c279  then I might be good?

fire-eggs, 2021-12-20:
> Maybe Looks to me like if I use commit [11c5203](https://github.com/raspberrypi/libcamera-apps/commit/11c5203a202777ddc339bba1dd8f85b53b60c279) then I might be good?

Yup, works.

closed 2021-12-20 by fire-eggs