raspberrypi/rpicam-apps #203: [BUG] fails to make -j4 in ~/libcamera-apps/build

yichenemma, 2021-12-22:
**Describe the bug**

I am trying to post process the video with annotate_cv. I have installed the OpenCV on the raspberry pi and when I `import cv2` to in python it works. Then I followed the documentation `Building libcamera-apps without rebuilding libcamera`. 

**Bug report**

But when I am tring to build libcamera-apps,  `make -j4` command fails to execute due to

> /home/pi/libcamera-apps/core/libcamera_app.cpp: In member function ‘void LibcameraApp::setupCapture()’:
> /home/pi/libcamera-apps/core/libcamera_app.cpp:666:54: error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’
> 666 |     if (i == buffer->planes().size() - 1 || plane.fd.get()!= buffer->planes()[i + 1].fd.get())
>     &nbsp;   &nbsp; &nbsp;&nbsp; | &nbsp;&nbsp;&nbsp;&nbsp;  &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;     &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; &nbsp;&nbsp;&nbsp; ^~~
> compilation terminated due to -Wfatal-errors.
> make[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:82: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:272: core/CMakeFiles/libcamera_app.dir/all] Error 2
> make: *** [Makefile:171: all] Error 2

**Additional context**

* Operating system version: `Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux`
* libcamera-apps build: `2a38ae93f143 21-10-2021 (14:49:21)`
* libcamera build: `v0.0.0+3156-f4070274`
* raspberry pi module: 4 Model B. 

|  | total | used | free | shared | buff/cache | available|
| :----------:  | :----------: | :----------: | :----------: | :----------: | :----------: | :----------: |
|Mem:| 3838|722|738|315|2377|2643|
|Swap:|99|0|99|

Thank you so much for the help in advanced

davidplowman, 2021-12-22:
Hi, unfortunately libcamera doesn't have a stable API yet so sometimes it moves on and we update libcamera-apps to match. This means that, at the moment, if you want to rebuild libcamera-apps then you must first rebuild libcamera. There is one more catch: the latest libcamera also expects a newer kernel than is in the most recent Bullseye image so before rebuilding libcamera you need to `sudo rpi-update`. (Note that rpi-update is not part of the normal update process, so be sure to back up any critical data first).

I'm sorry this is all a bit awkward at the moment. We are working to release updated Bullseye images shortly, at which point this situation will improve significantly.

yichenemma, 2021-12-22:
Thank you for the fast response! And I guess I have to rebuild the Libcamera-apps and Libcamera in order to use annoote_cv? Or is there another walk around?

davidplowman, 2021-12-22:
Yes, I'm afraid that at the moment you will have to do both an rpi-update (back up any important data first) and rebuild libcamera before rebuilding libcamera-apps. But this situation should improve shortly.

fire-eggs, 2021-12-22:
> Thank you for the fast response! And I guess I have to rebuild the Libcamera-apps, Libcamera and write my own Post-processing stage in order to use annoote_cv? Or is there another walk around?

Or use the workaround I found in issue #201: change the few instances of `fd.get()` to `fd.fd()` [about 5 instances across two files]. The code then builds / runs for me.

davidplowman, 2021-12-22:
Glad you've found a way forward!

davidplowman, 2021-12-24:
I think this is solved, but please re-open it if anything else comes up!

closed 2021-12-24 by davidplowman

yichenemma, 2021-12-24:
Thank you so much for the help! And changing to .fd() works perfectly for me. So after this, I tried to ran the command `libcamera-hello -t 0` and I got this error:

> [0:04:51.750303040] [1592]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
> ERROR: *** no cameras available ***
> Segmentation fault

Previously, this command will show up the window for the video streaming.  I also tried to enable the `Legacy Camera` option in `sudo raspi-config` but I received the same error. I have also checked out #168 but unfortunately no luck.

Thank you again in advanced for the help!

davidplowman, 2021-12-24:
I think these crashes are sometimes because the executables aren't quite linked the way you want. Firstly, make sure you rebuild libcamera-apps _after_ libcamera. You may want to run `hash -r` to make sure you pick up the local executable. Also run `sudo ldconfig /usr/local/lib` to be sure the runtime linker will pick up the right libraries. Does this help?

yichenemma, 2021-12-24:
Thank you for the fast response. I just tried these and received a new error now: `ERROR: the system appears to be configured for the legacy camera stack
`

davidplowman, 2021-12-24:
You need to check your /boot/config.txt. On Buster you need `dtoverlay=<your-camera>` where `<your-camera>` is either `ov5647` (v1 camera), `imx219` (v2 camera) or `imx477` (HQ cam). On Bullseye you can use the same or alternatively `camera_auto_detect=1`.