raspberrypi/rpicam-apps #204: UDP Stream + video record

smababou, 2021-12-23:
I would like to know, how to use libcamera-apps, to stream video (udp/tcp) with specific resolution and record a video with different resolution and store it on my Raspi

davidplowman, 2021-12-23:
Hi, so are you wanting to encode two different resolution streams (as h.264, for example) at the same time? Currently libcamera-apps don't support that though it wouldn't be a huge job to add that kind of feature. If you were OK with the same resolution for both streams, and just wanted to output them twice, once over the network and once to a file, that would probably be an easier change. You could possibly even create a basic version of that by outputting to stdout (`-o -`) and using `tee`. Does that help at all?

smababou, 2021-12-24:
It would be helpful if there is an example of making the same resolution for both streams, and just wanted to output them twice, once over the network and once to a file. In order to understand your explanation

davidplowman, 2021-12-24:
So if the destination computer for the stream has IP address 1.2.3.4 you start something like
```
ffplay udp://1.2.3.4:12345
```
on it, and then on the Pi run
```
libcamera-vid --inline -o - | tee save.h264 | nc -u 1.2.3.4 123454
```
and this would both send to stream to the destination computer using udp, and save the file locally as `save.h264`.

If you wanted to make libcamera-vid do this all by itself then you'd have to get involved in C++ programming. You might want to create an `Output` class that "demultiplexes" the stream it's being given to several other `Output` objects. It could be an interesting feature so we'd certainly consider a pull request for something like that.

naushir, 2022-01-05:
I think this can be closed now.  Feel free to re-open if there are any more queries.

closed 2022-01-05 by naushir