raspberrypi/rpicam-apps #205: Full Range vs. Limited / Studio Range color output

dustinkerstein, 2021-12-23:
_This may be more of a libcamera discussion, but I imagine it could overlap libcamera-apps as well. Let me know if I should post over there instead._ 

It appears as though libcamera-vid outputs in Limited / Studio range (16-235) regardless of the codec selected. While this is probably the desired / most compatible approach for h264 encoding, when using mjpeg the choice is a little blurrier, as is when outputting yuv420. 

Currently, since mjpeg is using 16-235, when played back in VLC (and others) the black levels are incorrect. This is the only real "bug" I have seen related to levels, but ideally, I would love to see the ability for the user to configure Full / Limited range for both still and video capture (across all codecs). As long as the metadata is properly set (except for YUV which requires manually telling the encoder about the video data), most modern video players have no issue with either Full or Limited range colors. For my particular use-case (capturing both stills and video in YUV), I'd prefer to use 0-255 to maximize the amount of Dynamic Range captured.

What do you all think?

I am testing with an Arducam Mini IMX477 module + Compute Module 4 + Official IO Board + Bullseye Lite 64bit + 5.10.63-v8+ Kernel running headless over ssh. Full system info here - https://pastebin.com/654vBepL

```
$ libcamera-hello --version
libcamera-apps build: 4872b3dfba4d-intree 22-12-2021 (19:53:20)
libcamera build: v0.0.0+3369-8ff5a8d5
$ vcgencmd version
Nov 18 2021 16:16:49
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
```

davidplowman, 2021-12-23:
Colour space handling is currently just being sorted out. In the "official released" versions in our apt repositories I think everything comes out as being in the JPEG colour space, i.e. fullrange, because libcamera (and therefore libcamera-apps) simply knows nothing about colour spaces.

Very recently (just a couple of weeks ago in fact) basic colour space support has gone into libcamera after many months under discussion and review. The upshot is that certain default choices are made and the libcamera-apps, not yet dealing with colour spaces either, just inherit those. These defaults cause still images to use the JPEG colour space (full range) and video gets Rec.709 (so limited range).

There's a pull request (https://github.com/raspberrypi/libcamera-apps/pull/198) which adds the necessary code to libcamera-apps though I don't think it watches for the MJPEG codec - that needs fixing. Realistically it's going to be the New Year before we get round to merging that. Does it sound like that will take care of things for you?

dustinkerstein, 2021-12-23:
Ok cool. That all sounds good. As long as I can force Full Range for YUV video (which looks like I can via this [line](https://github.com/raspberrypi/libcamera-apps/pull/198/files#diff-e23d0d57a87dbb19d3880167484fb54358abbdcb1096ec0c13f70a66951a2890R70)) then I think my use-case will be covered. Though it might be nice to have this control at the command line level at some point. 

Also, small side note - but you may want to consider renaming the flags like `FLAG_VIDEO_JPEG_COLOURSPACE` as Rec 709 Full Range is valid / used in some production workflows. JPEG will always be in Full Range, but for video there are valid cases to use both Limited and Full, so the Full Range naming being based on "JPEG" might confuse some folks. 

Feel free to close this issue now or when that PR is merged. Thanks for the background info.

davidplowman, 2021-12-29:
Thanks for the suggestions. I'm not quite sure where it will all go at the moment, on the one hand these apps were never meant "to do everything for everyone", but feature-creep is hard to resist...!

closed 2021-12-29 by davidplowman

dustinkerstein, 2021-12-29:
Totally understand :) Thanks for pointing out that PR. It should cover exactly what I need.