raspberrypi/rpicam-apps #206: Initial latency difference between still and video capture

dustinkerstein, 2021-12-23:
_Apologies as this isn't really a bug / issue, but just something that I'm trying to understand in regards to the code design_

I'm trying to use `libcamera-vid` as a basis for a custom libcamera app that captures both stills and video with the same code. It's currently functional, but it seems to take 2x the shutter speed to actually capture a frame compared to libcamera-still. I'm sure there's a simple reason / config that I missed when browsing through the `libcamera-still.cpp` and `libcamera-app.cpp` code, but if anyone has any pointers on how to avoid that extra frame delay, I'd appreciate it. Here are the two commands I'm comparing:

```
time libcamera-still -q 100 --denoise cdn_hq --shutter 10000000 --gain 1 --awbgains 1,1 --immediate --nopreview -o ~/10sec.jpg
time libcamera-vid --frames 1 -q 100 --framerate 0.1 --denoise cdn_hq --shutter 10000000 --gain 1 --awbgains 1,1 --codec mjpeg --nopreview -o ~/10sec.mjpeg
```

The former takes ~11 seconds, and the latter takes ~21 seconds. I thought it could have something to do with the framerate (based on this [code](https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L404)), so I tried slightly above / below 0.1 but got the same results. Thanks! 

PS - Are you all thinking of creating a similar IRC channel like libcamera? I'd much rather ask a question like this in that kind of setting to avoid annoying folks with pseudo/non-bugs.

fire-eggs, 2021-12-24:
I'll be interested in the answer!

davidplowman, 2021-12-24:
I think the answer is simply that when it gets the first frame (taking 10 seconds), libcamera-still immediately stops the camera, saves the image and quits.

libcamera-vid, in contrast, doesn't stop the camera immediately and goes round event loop again which waits for another frame (the second frame, taking another 10s). Moving the `frameout` test above the `app.Wait()` would resolve this, I think.

dustinkerstein, 2021-12-24:
> libcamera-vid, in contrast, doesn't stop the camera immediately and goes round event loop again which waits for another frame (the second frame, taking another 10s). Moving the `frameout` test above the `app.Wait()` would resolve this, I think.

And that was exactly it, thanks! Should have easily caught that myself, but thanks for the help. Hope the info will be useful for anyone else trying to build a consolidated still/video app.

closed 2021-12-24 by dustinkerstein