raspberrypi/rpicam-apps #207: feature request: test pattern

CarlFK, 2021-12-25:
please add an option to generate an image.

When testing apps, like libcamera-hello, it would be nice if I could swap out the camera data for generated data.

for instance I am trying to get an image on my screen, and I can't be sure if the problem is not getting data from the camera hardware, or the app can't display it on the screen.  (don't use this issue to debug my camera display problem, this issue is about adding a feature.)

I have added such a feature to other video stacks and find it very helpful.  Like streaming data over a lan has so many bits that can go wrong, it is very handy to swap the video feed for color bars - when I see the color bars on the remote viewer, I know all of the streaming is working.

One implementation of this concept: 
https://www.kernel.org/doc/html/v4.9/media/v4l-drivers/vivid.html

Personally that is way more complex than I want for the sort of testing I want to do. A single color at a single res/fps would be fine. 

Something other than a black frame so that I can tell it apart from a black background of no video data.  
Raspberry pi color would be nice:

    Name: Bright Maroon
    Hex: #c51a4a
    RGB: (197, 26, 74)
    CMYK: 0, 0.868, 0.624, 0.227
https://www.schemecolor.com/raspberry-pi-logo-colors.php

davidplowman, 2021-12-29:
Hi, thanks for the suggestion. I'm not quite sure exactly what you're wanting to be able to test, perhaps you could explain a bit more.

* Are you wanting to test the image all the way from the camera to the display? Cameras usually have test pattern modes that could be used, though there would be some plumbing involved, and possibly fiddling with drivers if they don't exposure these features.
* Or are you more interested in the path to the display (ignoring the camera). In this case overwriting the camera image with (for example) colour bars may be a simpler approach. It might lend itself to a post-processing stage.
* Or is it that you want to be able to guarantee something is coming out of libcamera so that whatever downstream processing you're doing is guaranteed to have a recognisable input? I think this is probably the same as the previous case.

Thanks!

CarlFK, 2021-12-29:
> Cameras usually have test pattern modes  

I did not know that.  and now I want that too!  for similar reasons, but in addition to what I was asking for:

yes, #2/3 are the same feature used for 2 uses.   add a 2 more: unit tests for my system and unit tests for your system.  I might even contribute some tests.

Here are some examples of where I have used test patterns:
Note: I don't think any of this would be replaced by this libcamera test feature,  but this should give you a feel for how test patterns are used in the wild.

ingest.py connects a video source to the video mixer process:
https://github.com/CarlFK/voctomix-outcasts/blob/master/ingest.py#L79-L114
hdmi/sdi camera card made by BlackMagic company:
```
 elif args.video_source == 'blackmagic':
        video_src = decklinkvideosrc
```
test patterned generated by gstreamer:
```
elif args.video_source == 'test':
    video_src =  videotestsrc
```
"blackmagic" is used for production, "test" is used as part of setup to confirm everything downstream is functioning.  And when something stops functioning and it isn't obvious what broke. Kill the camera feed and bring up the test feed.  If the test feed works, look at the camera, see what cable fell out.  If the test doesn't show up on the other end, look for what process died.

I also have a tests for post processing too:
https://github.com/CarlFK/veyepar/blob/master/dj/scripts/run_tests.py#L222
Generate a few seconds of video, add a title slide to the start, encode it, upload it to youtube/vimeo/peertube/cdn , email and tweet the URLs, check result to see if it sounds and looks like it should (using ocr and speech recognition which is currently broken because no one is maintaining the package :(

Summary: video processing has tons fiddly code that falls over all the time, and debugging is much easier if the camera hardware can be removed from the system.

davidplowman, 2021-12-30:
The camera test pattern modes would be useful to us, though I'm not sure if we're likely to do the necessary work in the very near future.

Having a "test pattern output mode", where we basically overwrite the camera image with something of our own is probably an easier change, as you could do it simply with a "post-processing stage". But given all the areas where we're still playing catch-up I again think it's unlikely we'll do anything like this ourselves in the near future. I was wondering if you had some expertise with C++? Post-processing stages are not difficult to write - see for example the ["negate stage"](https://github.com/raspberrypi/libcamera-apps/blob/main/post_processing_stages/negate_stage.cpp). We'd be happy to help with the process and accept a pull request if that was of any interest to you.

CarlFK, 2021-12-30:
> overwrite the camera image

If the camera is not working (like not plugged in), will this still create output?

davidplowman, 2021-12-30:
No, I'm afraid the whole thing is really driven by frames coming out of the camera, and that's pretty fundamental to our pipeline handler. You might want to have a look at some of libcamera's other pipeline handlers, I think it has some "test" ones that might behave more like this, but I'm not too sure.

CarlFK, 2021-12-30:
bummer.   that completely removes the value I am looking for.  Thanks for explaining. 

All of this background is kinda relevant to #215 so now it is kinda referenced ;)

naushir, 2022-01-04:
The _VIVID_ V4L2 driver and associated libcamera pipeline handler might be exactly what you want here.  See [here](https://www.kernel.org/doc/html/v4.9/media/v4l-drivers/vivid.html) for more details.  This has a whole host of test patterns that could be selected for use. This should work mostly transparently with libcamera-apps.

Disclaimer: I have not tried to run this on a Raspberry Pi or any other platform myself.

6by9, 2022-01-04:
The vivid pipeline handler should work against vivid on the Pi.

I wonder if we can substitute vivid for unicam with the RPI pipeline handler to avoid changing the pipeline handler. In theory that should just produce test frames at the defined rates, but I haven't followed it through as to how well it will generate Bayer data. AE/AWB will do nothing useful as there aren't the relevant controls.

kbingham, 2022-01-04:
Note that the VIVID pipeline handler is actually kept separate from libcamera, however you might find the VIMC pipeline handler more appropriate, and directly supported by libcamera.

If the VIMC device is probed (sudo modprobe vimc), and the vimc pipeline is enabled in the libcamera build, then there will be a 'virtual' camera device available called "Sensor B"

This will show up with cam --list as :
1: 'Sensor B' (platform/vimc.0 Sensor B)

kbingham, 2022-01-04:
(For completeness, the vivid pipeline handler is available at https://git.libcamera.org/libcamera/vivid.git/log/ however it is kept separate as it forms part of the 'pipeline writers guide', and we don't consider Vivid to really be a camera, while the VIMC is designed to be a virtual camera device).

kbingham, 2022-01-04:
(And confirming it works on RPi...)

pi@earth:~ $ sudo modprobe vimc
pi@earth:~/libcamera-gadget $ ./build/gcc/src/cam/cam --list
Available cameras:
1: 'Sensor B' (platform/vimc.0 Sensor B)

./build/gcc/src/cam/cam -c1 -C
[110:10:50.395364018] [12724]  INFO IPAManager ipa_manager.cpp:138 libcamera is not installed. Adding '/home/pi/libcamera-gadget/build/gcc/src/ipa' to the IPA search path
[110:10:50.403081839] [12724]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3374-eca6c40d
[110:10:50.604162910] [12725]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'Sensor B'
[110:10:50.604226816] [12725]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[110:10:50.604262574] [12725]  WARN CameraSensor camera_sensor.cpp:396 'Sensor B': Failed to retrieve the camera location
[110:10:50.613215352] [12725]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/pi/libcamera-gadget/src/ipa/vimc/data'
Using camera platform/vimc.0 Sensor B as cam0
[110:10:50.614966482] [12724]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-BGR888
cam0: Capture until user interrupts by SIGINT
396650.701947 (0.00 fps) cam0-stream0 seq: 000000 bytesused: 6220800
396650.776704 (13.38 fps) cam0-stream0 seq: 000001 bytesused: 6220800
396650.848693 (13.89 fps) cam0-stream0 seq: 000002 bytesused: 6220800
396650.920689 (13.89 fps) cam0-stream0 seq: 000003 bytesused: 6220800
396650.992682 (13.89 fps) cam0-stream0 seq: 000004 bytesused: 6220800
396651.068709 (13.15 fps) cam0-stream0 seq: 000005 bytesused: 6220800
396651.140710 (13.89 fps) cam0-stream0 seq: 000006 bytesused: 6220800
396651.212701 (13.89 fps) cam0-stream0 seq: 000007 bytesused: 6220800
396651.284704 (13.89 fps) cam0-stream0 seq: 000008 bytesused: 6220800
396651.360698 (13.16 fps) cam0-stream0 seq: 000009 bytesused: 6220800
396651.432713 (13.89 fps) cam0-stream0 seq: 000010 bytesused: 6220800
396651.504709 (13.89 fps) cam0-stream0 seq: 000011 bytesused: 6220800
396651.576706 (13.89 fps) cam0-stream0 seq: 000012 bytesused: 6220800
396651.652702 (13.16 fps) cam0-stream0 seq: 000013 bytesused: 6220800
396651.724700 (13.89 fps) cam0-stream0 seq: 000014 bytesused: 6220800
396651.796707 (13.89 fps) cam0-stream0 seq: 000015 bytesused: 6220800
396651.868695 (13.89 fps) cam0-stream0 seq: 000016 bytesused: 6220800
396651.944701 (13.16 fps) cam0-stream0 seq: 000017 bytesused: 6220800

naushir, 2022-01-13:
Seems like the vimc module is the way to go. If there are no objection, I will close this shortly.

closed 2022-01-13 by naushir