raspberrypi/rpicam-apps #208: [BUG] ERROR: *** no cameras available ***

CarlFK, 2021-12-25:
**Describe the bug**
I used the camera, did an "apt update/upgrade" reboot and now "no cameras available"

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.
-bash: camera-bug-report: command not found

**Additional context**

ERROR: *** no cameras available ***
pi@localhost:~ $ DISPLAY=:0 libcamera-hello -t 0
libEGL warning: DRI2: failed to authenticate
[0:30:58.664183978] [3733]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:30:58.824130404] [3733]  INFO Camera camera.cpp:937 configuring streams: (0) 1296x972-YUV420
[0:30:58.824348329] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 2067 (best 2067)
[0:30:58.824445217] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1849.67 (best 1849.67)
[0:30:58.824488809] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 1500 (best 1500)
[0:30:58.824529605] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 3796 (best 1500)
[0:30:58.824571679] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 4796 (best 1500)
[0:30:58.824609494] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 1567 (best 1500)
[0:30:58.824646383] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1349.67 (best 1349.67)
[0:30:58.824684827] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 1000 (best 1000)
[0:30:58.824721327] [3747]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 3296 (best 1000)
[0:30:58.824853733] [3747]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1296x972-pBAA
[0:30:58.851814071] [3747]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:30:58.851868534] [3747]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:30:58.851895348] [3747]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:30:58.851918774] [3747]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:30:58.933897819] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[0:30:58.970669142] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:30:59.004859626] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:30:59.039053037] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:30:59.072555432] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:30:59.104856094] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:30:59.138246842] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:30:59.171742830] [3747]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 21
Aborted
pi@localhost:~ $ sudo reboot 
Connection to 192.168.1.145 closed by remote host.
Connection to 192.168.1.145 closed.
carl@twist:~/sbc/rpi$ ssh pi@192.168.1.145
Linux localhost 5.10.63-v7l+ #1488 SMP Thu Nov 18 16:15:28 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Dec 25 11:24:11 2021

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@localhost:~ $ DISPLAY=:0 libcamera-hello -t 0
libEGL warning: DRI2: failed to authenticate
[0:01:18.941940750] [1231]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
ERROR: *** no cameras available ***

davidplowman, 2021-12-29:
Hi, I'm a bit puzzled that you get a "no cameras available", but the next run clearly shows it finding your ov5647. Please could you report:

* The output of `uname -a`
* The output of `vcgencmd version`
* The output of `libcamera-hello --version`
* The contents of your /boot/config.txt.

The "failed to import fd" error is related to running over ssh. You need to do one of:

* Use the `-n` option to suppress the preview window altogether, or
* Use the `--qt-preview` option to use an implementation of the preview that works with X forwarding (the default implementations use hardware acceleration which does not work across the network).
* If you have a display attached to the Pi you can force X to display the preview there, though you would need to give your ssh session the necessary X authorisation to open a window there (beyond my knowledge, I'm afraid)
* Again if you have a display attached to the Pi you could not run X Windows there at all, and you should get a DRM/KMS preview window.

janmg, 2022-01-08:
I had the same "no cameras available" issue. After working out the exact stack from boot of the kernel,overlays,modules and cleaning up the old libcamera from apt and building everything clean. Then I did an rpi-update and still had the same issue. Now I found that rpi-update changed repository and that in the new repo there is a commit from Jan 6 with "firmware: improve firmware camera detection" as commit message.

I assume doing the first rpi-update updated the firmware to the latest from Hexxeh, but the second time I got the firmware that got actually fixed. So your apps depend on the libcamera libraries that depend on the lower levels of RPI.

https://github.com/raspberrypi/rpi-firmware

For me this now works
libcamera-hello -n

CarlFK, 2022-01-10:
I took a shot at reproducing what I saw - image sd, libcamera-hello, apt update/upgrade, reboot, libcamera-hello .. worked fine.

ohhai, 2022-01-12:
I have same "no cameras available" error with HQ camera (IMX477) and Raspberry Zero W 1.1
OS: current "Raspberry Pi OS with desktop" build from here: https://www.raspberrypi.com/software/operating-systems/

raspivid works fine but libcamera-hello tells this:

```
libEGL warning: DRI2: failed to authenticate
[0:11:24.968207824] [1273]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
ERROR: *** no cameras available ***
```
Also:
```
$ vcgencmd get_camera
supported=1 detected=1
```
What package shall be installed to provide camera-bug-report?
This shows me nothing:
```
# apt-file search  camera-bug-report
# apt-cache search  camera-bug-report
```

naushir, 2022-01-13:
camera-bug-report is a tool that is available in the libcamera-apps repo (``.//utils/camera-bug-report``).  In the next official OS release, it will be installed in the system directories, so will be available to use without copying from this repo.

Can you post the output if your ``/boot/config.txt`` file please? Additionally, is this an official Raspberry Pi HQ camera?

naushir, 2022-01-13:
camera-bug-report is a tool that is available in the libcamera-apps repo (``./utils/camera-bug-report``).  In the next official OS release, it will be installed in the system directories, so will be available to use without copying from this repo.

Can you post the output if your ``/boot/config.txt`` file please? Additionally, is this an official Raspberry Pi HQ camera?

ohhai, 2022-01-13:
@naushir
I have to add that my OS is up to date and not identical to the original `2021-10-30-raspios-bullseye-armhf-full.zip`

1)
Camera is this exact item. Should be same as original with mechanical part added, but not sure for 100%
https://www.arducam.com/product/arducam-high-quality-ir-cut-camera-for-raspberry-pi-12-3mp-1-2-3-inch-imx477-hq-camera-module-with-6mm-cs-lens-for-pi-4b-3b-2b-3a-pi-zero-and-more/
Anyway, `raspivid` and `vcgencmd get_camera` detect it successfully.

2)
This is diff of my config with original file from same image. My complete config is attached: [config-my.txt](https://github.com/raspberrypi/libcamera-apps/files/7862886/config-my.txt)
I can try to replace it completely with original version or restore changed options one by one and check what will happen.

```
$ diff -U 1 /tmp/config-orig.txt /tmp/config-my.txt
--- /tmp/config-orig.txt      2022-01-13 14:34:56.252787876 +0200
+++ /tmp/config-my.txt 2022-01-13 14:34:14.624850125 +0200
@@ -41,5 +41,5 @@
 # Uncomment some or all of these to enable the optional hardware interfaces
-#dtparam=i2c_arm=on
+dtparam=i2c_arm=on
 #dtparam=i2s=on
-#dtparam=spi=on
+dtparam=spi=on
 
@@ -55,3 +55,3 @@
 # Automatically load overlays for detected cameras
-camera_auto_detect=1
+start_x=1
 
@@ -61,3 +61,3 @@
 # Enable DRM VC4 V3D driver
-dtoverlay=vc4-kms-v3d
+#dtoverlay=vc4-kms-v3d
 max_framebuffers=2
@@ -76,2 +76,3 @@
 [pi4]
+dtoverlay=vc4-fkms-v3d
 # Run as fast as firmware / board allows
@@ -80 +81,4 @@
 [all]
+enable_uart=1
+dtoverlay=w1-gpio
+gpu_mem=128
```

3)
I downloaded `camera-bug-report` script (only this file), but it doesn't launch.
Does it need to be run from full repository cloned locally?
```
$ ./camera-bug-report
Traceback (most recent call last):
  File "/my/./camera-bug-report", line 87, in <module>
    if not any([s in app for s in ['libcamera-still', 'libcamera-vid', 'libcamera-hello', 'libcamera-raw', 'libcamera-jpeg']]):
  File "/my/./camera-bug-report", line 87, in <listcomp>
    if not any([s in app for s in ['libcamera-still', 'libcamera-vid', 'libcamera-hello', 'libcamera-raw', 'libcamera-jpeg']]):
TypeError: argument of type 'NoneType' is not iterable

$ ls -l /usr/bin/libcamera-hello
-rwxr-xr-x 1 root root 46972 Oct 21 17:33 /usr/bin/libcamera-hello

$ dpkg -l | grep camera
ii  libcamera-apps                        0~git20211021+2a38ae9-1           armhf        libcamera-apps
ii  libcamera-tools                       0~git20211020+f4070274-3          armhf        complex camera support library (tools)
ii  libcamera0:armhf                      0~git20211020+f4070274-3          armhf        complex camera support library
ii  libdc1394-25:armhf                    2.2.6-3                           armhf        high level programming interface for IEEE 1394 digital cameras
ii  libgphoto2-6:armhf                    2.5.27-1                          armhf        gphoto2 digital camera library
ii  libgphoto2-port12:armhf               2.5.27-1                          armhf        gphoto2 digital camera port library
ii  python3-picamera                      1.13                              armhf        Pure Python interface to the Raspberry Pi's camera module.
```

naushir, 2022-01-13:
You config file is missing either ``camera_auto_detect=1`` or ``dtoverlay=imx477``, so the camera would not be detected in linux.  You probably also want to get rid of the ``gpu_mem=128`` and ``dtparam=i2c_arm=on`` lines.

I suggest you revert back to the original ``config.txt`` file and it ought to work.

ohhai, 2022-01-13:
@naushir

Interesting. I did not edit config.txt manually - did it just by raspi-config with disabling no any option, only enabling some.

Tried 5 steps:
1) Enable only `camera_auto_detect=1`
2) Enable only `dtoverlay=imx477`
3) Enable both
4) Additionally revert (enable) `dtoverlay=vc4-kms-v3d`
5) Revert all, with this enabled:
```
camera_auto_detect=1
dtoverlay=imx477
```

Final diff of config.txt:
```
 camera_auto_detect=1
+dtoverlay=imx477
+#start_x=1
+#dtoverlay=vc4-fkms-v3d
+#enable_uart=1
+#gpu_mem=128
```

**Results**

`libcamera-hello` has similar output for cases 1,2,3:

```
$ libcamera-hello
libEGL warning: DRI2: failed to authenticate
[0:02:43.358980000] [905]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:02:43.602462000] [917]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:02:43.602856000] [917]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:02:44.042488000] [905]  INFO Camera camera.cpp:937 configuring streams: (0) 2028x1520-YUV420
[0:02:44.049630000] [917]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt RG10 Score: 3956.22 (best 3956.22)
[0:02:44.052477000] [917]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt pRCC Score: 1083.84 (best 0)
[0:02:44.053264000] [917]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC
[0:02:44.623023000] [917] ERROR V4L2 v4l2_videodevice.cpp:1158 /dev/video15[17:cap]: Unable to request 4 buffers: Cannot allocate memory
[0:02:44.623987000] [917] ERROR RPI raspberrypi.cpp:817 Failed to allocate buffers

# or this, `v4l2_videodevice.cpp` line is different

$ libcamera-hello
libEGL warning: DRI2: failed to authenticate
[0:04:51.899199000] [916]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:04:52.116144000] [927]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:04:52.116528000] [927]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:04:52.585964000] [916]  INFO Camera camera.cpp:937 configuring streams: (0) 2028x1520-YUV420
[0:04:52.591528000] [927]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt RG10 Score: 3956.22 (best 3956.22)
[0:04:52.594279000] [927]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt pRCC Score: 1083.84 (best 0)
[0:04:52.595078000] [927]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pRCC
[0:04:53.343062000] [927] ERROR V4L2 v4l2_videodevice.cpp:1165 /dev/video15[17:cap]: Not enough buffers provided by V4L2VideoDevice
[0:04:53.388562000] [927] ERROR RPI raspberrypi.cpp:817 Failed to allocate buffers
```

`libcamera-hello` output for cases 4 and 5:
```
libEGL warning: DRI2: failed to authenticate
[0:04:53.114457000] [968]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[0:04:53.457409000] [979]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:04:53.490152000] [979]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:04:54.096295000] [968]  INFO Camera camera.cpp:937 configuring streams: (0) 2028x1520-YUV420
[0:04:54.100632000] [979]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt RG10 Score: 3956.22 (best 3956.22)
[0:04:54.101164000] [979]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt RG12 Score: 1387 (best 1387)
[0:04:54.101530000] [979]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1520 fmt RG12 Score: 500 (best 500)
[0:04:54.101879000] [979]  INFO RPI raspberrypi.cpp:122 Mode: 2028x1080 fmt RG12 Score: 1583.84 (best 500)
[0:04:54.102233000] [979]  INFO RPI raspberrypi.cpp:122 Mode: 1332x990 fmt pRAA Score: 3456.22 (best 500)
[0:04:54.102592000] [979]  INFO RPI raspberrypi.cpp:122 Mode: 4056x3040 fmt pRCC Score: 887 (best 500)
[... 52 lines cut ...]
[0:04:58.490871000] [979]  WARN RPI raspberrypi.cpp:1771 Dropping unmatched input frame in stream Unicam Embedded
[0:04:58.491510000] [979]  WARN RPI raspberrypi.cpp:1808 Dropping unmatched input frame in stream Unicam Image
[0:04:58.577332000] [979]  WARN RPI raspberrypi.cpp:1771 Dropping unmatched input frame in stream Unicam Embedded
[0:04:58.578099000] [979]  WARN RPI raspberrypi.cpp:1808 Dropping unmatched input frame in stream Unicam Image
[0:04:58.578557000] [979]  WARN RPI raspberrypi.cpp:1771 Dropping unmatched input frame in stream Unicam Embedded
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to import fd 20
Aborted
```

`vcgencmd` output for cases 1-4:
```
$ vcgencmd get_camera
supported=1 detected=0
```

`vcgencmd` output for case 5:
```
$ vcgencmd get_camera
supported=0 detected=0
```

Just in case: rpi doc tell this:  "start_x=1 should be specified when using the camera module. Enabling the camera via raspi-config will set this automatically."

https://www.raspberrypi.com/documentation/computers/config_txt.html

closed 2022-01-13 by naushir