raspberrypi/rpicam-apps #209: [BUG] libcamera-hello doesn't show preview on console

CarlFK, 2021-12-27:
**Describe the bug**
libcamera-hello doesn't show preview on console 

**Bug report**
2021-10-30-raspios-bullseye-armhf-lite  (so no gui/X)
libcamera-hello - no image and a message: "Preview window unavailable"

enable legacy,  raspistill will show  an image

davidplowman, 2021-12-29:
Just so we can be sure what you're running, can you please report the output of:

* `uname -a`
* `vcgencmd version`
* `libcamera-hello --version`

It sounds like you have a keyboard and screen plugged directly into your Pi, and you're logged in directly rather than through ssh. Can you confirm that?

Running without X WIndows, you should get the DRM/KMS-based preview window automatically. Could you check your /boot/config.txt contains `dtovleray=vc4-kms-v3d`. Actually pasting the whole thing in here would be helpful. Thanks!

CarlFK, 2022-01-04:
gah - my bad.  I was using an old config.txt that did not have dtovleray=vc4-kms-v3d
using the stock settings, I get the DRM/KMS-based preview.
sorry!

closed 2022-01-04 by CarlFK