raspberrypi/rpicam-apps #210: Segmentation fault during long stream

dustinkerstein, 2021-12-27:
When running `LIBCAMERA_LOG_LEVELS=0 libcamera-vid --frames 1000000 --framerate 30 --width 1600 --height 1200 --gain 12 --awb auto --shutter 55000 --codec mjpeg --quality 90 -o /dev/null` I eventually get a segmentation fault. Log here - [segfault.txt](https://github.com/raspberrypi/libcamera-apps/files/7781249/segfault.txt)

Are others able to replicate?

I am testing with an Arducam Mini IMX477 module + Compute Module 4 + Official IO Board + Bullseye Lite 64bit + 5.10.87-v8+ Kernel running headless over ssh.

```
$ libcamera-hello --version
libcamera-apps build: 856857bb698a-intree-dirty 26-12-2021 (22:39:04)
libcamera build: v0.0.0+3369-8ff5a8d5
$ vcgencmd version
Nov 18 2021 16:16:49
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
```

fire-eggs, 2021-12-28:
My attempt ran to completion overnight. Bullseye 64big; RPi 4Bx8G; Raspberry Pi High Quality camera module.

dustinkerstein, 2021-12-28:
Interesting. I just tested again with a different Arducam IMX477 module (B2040 vs. B2049) and I was able to replicate the seg fault after 10 or so minutes. Could be specific to the CM4. Is there any other debug that I could get that would be helpful?

davidplowman, 2021-12-29:
If you could run it under gdb a call stack from the crash might be helpful, even if it's non-debug build. Obviously a call stack from a debug build would be even better!

dustinkerstein, 2021-12-29:
Here's a backtrace on a non-debug build:

```
Thread 2 "libcamera-vid" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ff67acf40 (LWP 1200)]
0x0000007ff7d629d0 in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /lib/aarch64-linux-gnu/libstdc++.so.6
(gdb) backtrace
#0  0x0000007ff7d629d0 in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#1  0x0000007ff7f8fd6c in std::pair<std::_Rb_tree_iterator<CompletedRequest*>, bool> std::_Rb_tree<CompletedRequest*, CompletedRequest*, std::_Identity<CompletedRequest*>, std::less<CompletedRequest*>, std::allocator<CompletedRequest*> >::_M_insert_unique<CompletedRequest* const&>(CompletedRequest* const&) () from /home/pi/GitHub/libcamera-apps/build/core/libcamera_app.so
#2  0x0000007ff7f8552c in LibcameraApp::requestComplete(libcamera::Request*) () from /home/pi/GitHub/libcamera-apps/build/core/libcamera_app.so
#3  0x0000007ff7a1e8a0 in libcamera::Camera::requestComplete(libcamera::Request*) () from /usr/local/lib/aarch64-linux-gnu/libcamera.so.0
#4  0x0000007ff7a4ede0 in libcamera::PipelineHandler::completeRequest(libcamera::Request*) () from /usr/local/lib/aarch64-linux-gnu/libcamera.so.0
#5  0x0000007ff7a7d918 in libcamera::RPiCameraData::checkRequestCompleted() () from /usr/local/lib/aarch64-linux-gnu/libcamera.so.0
#6  0x0000007ff7a7ed48 in libcamera::RPiCameraData::handleState() () from /usr/local/lib/aarch64-linux-gnu/libcamera.so.0
#7  0x0000007ff7a11b50 in libcamera::ipa::RPi::IPAProxyRPi::statsMetadataCompleteThread(unsigned int, libcamera::ControlList const&) () from /usr/local/lib/aarch64-linux-gnu/libcamera.so.0
#8  0x0000007ff7999740 in libcamera::Object::message(libcamera::Message*) () from /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0
#9  0x0000007ff799b880 in libcamera::Thread::dispatchMessages(libcamera::Message::Type) () from /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0
#10 0x0000007ff7993660 in libcamera::EventDispatcherPoll::processEvents() () from /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0
#11 0x0000007ff799b5d0 in libcamera::Thread::exec() () from /usr/local/lib/aarch64-linux-gnu/libcamera-base.so.0
#12 0x0000007ff7a24058 in libcamera::CameraManager::Private::run() () from /usr/local/lib/aarch64-linux-gnu/libcamera.so.0
#13 0x0000007ff7d76cac in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#14 0x0000007ff7aea628 in start_thread (arg=0x7ff67ac840) at pthread_create.c:477
#15 0x0000007ff7be901c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

I'll get a debug build setup soon and will post a call stack.

davidplowman, 2021-12-29:
Thanks, that's pretty interesting already. It means the crash is probably happening here: 
https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L724

So perhaps that means someone else is trampling on that "set" object, such as here:
https://github.com/raspberrypi/libcamera-apps/blob/main/core/libcamera_app.cpp#L511

Looking at that I guess it seems pretty obvious that it needs to have a mutex to prevent the concurrent access, though it's probably worth having a think in case there's a better solution.

dustinkerstein, 2021-12-29:
And here's a debug stack trace:

```
Thread 2 "libcamera-vid" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ff6176f40 (LWP 1313)]
0x0000007ff7bcba18 in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /lib/aarch64-linux-gnu/libstdc++.so.6
(gdb) backtrace
#0  0x0000007ff7bcba18 in std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) () from /lib/aarch64-linux-gnu/libstdc++.so.6
#1  0x0000007ff7f46b44 in std::_Rb_tree<CompletedRequest*, CompletedRequest*, std::_Identity<CompletedRequest*>, std::less<CompletedRequest*>, std::allocator<CompletedRequest*> >::_M_insert_<CompletedRequest* const&, std::_Rb_tree<CompletedRequest*, CompletedRequest*, std::_Identity<CompletedRequest*>, std::less<CompletedRequest*>, std::allocator<CompletedRequest*> >::_Alloc_node> (this=0x7fffffed58,
    __x=0x0, __p=0x7ff001fc50, __v=@0x7ff6175278: 0x7ff001fc80, __node_gen=...)
    at /usr/include/c++/10/bits/stl_tree.h:1818
#2  0x0000007ff7f40968 in std::_Rb_tree<CompletedRequest*, CompletedRequest*, std::_Identity<CompletedRequest*>, std::less<CompletedRequest*>, std::allocator<CompletedRequest*> >::_M_insert_unique<CompletedRequest* const&> (this=0x7fffffed58, __v=@0x7ff6175278: 0x7ff001fc80)
    at /usr/include/c++/10/bits/stl_tree.h:2159
#3  0x0000007ff7f3ab60 in std::set<CompletedRequest*, std::less<CompletedRequest*>, std::allocator<CompletedRequest*> >::insert (this=0x7fffffed58, __x=@0x7ff6175278: 0x7ff001fc80)
    at /usr/include/c++/10/bits/stl_set.h:512
#4  0x0000007ff7f307b4 in LibcameraApp::requestComplete (this=0x7fffffebf0, request=0x5555607270)
    at /home/pi/GitHub/libcamera-apps/core/libcamera_app.cpp:724
#5  0x0000007ff7f58380 in libcamera::BoundMethodMember<LibcameraApp, void, libcamera::Request*>::activate (this=0x5555607eb0, args#0=0x5555607270, deleteMethod=false)
    at /usr/local/include/libcamera/libcamera/base/bound_method.h:180
#6  0x0000007ff777316c in libcamera::Signal<libcamera::Request*>::emit (this=0x7ff0025910,
    args#0=0x5555607270) at ../include/libcamera/base/signal.h:145
#7  0x0000007ff7770520 in libcamera::Camera::requestComplete (this=0x7ff0025890, request=0x5555607270)
    at ../src/libcamera/camera.cpp:1237
#8  0x0000007ff77d9ab0 in libcamera::PipelineHandler::completeRequest (this=0x7ff0000e10,
    request=0x5555607270) at ../src/libcamera/pipeline_handler.cpp:472
#9  0x0000007ff78422dc in libcamera::RPiCameraData::checkRequestCompleted (this=0x7ff00011b0)
    at ../src/libcamera/pipeline/raspberrypi/raspberrypi.cpp:1780
--Type <RET> for more, q to quit, c to continue without paging--
#10 0x0000007ff7842254 in libcamera::RPiCameraData::handleState (this=0x7ff00011b0)
    at ../src/libcamera/pipeline/raspberrypi/raspberrypi.cpp:1751
#11 0x0000007ff7841434 in libcamera::RPiCameraData::statsMetadataComplete (this=0x7ff00011b0,
    bufferId=0, controls=...) at ../src/libcamera/pipeline/raspberrypi/raspberrypi.cpp:1499
#12 0x0000007ff78588e8 in libcamera::BoundMethodMember<libcamera::RPiCameraData, void, unsigned int, libcamera::ControlList const&>::activate (this=0x7ff0012700, args#0=0, args#1=..., deleteMethod=false)
    at ../include/libcamera/base/bound_method.h:180
#13 0x0000007ff7753270 in libcamera::Signal<unsigned int, libcamera::ControlList const&>::emit (
    this=0x7ff0010438, args#0=0, args#1=...) at ../include/libcamera/base/signal.h:145
#14 0x0000007ff774d968 in libcamera::ipa::RPi::IPAProxyRPi::statsMetadataCompleteThread (
    this=0x7ff0010350, bufferId=0, controls=...) at src/libcamera/proxy/raspberrypi_ipa_proxy.cpp:577
#15 0x0000007ff7762088 in libcamera::BoundMethodMember<libcamera::ipa::RPi::IPAProxyRPi, void, unsigned int, libcamera::ControlList const&>::invoke (this=0x7ff0010310, args#0=0, args#1=...)
    at ../include/libcamera/base/bound_method.h:191
#16 0x0000007ff7762638 in libcamera::BoundMethodArgs<void, unsigned int, libcamera::ControlList const&>::invokePack<0ul, 1ul, void> (this=0x7ff0010310, pack=0x7fd4004020)
    at ../include/libcamera/base/bound_method.h:115
#17 0x0000007ff7761e64 in libcamera::BoundMethodArgs<void, unsigned int, libcamera::ControlList const&>::invokePack (this=0x7ff0010310, pack=0x7fd4004020) at ../include/libcamera/base/bound_method.h:124
#18 0x0000007ff743dae0 in libcamera::InvokeMessage::invoke (this=0x7fd4003b80)
    at ../src/libcamera/base/message.cpp:153
#19 0x0000007ff743e0a8 in libcamera::Object::message (this=0x7ff00103e8, msg=0x7fd4003b80)
    at ../src/libcamera/base/object.cpp:186
#20 0x0000007ff7444394 in libcamera::Thread::dispatchMessages (this=0x55555f13c0,
    type=libcamera::Message::None) at ../src/libcamera/base/thread.cpp:627
#21 0x0000007ff742fea4 in libcamera::EventDispatcherPoll::processEvents (this=0x7ff00069f0)
    at ../src/libcamera/base/event_dispatcher_poll.cpp:147
#22 0x0000007ff7443940 in libcamera::Thread::exec (this=0x55555f13c0)
--Type <RET> for more, q to quit, c to continue without paging--
    at ../src/libcamera/base/thread.cpp:341
#23 0x0000007ff7780974 in libcamera::CameraManager::Private::run (this=0x55555f13b0)
    at ../src/libcamera/camera_manager.cpp:120
#24 0x0000007ff74438bc in libcamera::Thread::startThread (this=0x55555f13c0)
    at ../src/libcamera/base/thread.cpp:319
#25 0x0000007ff744790c in std::__invoke_impl<void, void (libcamera::Thread::*)(), libcamera::Thread*>
    (
[... cut ...]
#29 0x0000007ff7447720 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> > >::_M_run (this=0x5555601b60) at /usr/include/c++/10/thread:215
#30 0x0000007ff7bdfcac in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#31 0x0000007ff7953628 in start_thread (arg=0x7ff6176840) at pthread_create.c:477
#32 0x0000007ff7a5201c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

Let me know if there's anything else I can help with.

davidplowman, 2021-12-29:
Thanks, I think that too agrees with the prevailing theory. There's what I believe should be a fix here: https://github.com/raspberrypi/libcamera-apps/pull/214. Actually there was a bit of simplification possible, which I hope has resolved things without chucking in an extra mutex.

dustinkerstein, 2021-12-29:
Sounds good!

naushir, 2022-01-04:
Closing this as https://github.com/raspberrypi/libcamera-apps/pull/214 has been merged.

closed 2022-01-04 by naushir