raspberrypi/rpicam-apps #211: [BUG]

sv87411, 2021-12-27:
**Describe the bug**
Apologies if this is specific to the Raspbian version of the libcamera-apps build. Also apologies if this isn't a bug and the libcamera apps have been specifically designed not to send command output to stdout. I still feel this is a bug though.

Experienced on Raspbian version:

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
Using libcamera version:

```
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:49:17)
libcamera build: v0.0.0+3156-f4070274
```

Command output from all libcamera apps does not appear to be sent to stdout. I cannot redirect any command output to a file. This happens both on the direct connect console and an SSH session to the Pi.

Example. Redirect version information output to a file called 'libcamera_version.txt'

libcamera-jpeg --version > libcamera_version.txt

Result. Version information only displayed on the console and libcamera_version.txt file remains empty.

This appears to happen with all libcamera app commands.

I would have expected that all command output would be sent to stdout for a linux command.

**Bug report**
``camera-bug-report`` tool not available on Raspbian.

**Additional context**
None

fire-eggs, 2021-12-28:
Unfortunately, unlike most *nix apps, all the log/help output from the libcamera apps is being sent to `stderr`, not `stdout`.

Which means to redirect the output to a file, you must redirect `stderr`, e.g.:
```
libcamera-jpeg --version 2> libcamera_version.txt
```

Note the '2' in front of the angle bracket.

6by9, 2021-12-28:
This is intentional.
In the same way as raspistill/vid do, libcamera-jpeg supports sending the encoded output to stdout, at which point you do not want any logging going to stdout and mixing with the encoded data.

There would be a case for the help text to be sent to stdout as that is only generated when there is no encoded output data.

sv87411, 2021-12-28:
Thank you. I wasn't aware that this is how raspistill etc behaved. Still seems odd to me, but I can re-direct ```stderr```. 

I just checked and ```raspistill --help``` output does get sent to ```stdout```, so if ```libcamera-still``` is mirroring ```raspistill``` behaviour it should at least send help (and maybe version info) to ```stdout```.

It was while I was trying to pipe the help output of ```libcamera-jpeg``` to grep to find a specific setting that I noticed this behaviour. So it would be good if at least the help output went to ```stdout```.

I still think it's odd behaviour though. Surely if the actual camera output is sent to ```stdout``` using ```-o -``` then logging could go to ```stderr``` in that case, but at all other times when output goes to file then logging gets sent to ```stdout```. This would be more in line with standard *nix apps.

closed 2021-12-28 by sv87411