raspberrypi/rpicam-apps #213: [BUG] libcamera-apps do not compile

emsruderer, 2021-12-28:
**Describe the bug**
A clear and concise description of what the bug is.
[ 73%] Building CXX object core/CMakeFiles/libcamera_app.dir/options.cpp.o
/home/pi/libcamera-apps/core/libcamera_app.cpp: In member function ‘void LibcameraApp::setupCapture()’:
/home/pi/libcamera-apps/core/libcamera_app.cpp:666:54: error: ‘const class libcamera::FileDescriptor’ has no member named ‘get’
  666 |     if (i == buffer->planes().size() - 1 || plane.fd.get() != buffer->planes()[i + 1].fd.get())
      |                                                      ^~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [core/CMakeFiles/libcamera_app.dir/build.make:82: core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o] Fehler 1
make[2]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[1]: *** [CMakeFiles/Makefile2:274: core/CMakeFiles/libcamera_app.dir/all] Fehler 2
make: *** [Makefile:171: all] Fehler 2
*

fire-eggs, 2021-12-28:
See issues #203 and #201.

emsruderer, 2021-12-29:
Thanks, I am sorry to have missed those claims.

closed 2021-12-29 by emsruderer