raspberrypi/rpicam-apps #215: feature request:  --identify

CarlFK, 2021-12-30:
I wold like something to help determine if and what camera is plugged in.

libcamera-hello alone (example below) is too verbose and does too much more which dilutes my goal.
I just want to know if a camera is hooked up and somewhat functioning (having the ribbon cable backwards does not count ;)

I don't want it to send output anywhere, (screen or file or anything that might cause an error, like Read Only fs)
I'm not sure I even want it to grab a frame.  perhaps this could be optional: --identify [frame]  would display stats about the frame: res, depth and whatever other metadata comes with it.

maybe the model, or name of the driver/module/whatever code is specific to different cameras.
I'm not sure where this fits, but if this can help with a Pi's overlay files / device tree, 
```
pi@rpi-x:~ $ libcamera-hello -v 
Acquired camera /base/soc/i2c0mux/i2c@1/ov5647@36
```
I'm assuming it is getting that information from 
`/boot/config.txt:  dtoverlay=ov5647`
```
strings /boot/overlays/ov5647.dtbo
ov5647@36
```
Which isn't exactly redundant given it could have started from:
camera_auto_detect=1 

I'd like to know what it detected, and know if what is in the DT even makes sense - like can the camera hardware confirm that it is indeed a ov5647 compliant device? 

This is in the direction of what I want:
`libcamera-hello 2>&1 | grep "no cameras"`
except that is fragile, like it assumes grep is installed and on the path, the local is english, the message doesn't change, etc.

```
pi@rpi-x:~ $ libcamera-hello
Preview window unavailable
[88:43:34.966917025] [4498]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[88:43:35.030206950] [4500] ERROR V4L2 v4l2_device.cpp:87 /dev/video0[-1:cap]: Failed to open V4L2 device: Remote I/O error
ERROR: *** no cameras available ***
```

```
pi@rpi-x:~ $ libcamera-hello
Preview window unavailable
[88:31:13.940499864] [4339]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
[88:31:14.024815154] [4339]  INFO Camera camera.cpp:937 configuring streams: (0) 1296x972-YUV420
[88:31:14.024956727] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt BG10 Score: 2067 (best 2067)
[88:31:14.025022505] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt BG10 Score: 1849.67 (best 1849.67)
[88:31:14.025062653] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt BG10 Score: 1500 (best 1500)
[88:31:14.025097634] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BG10 Score: 3796 (best 1500)
[88:31:14.025135356] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt BA81 Score: 4796 (best 1500)
[88:31:14.025169838] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 2592x1944 fmt pBAA Score: 1567 (best 1500)
[88:31:14.025205097] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pBAA Score: 1349.67 (best 1349.67)
[88:31:14.025240245] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 1296x972 fmt pBAA Score: 1000 (best 1000)
[88:31:14.025274633] [4341]  INFO RPI raspberrypi.cpp:122 Mode: 640x480 fmt pBAA Score: 3296 (best 1000)
[88:31:14.025396874] [4341]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected mode: 1296x972-pBAA
[88:31:14.046384201] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[88:31:14.046469312] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[88:31:14.046506867] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[88:31:14.046539997] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[88:31:14.046571756] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[88:31:14.046605015] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[88:31:14.046680145] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[88:31:14.046724996] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[88:31:14.046761866] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[88:31:14.046792422] [4345]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[88:31:14.048559637] [4341]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[88:31:14.048595285] [4341]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[88:31:14.048622100] [4341]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[88:31:14.048646267] [4341]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[88:31:14.121209583] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (7 left)
[88:31:14.166216652] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[88:31:14.199893839] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[88:31:14.234019894] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[88:31:14.266403030] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[88:31:14.299909773] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[88:31:14.333300924] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[88:31:14.366647556] [4341]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```

davidplowman, 2021-12-30:
In the latest version of libcamera-apps (which we're planning to release shortly now) you can use `libcamera-hello --list` which lists all the attached cameras and their native modes, like so:
```
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 
           'SRGGB12_CSI2P' : 2028x1080 2028x1520 4056x3040
```
This example shows a single camera (an imx477, aka. the HQ cam), with a single 10-bit readout mode and 3 12-bit modes (one of which is the full native sensor resolution).

CarlFK, 2021-12-30:
nice. I was wondering about how multiple cameras should be handled, this is it.  issue closed!

closed 2021-12-30 by CarlFK