raspberrypi/rpicam-apps #217: StopCamera() takes a long time after streaming many frames

dustinkerstein, 2022-01-06:
When running: `libcamera-vid --frames 18000 --framerate 30 --width 1920 --height 1080 --codec h264 -o /dev/shm/camera.mp4` with the following stopwatch debug:
```
auto start = high_resolution_clock::now();
app.StopCamera(); // stop complains if encoder very slow to close
auto stop = high_resolution_clock::now();
auto duration = duration_cast<seconds>(stop - start);
std::cerr << "StopCamera took: " << duration.count() << " seconds" << std::endl;
start = high_resolution_clock::now();
app.StopEncoder();
stop = high_resolution_clock::now();
duration = duration_cast<seconds>(stop - start);
std::cerr << "StopEncoder took: " << duration.count() << " seconds" << std::endl;
```
Results in:
```
StopCamera took: 7 seconds
StopEncoder took: 0 seconds
```
Bug report here - [bug18000.txt](https://github.com/raspberrypi/libcamera-apps/files/7823421/bug18000.txt)

I happened to let a 1600x1200 mjpeg stream run for ~2 hours yesterday and it took >10minutes for that function to finish. There doesn't seem to be any ill effects beyond the long delay, but is there anything that could be tweaked to minimize this delay?
```
$ uname -a
Linux raspberrypi 5.10.87-v8+ #1 SMP PREEMPT Sat Dec 18 19:47:37 GMT 2021 aarch64 GNU/Linux
$ ./libcamera-hello --version
libcamera-apps build: ce0c6360ed67-intree-dirty 06-01-2022 (16:21:40)
libcamera build: v0.0.0+3369-8ff5a8d5
$ vcgencmd version
Nov 18 2021 16:16:49
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
```

davidplowman, 2022-01-07:
Thanks for reporting this. Indeed I see this too, it seems to be deleting the libcamera Request objects that takes "forever". Each Request seems to be hanging on to a huge number of "signals" (whatever those are) which seems surprising. But I don't really understand anything down there so I'll probably go and ask over at libcamera.

naushir, 2022-01-07:
@kbingham might be able to provide some insight into this.

davidplowman, 2022-01-07:
What I see happening is:

It sits for ever in `Object::disconnect()` iterating through thousands of entries in `signals_`. As far as I can see, `Object::connect()` adds another entry to this list every time I queue a request (`PipelineHandler::queueRequest()`). I'm not clear on how those entries should get removed (or indeed what they are!).

kbingham, 2022-01-07:
Yikes! That doesn't sound good.
The request is connected with signals to allow the pipeline handler to inform the Request that it has completed, or that a buffer is completed. Applications can listen to those signals (per request) if they wish to know that a buffer has completed (before the full request).

So if you had a request that had 3 streams, you could theoretcially start processing the raw stream as soon as it has been captured,  and then process the ISP streams after they complete, or you can do as most implementations do now - and wait for the full request to complete.

I expect we're re-connecting signals on every queue - and not clearing them down so it could easily be a core libcamera bug and worth posting on the list.

kbingham, 2022-01-07:
that was easily confirmed anyway:

```
diff --git a/src/libcamera/base/signal.cpp b/src/libcamera/base/signal.cpp
index 9df45d079a42..000797109158 100644
--- a/src/libcamera/base/signal.cpp
+++ b/src/libcamera/base/signal.cpp
@@ -9,6 +9,8 @@
 
 #include <libcamera/base/mutex.h>
 
+#include <libcamera/base/log.h>
+
 /**
  * \file base/signal.h
  * \brief Signal & slot implementation
@@ -35,6 +37,10 @@ void SignalBase::connect(BoundMethodBase *slot)
        if (object)
                object->connect(this);
        slots_.push_back(slot);
+
+
+       if (slots_.size() > 10)
+               LOG(Fatal) << "Signal slots are connected more than 10 times?";
 }
```
 
Results in:

```
./build/gcc/src/qcam/qcam -r gles
[49:14:00.684424925] [1388856]  INFO IPAManager ipa_manager.cpp:138 libcamera is not installed. Adding '/home/kbingham/iob/libcamera/libcamera/build/gcc/src/ipa' to the IPA search path
[49:14:00.686451868] [1388856]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3444-417cdd55-dirty (2022-01-07T11:27:15+00:00)
[49:14:00.765146711] [1388864]  INFO V4L2 v4l2_subdevice.cpp:258 'Sensor B': Using /dev/v4l-subdev1
[49:14:00.766630775] [1388864]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'Sensor B'
[49:14:00.766648248] [1388864]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[49:14:00.766658096] [1388864]  WARN CameraSensor camera_sensor.cpp:405 'Sensor B': Failed to retrieve the camera location
[49:14:00.766673686] [1388864]  INFO V4L2 v4l2_subdevice.cpp:258 'Debayer B': Using /dev/v4l-subdev3
[49:14:00.766720924] [1388864]  INFO V4L2 v4l2_subdevice.cpp:258 'Scaler': Using /dev/v4l-subdev5
[49:14:00.767090518] [1388864]  WARN IPAManager ipa_manager.cpp:280 IPA module /home/kbingham/iob/libcamera/libcamera/build/gcc/src/ipa/vimc/ipa_vimc.so signature regarded as valid
[49:14:00.767597910] [1388864]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/kbingham/iob/libcamera/libcamera/src/ipa/vimc/data'
[49:14:03.475416599] [1388856]  INFO Camera camera.cpp:1030 configuring streams: (0) 1920x1080-YUYV
[49:14:12.665088025] [1388864] FATAL default signal.cpp:43 Signal slots are connected more than 10 times?
Backtrace:
libcamera::SignalBase::connect(libcamera::BoundMethodBase*)+0x10e (../../src/libcamera/base/signal.cpp:44)
void libcamera::Signal<>::connect<libcamera::PipelineHandler, libcamera::PipelineHandler::queueRequest(libcamera::Request*)::{lambda()#1}, (void*)0>(libcamera::PipelineHandler*, libcamera::PipelineHandler::queueRequest(libcamera::Request*)::{lambda()#1}, libcamera::ConnectionType)+0x72 (../../include/libcamera/base/signal.h:71)
libcamera::PipelineHandler::queueRequest(libcamera::Request*)+0x155 (../../src/libcamera/pipeline_handler.cpp:372)
libcamera::BoundMethodMember<libcamera::PipelineHandler, void, libcamera::Request*>::invoke(libcamera::Request*)+0x7f (../../include/libcamera/base/bound_method.h:191)
std::enable_if<std::is_void<void>::value, void>::type libcamera::BoundMethodArgs<void, libcamera::Request*>::invokePack<0ul, void>(libcamera::BoundMethodPackBase*, std::integer_sequence<unsigned long, 0ul>)+0x4a (../../include/libcamera/base/bound_method.h:116)
libcamera::BoundMethodArgs<void, libcamera::Request*>::invokePack(libcamera::BoundMethodPackBase*)+0x27 (../../include/libcamera/base/bound_method.h:125)
libcamera::InvokeMessage::invoke()+0x46 (../../src/libcamera/base/message.cpp:154)
libcamera::Object::message(libcamera::Message*)+0x50 (../../src/libcamera/base/object.cpp:188)
libcamera::Thread::dispatchMessages(libcamera::Message::Type)+0x317 (../../src/libcamera/base/thread.cpp:628)
libcamera::EventDispatcherPoll::processEvents()+0x38 (../../src/libcamera/base/event_dispatcher_poll.cpp:150)
libcamera::Thread::exec()+0x70 (../../src/libcamera/base/thread.cpp:340)
libcamera::CameraManager::Private::run()+0x112 (../../src/libcamera/camera_manager.cpp:122)
libcamera::Thread::startThread()+0xf2 (../../src/libcamera/base/thread.cpp:320)
void std::__invoke_impl<void, void (libcamera::Thread::*)(), libcamera::Thread*>(std::__invoke_memfun_deref, void (libcamera::Thread::*&&)(), libcamera::Thread*&&)+0x6a (/usr/include/c++/11/bits/invoke.h:74)
std::__invoke_result<void (libcamera::Thread::*)(), libcamera::Thread*>::type std::__invoke<void (libcamera::Thread::*)(), libcamera::Thread*>(void (libcamera::Thread::*&&)(), libcamera::Thread*&&)+0x3b (/usr/include/c++/11/bits/invoke.h:97)
void std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> >::_M_invoke<0ul, 1ul>(std::_Index_tuple<0ul, 1ul>)+0x47 (/usr/include/c++/11/bits/std_thread.h:253)
std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> >::operator()()+0x1c (/usr/include/c++/11/bits/std_thread.h:260)
std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (libcamera::Thread::*)(), libcamera::Thread*> > >::_M_run()+0x20 (/usr/include/c++/11/bits/std_thread.h:211)
??? [0x00007f54112b2694] (/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.29 [0x00007f54112b2694])
start_thread+0x277 (./nptl/pthread_create.c:435)
__clone+0x44 (../sysdeps/unix/sysv/linux/x86_64/clone.S:102)
make: *** [Makefile:92: gcc-run] Aborted (core dumped)

```

davidplowman, 2022-01-07:
Thanks, Kieran. I guess I'll go and file a bug for this and link it back to this issue here.

kbingham, 2022-01-07:
At least already found a fix, will need confirming with the others.

kbingham, 2022-01-07:
```
diff --git a/src/libcamera/request.cpp b/src/libcamera/request.cpp
index fe1bccd40b23..7dab9760fd51 100644
--- a/src/libcamera/request.cpp
+++ b/src/libcamera/request.cpp
@@ -178,6 +178,8 @@ void Request::Private::emitPrepareCompleted()
 {
        prepared_ = true;
        prepared.emit();
+
+       prepared.disconnect();
 }
 
 /**

```

davidplowman, 2022-01-07:
Cool! Shall I leave off the bug report, then, as we seem to be ahead of the game on this one?

kbingham, 2022-01-07:
Yup, I'll handle it from here.

dustinkerstein, 2022-01-07:
Great stuff! Thanks for the quick investigation. I had a hunch it might be more of a core libcamera issue, but forgot to mention that. Will give the test a fix as soon as it's pushed.

kbingham, 2022-01-07:
@dustinkerstein The patch is posted to the development list : https://patchwork.libcamera.org/patch/15259/ - https://patchwork.libcamera.org/patch/15259/mbox/ if you wish to try it.

dustinkerstein, 2022-01-07:
@kbingham patch looks good to me from quick testing of ~15 minute streams.

naushir, 2022-01-13:
I'll close this down now as the patches are going to be applied in libcamera.

closed 2022-01-13 by naushir