raspberrypi/rpicam-apps #218: [BUG] Could not open any dmaHeap device

lukexi, 2022-01-07:
Hi, I'm trying to run libcamera-apps on a Pi 4 8GB running Arch Linux ARM. It fails like this:
```
libcamera-still -t 1000 -o test.jpg
--------------------------------------------------------------------------------
[0:21:32.190277860] [634]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3369-8ff5a8d5
[0:21:32.200527323] [635] ERROR RPI dma_heaps.cpp:53 Could not open any dmaHeap device
[0:21:32.200579138] [635] ERROR RPI raspberrypi.cpp:1074 Failed to register camera: -12
```

I've just build libcamera and libcamera-apps from git.

Here is the bug.txt that contains all the system details...
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/7827611/bug.txt)

naushir, 2022-01-07:
I see you are running an Arch Linux distribution.  It looks like this distro does not have the dmabuf heap allocation kernel drivers used by the libcamera pipeline handler, and is causing this failure.

Any reason you are using that particular distribution and not Raspberry Pi OS?

lukexi, 2022-01-07:
Yes, the other software system I'm working with currently expects to run specifically on Arch Linux... I guess I can try to add the drivers?

naushir, 2022-01-07:
Yes, you could try adding the dma-heap drivers and see if that works.
I'll close this issue down now, please do re-open if you have any additional queries related to libcamera-apps.

closed 2022-01-07 by naushir

lukexi, 2022-01-07:
Thanks! I'm investigating this and it looks like Arch Linux ARM gets its raspberry pi kernel mostly unmodified from github.com/raspberrypi/linux, and it's pointing to a commit only a few days old. The DMABUF_HEAPS option is enabled https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-rpi/config . And I see a dma_heap device in my device tree.
```
[dynamic@dynamite2 linux]$ ls /dev/dma_heap/
linux,cma  system
```

lukexi, 2022-01-07:
I see, `/dev/dma_heap/linux,cma` has root-only permissions (is that expected?). Running with sudo gives me a different error.
```
[dynamic@dynamite2 ~]$ sudo libcamera-still -t 1000 -o test.jpg
[14:05:14.947659624] [2880]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3369-8ff5a8d5
[14:05:14.958320458] [2881]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[14:05:14.958361569] [2881]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[14:05:14.958388180] [2881] ERROR CameraSensor camera_sensor.cpp:535 'imx477 10-001a': Camera sensor does not support test pattern modes.
[14:05:14.975645754] [2881] ERROR RPI raspberrypi.cpp:1173 Unicam driver does not use the MediaController, please update your kernel!
[14:05:14.976299402] [2881] ERROR RPI raspberrypi.cpp:1074 Failed to register camera: -22
ERROR: *** no cameras available ***
```

lukexi, 2022-01-07:
OK, and then it looks like those errors were just due to some incorrect config.txt settings — all good now, thanks for the help!
```
[dynamic@dynamite2 ~]$ sudo libcamera-still -t 1000 -o test.jpg
[0:25:38.094620509] [590]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3369-8ff5a8d5
[0:25:38.110484528] [591]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:25:38.110532398] [591]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:25:38.110570232] [591] ERROR CameraSensor camera_sensor.cpp:535 'imx477 10-001a': Camera sensor does not support test pattern modes.
[0:25:38.144850898] [591]  INFO RPI raspberrypi.cpp:1256 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:25:38.145926806] [590]  INFO Camera camera.cpp:1028 configuring streams: (0) 2028x1520-YUV420
[0:25:38.146309306] [591]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[0:25:38.253943731] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:25:38.254107435] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:25:38.254182435] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:25:38.254245083] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:25:38.254308194] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:25:38.254865898] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:25:38.254972046] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:25:38.255036194] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:25:38.255113565] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:25:38.255174324] [596]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[0:25:38.256990787] [591]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:25:38.257064213] [591]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:25:38.257116250] [591]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:25:38.257165583] [591]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:25:38.257213713] [591]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:25:38.257262528] [591]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:25:39.998647490] [590]  INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[0:25:39.999665175] [591]  INFO RPI raspberrypi.cpp:730 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[0:25:40.077157304] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:25:40.077256434] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:25:40.077296471] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:25:40.077335416] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:25:40.077373045] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:25:40.077410990] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:25:40.077448749] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:25:40.077482841] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[0:25:40.077521841] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:25:40.077557804] [600]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 2
Still capture image received
```

naushir, 2022-01-08:
Great, good it worked!

lukexi, 2022-01-25:
For future travelers: to allow accessing /dev/dma_heap on Arch Linux ARM for Pi without needing `sudo` (which is what causes the error in this bug report), add the following line to 
`/etc/udev/rules.d/raspberrypi.rules`:
`SUBSYSTEM=="dma_heap", GROUP="video", MODE="0660"`

(and ensure your user is a member of the `video` group)