raspberrypi/rpicam-apps #219: libcamera-app fails to compile

ahatstat, 2022-01-07:
When attempting to build libcamera-apps commit b9a6923... on Bullseye 32-bit OS following [these directions](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps) the build fails.
Before building libcamera-apps I did sudo rpi-update and built libcamera.  Any advice?

```
[ 76%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
/home/pi/libcamera-apps/core/libcamera_app.cpp: In member function ‘void LibcameraApp::ConfigureStill(unsigned int)’:
/home/pi/libcamera-apps/core/libcamera_app.cpp:257:24: error: ‘struct libcamera::StreamConfiguration’ has no member named ‘colorSpace’
  257 |  configuration_->at(0).colorSpace = libcamera::ColorSpace::Jpeg;
      |                        ^~~~~~~~~~
```

fire-eggs, 2022-01-07:
Looking at `stream.h` in the [libcamera source](https://git.libcamera.org/libcamera/libcamera.git/log/include/libcamera/stream.h), that field was added on Dec 13,2021. Did you fetch the latest [libcamera code](https://git.libcamera.org/libcamera/libcamera.git) before building it?

ahatstat, 2022-01-08:
The guide points to git://linuxtv.org/libcamera.git.  I cloned the latest from git.libcamera.org as you suggested.  However, after building and installing per the guide, libcamera-app still failed to compile for the same reason because I believe it was referring to pre-existing headers in /usr/include/libcamera instead of /usr/local/include/libcamera/libcamera where the new ones were installed.  After resolving this I was able to get libcamera-app to build.  Thank you.

fire-eggs, 2022-01-08:
Glad that helped. Sounds like a bug in the docs then.

davidplowman, 2022-01-09:
I'd have thought that the system should automatically look for headers first in /usr/local/include rather than in /usr/include, just as the runtime linker does a similar thing with libraries. But I'm happy to be corrected on that.

Anyway, I'm glad you solved the problem. I'll mark this issue as resolved shortly, if there's nothing further to add. Thanks!

davidplowman, 2022-01-10:
Please feel free to open another issue if anything else comes up. Thanks!

closed 2022-01-10 by davidplowman