raspberrypi/rpicam-apps #220: [BUG] Preview windows only support YUV420

errolt, 2022-01-09:
**Describe the bug**
Trying to use libcamera-still with a usb camera. I understand that a preview can't be shown, but setting --nopreview doesn't help, and that error message is still displayed. Is this intended behavior?

**Bug report**
I don't know where to get camera-bug-report, it isn't part of a default lite os image. This is the output I'm getting from libcamera-still

> pi@camera:~ $ libcamera-still --nopreview 1 -o test2.jpg
> [0:29:13.224418338] [752]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3156-f4070274
> Stream configuration adjusted
> [0:29:13.399755927] [752]  INFO Camera camera.cpp:937 configuring streams: (0) 640x360-MJPEG
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  Preview windows only support YUV420
> Aborted

**Additional context**
I'm running libcamera-still as it is in apt repo:

> pi@camera:~ $ libcamera-still --version
> libcamera-apps build: 2a38ae93f143 21-10-2021 (14:49:17)
> libcamera build: v0.0.0+3156-f4070274

davidplowman, 2022-01-09:
Hi, it looks like your USB webcam is supplying an MJPEG stream, would that be correct? I'm afraid our libcamera implementation only supports embedded (raw Bayer) sensors, not webcams that produce fully processed images or MJPEG streams.

I agree that the error message that comes out seems to have got a bit confused, but I think the basic problem is that this sort of camera just isn't supported.

I think libcamera does have some code ("pipeline handlers" as they are known) to handle webcams, so it might be worth trying to compile that and see if you get any further. Alternatively you can presumably address your camera as a regular V4L2 sensor through `/dev/video<n>` (for some value of `<n>`).

errolt, 2022-01-10:
Thanks @davidplowman. I take it then that it isn't just the preview being an issue, but that my camera doesn't even want to supply a raw image. It is a Microsoft Corp. LifeCam HD-3000.

Thank you,
Errol

closed 2022-01-10 by errolt

kbingham, 2022-01-10:
It looks like you already have the libcamera pipeline handler for UVC there, as it identified the camera, and selected an MJPEG format. But the issue sounds like libcamera-still doesn't accept a camera that already produces a jpeg image stream.

davidplowman, 2022-01-10:
So far as I know we only ask for the raspberrypi PH, but maybe something has happened to cause uvc to be dragged in. Nonetheless, handling non-raw cameras is well beyond the scope of what we're doing with these applications and unlikely to change for the foreseeable future - so I think we can probably close this one.