raspberrypi/rpicam-apps #221: Exposures with libcamera-still are very different from raspistill.

knutegit, 2022-01-10:
libcamera-still images are much darker in low light than raspistill images.  The images below were taken about 20 minutes apart in late afternoon with some overcast.  The libcamera-still image is from a Java motion detection program and that is why I'm holding the piece of paper in front of the lense.  The second image is using a Java program to capture from raspistill.

RPi3B+, RPi camera version 2.1, up to date bullseye with recently compiled libcamera, libcamera-apps and userland.

[image]
[image]

knutegit, 2022-01-10:
Better pictures.  The raspistill image was taken 5-10 minutes later than the libcamera-still image.

[image]
[image]

naushir, 2022-01-10:
This is not entirely unexpected, the libcamera stack uses an entirely different set of algorithms and image quality tuning parameters when compared with the old stack.  The advantage of libcamera is that everything is open source, so you are free to tweak the settings to match your application.  For your case, you might want to have a look at the AGC/AEC section of our camera tuning guide [here](https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf).

A possibly simpler way would be to use the `--ev` command line argument to increase the metering brightness.  This controls the Exposure Value (EV) in the AGC algorithm.  A value of +1 gives 2x the relative brightness from EV 0.  Likewise a value of -1 would give 1/2x of the relative brightness from EV 0.

naushir, 2022-01-17:
Is there more to discuss on this issue?  If not, I will close this down.

naushir, 2022-01-17:
Closing this now.  Please do reopen if there are further discussions needed.

knutegit, 2022-01-20:
Sorry to come back to this but I think there are two issues here.  The first being the significant light sensitivity difference between libcamera-still and raspistill and an apparent lack of what I can only describe as dynamic range (I'm sure there is another name for it) or exposure adjustment as light levels change.

The other is that if run for a period of time there is inadequate correction or a drift in exposure.  I ran my motion detection program from yesterday afternoon to this morning and the image looked really dark but the light outside was not that low.  I stopped and restarted the program and the image was much brighter.

See this post too:  https://forums.raspberrypi.com/viewtopic.php?t=328083

davidplowman, 2022-01-20:
Hi, did you try those suggestions from a week or so ago? I would recommend adding `--framerate 0`. If you want it even brighter (and have the latest libcamera-apps code) you could also add `--exposure long`.

The question about the exposure getting "stuck" is puzzling. When I first read about it I thought it might be because the timelapse interval is so short that it never gets to run any preview frames in between. But when I tried it, it seems it does still adapt (though slowly, as it should). Are you able to bump up the timelapse interval so that we're guaranteed a few preview frames?

If that doesn't explain it we can always turn on the AGC debug, though it sounds like that would get quite long.

As regards that forum post, it looks to be using OpenCV so I guess it must be using the proprietary Broadcom stack, which works totally differently and shares nothing in common. So I'm not really sure if it's at all related. Trying to figure out what that stack is even trying to do is quite difficult...

knutegit, 2022-01-21:
Sorry to be slow getting back to this.  I did try using --framerate 0 and it did not appear to have any impact.  I did try another test, I took a picture with the same Pi and same camera every three minutes alternating between raspistill and libcamera-still.  Selected images here: http://knutejohnson.com/cameratest1.  These were taken without the --framerate 0 and the camera program was run for each photo.  I will run these again this evening and tomorrow morning with --framerate 0 and --exposure long and see how it goes.

What would be a good timelapse interval to get the preview frames?  Do I need to enable preview?

davidplowman, 2022-01-21:
Thanks for posting those. It would be interesting to see some of the exif data for some of those images, specifically the exposure time and ISO. It's a bit strange how the difference seems more pronounced as they're getting darker. Is it the case that, when those images are rather dark, that stopping and doing a single "standalone" capture would be brighter, do you think? I'll try and do some experiments myself too.

As regards a good timelapse gap - I've actually just pushed a branch to the repo called "timelapse-fix" which enforces half a dozen preview frames between timelapse captures, however small the requested interval. You could give that a go and see if it makes any difference.

knutegit, 2022-01-21:
>Is it the case that, when those images are rather dark, that stopping and doing a single "standalone" capture would be brighter, do you think? I'll try and do some experiments myself too.

Those images are all single frame captures.  I start the program and capture the first image out of the stream.

I looked at the Exiftool and it doesn't give the exposure or ISO.  How do find that information?

knutegit, 2022-01-21:
For the last test I used a Java program to capture the image and save it as a file.  Turns out when the image data is converted to a Java image not all of the EXIF data is saved.  Tomorrow I'll try using just the programs to save the files and we'll get the complete EXIF.

knutegit, 2022-01-23:
Two new sets of images.  http://knutejohnson.com/cameratest2 are images from the stdout converted to Java images.  These have framerate 0 and exposure long.  Not much different from the ones that didn't have those options in cameratest1.

http://knutejohnson.com/cameratest3 are images taken directly with raspistill and libcamera-still and saved to a file.  Both raspistill and libcamera-still were fresh compiles yesterday.  Mouse over the image for a tooltip with the EXIF data for that image.  raspistill required a 2000ms timeout or the image was severely washed out (not sure what's up with that).  Command lines are on the top of the web page.

I've discovered that when you read a JPEG file and create a Java BufferedImage with that data and subsequently save it as a JPEG file again, that the EXIF data in the image is lost.

I hope this is somewhat helpful.

This is the kernel in my Pi3 after an rpi-update: Linux raspberrypi4B 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armvl.  It appears to be the same kernel that is current.  Does this kernel have the fix for the pink images?

davidplowman, 2022-01-24:
Hi, thanks very much for posting those two sets of images. I think they're very helpful, especially that second set where the EXIF pops up when you hover!

Anyway, broadly speaking I thought the exposures were relatively comparable. Looking at the "cameratest3" set first, I thought perhaps the biggest difference might be in images 1820-raspicam.jpg and 1822-libcam.jpg, though it didn't seem anywhere near as huge as you were seeing in those earlier image sets. Looking at the EXIF data:

1820-raspicam.jpg has an exposure of 1/8s and ISO of 320. We believe raspicam counts a gain of 1 as being about ISO 54, so this makes the gain here about 6x.

1822-libcam.jpg has an exposure of 1/8s and an ISO of 600. Looking at imx219.json, I think the exposure for "long" mode tops out at 120ms, so the EXIF seems to be approximating slightly there. In this case the code takes ISO 100 as a gain of 1, so the gain here is 6x too, and the ISO difference is not anything "real".

There is certainly some variation here in these images, I don't think that's too surprising, but overall I don't think I'm seeing too much wrong. If you wanted the libcamera images to be a bit brighter in those dark conditions, you could increase the maximum gain in the `imx219.json` tuning file. For imx219 you could go up to 10x. If you wanted all the images brighter (specifically the ones that are not being limited by the darkness) then you could dial in some extra EV on the command line (e.g. `--ev 0.5`). Does that help enough?

knutegit, 2022-01-24:
I think with the --framerate 0 and the --exposure long that they are close enough.  I'll try playing with the options you suggested to see if I can get a little better result.

I ran another whole set of images this weekend with libcamera running continuously using the STDOUT option.  The images are a little dark but it was overcast and the light levels were low.  You can see those at http://knutejohnson.com/cameratest4.  It's sunny today, I'll run another set of those this evening and tomorrow morning.  I was hoping with this latest set of images to capture the fade out that I have seen in the past.  I did however run my motion detection program today and captured a bunch of pink images followed by washed out images.  That program uses the same scheme to capture images as the set in cameratest4.

I really appreciate the time you have taken to help me get this running better.  Thanks!

davidplowman, 2022-01-25:
Well, thanks for taking the time to help us figure out most of the problems, it's much appreciated. I guess I'll close this one now, but if you want any more advice on anything, please don't hesitate to post back or create a new issue!

closed 2022-01-25 by davidplowman