raspberrypi/rpicam-apps #224: [BUG] Not able to get libcamera-hello working on RPi Zero W 1.1 @ current Raspberry Pi OS; raspivid works

ohhai, 2022-01-13:
Creating separate issue as requested here: https://github.com/raspberrypi/libcamera-apps/issues/208#issuecomment-1012195643

Hardware:
- Raspberry Pi Zero W 1.1
- HQ camera (IMX477) with mechanical IR modification: https://www.arducam.com/product/arducam-high-quality-ir-cut-camera-for-raspberry-pi-12-3mp-1-2-3-inch-imx477-hq-camera-module-with-6mm-cs-lens-for-pi-4b-3b-2b-3a-pi-zero-and-more/

Software:
https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-11-08/2021-10-30-raspios-bullseye-armhf-full.zip
plus current updates (`apt-get update && apt-get dist-upgrade`)

Output from `utils/camera-bug-report`:
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7863743/bug-report.txt)
Just in case: there are a lot of "Division by zero in kernel" errors in dmesg. Triggered by libcamera-hello inside video-related code.

`raspivid` is able to work fine with different config.txt (more details in the issue referenced above)

My diff of `/boot/config.txt` with the original file from 2021-10-30-raspios-bullseye-armhf-full.zip:
```
 camera_auto_detect=1
+dtoverlay=imx477
+#start_x=1
+#dtoverlay=vc4-fkms-v3d
+#enable_uart=1
+#gpu_mem=128
```
Full file: [config.txt](https://github.com/raspberrypi/libcamera-apps/files/7863752/config.txt)

naushir, 2022-01-14:
Thanks for the detailed report!

The div by zero that you are seeing is almost certainly a bug that has been fixed in https://github.com/raspberrypi/linux/commit/19359ba5442ffba6db180eaaeda7472e7377e368.  Are you able to rebuild your kernel with the latest changes and try again.  Alternatively, you should be able to do a ``rpi-update`` to get a newer kernel with the fixes.

If you do either of these to update the kernel, you will run into a mismatch with the libcamera userland libraries.  To fix this, you can either update the libcamera libraries (instructions found [here](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps)), or you can add a parameter to to dtoverlay in config.txt:

```
dtoverlay=imx477,media-controller=0
```

This incompatibility will be fixed in an imminent upcoming software release to apt.

davidplowman, 2022-01-14:
Hi, a few things to try:

1. If you want to use libcamera on a Pi Zero I would recommend undoing any edits to your /boot/config.txt. They may be causing more memory to be consumed which may hamper its ability to run. In particular, adding more GPU memory will only make things worse.
2. If you're using X Windows and are expecting to see the camera preview in a regular X window, you need to [re-enable Glamor](https://forums.raspberrypi.com/viewtopic.php?t=323547). Note that the rate at which a Pi Zero can get camera images through the X WIndows display stack onto the screen will be poor (as indeed will be anything that involves X Windows).
3. If you're logged in through ssh and want to display the preview using X forwarding there are some additional constraints as the default hardware-accelerated preview implementations don't work with X forwarding. You would have to use the `--qt-preview` option to get a software rendered preview though as noted earlier, a Pi Zero will struggle.  The hardware-accelerated preview windows also don't work very well with VNC, `--qt-preview` would probably be better in this case too.
4. Performance on a Pi Zero will be _way_ better if you don't use X Windows. Installing Raspberry Pi OS Lite would be best, though even quitting or suspending X Windows (Ctrl+Alt+F1 usually) will help massively. Of course, you always have the option to suppress the preview window entirely (the `-n` option).

ohhai, 2022-01-14:
@naushir  @davidplowman  thank you for all the info, I'd unlikely manage to find it all by myself!

I did `rpi-update`, added `,media-controller=0` to `dtoverlay=imx477` and enabled Glamor (just in case, it caused no changes in `/boot/config.txt`)

Current diff with the original `/boot/config.txt` :
```
-#hdmi_force_hotplug=1
+hdmi_force_hotplug=1
-#hdmi_group=1
-#hdmi_mode=1
+hdmi_group=2
+hdmi_mode=42
+dtoverlay=imx477,media-controller=0
+enable_uart=1
+dtoverlay=dwc2
```

Might be important: I connect to RPi with `tigervnc` on my Linux PC and `tigervnc-xorg-extension` on RPi.

What I have now:

1)
a) Black window of libcamera-app when launching just `libcamera-hello`
b) This error from vlc on PC and nothing more when launching `libcamera-vid -t 0 --inline --listen -o tcp://0.0.0.0:6123` on RPi :
```
$ cvlc tcp/h264://192.168.0.5:6123
VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
[000055d84af64f30] dummy interface: using the dummy interface module...
[00007f0530001110] h26x demux error: this doesn't look like a h264 ES stream, continuing anyway
```

2)
Console output of libcamera ends with following. Sensor control error might indicate some difference of my camera module from the original HQ camera, but (again, just in case) - raspivid is able to show image from my camera.
As for last buffers error, it has related part in dmesg (see below)
```
[0:11:36.844093549] [958]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pgCC
[0:11:36.853795437] [958] ERROR IPARPI raspberrypi.cpp:539 Unable to find sensor control 0x009e0903
[0:11:36.854002435] [958] ERROR IPARPI raspberrypi.cpp:351 Sensor control validation failed.
[0:11:36.854132433] [958] ERROR RPI raspberrypi.cpp:1306 IPA configuration failed!
[0:11:36.854289432] [958] ERROR RPI raspberrypi.cpp:757 Failed to configure the IPA: -32
[0:11:37.540281542] [958] ERROR V4L2 v4l2_videodevice.cpp:1158 /dev/video1[14:cap]: Unable to request 4 buffers: Invalid argument
```

3)
dmesg has no more "division by zero" errors. The only clear problem there shows following:

```
WARNING: CPU: 0 PID: 1015 at drivers/media/common/videobuf2/videobuf2-core.c:806 vb2_core_reqbufs+0x180/0x49c [videobuf2_common]
CPU: 0 PID: 1015 Comm: libcamera-hello Tainted: G        WC        5.10.90+ #1512
Backtrace: 
[<c0877908>] (dump_backtrace) from [<c0877c88>] (show_stack+0x20/0x24)
 r7:00000326 r6:00000009 r5:bf35e84c r4:c0a7429c
[<c0877c68>] (show_stack) from [<c087af04>] (dump_stack+0x28/0x30)
[<c087aedc>] (dump_stack) from [<c0023210>] (__warn+0xe8/0x110)
 r5:bf35e84c r4:bf361f78
[<c0023128>] (__warn) from [<c08782a0>] (warn_slowpath_fmt+0x68/0xd8)
 r7:00000009 r6:bf35e84c r5:00000326 r4:bf361f78
[<c087823c>] (warn_slowpath_fmt) from [<bf35e84c>] (vb2_core_reqbufs+0x180/0x49c [videobuf2_common])
 r9:bf45147c r8:bf44b648 r7:c3837e3c r6:c0bb9028 r5:c3bc5a2c r4:c3bc59ec
[<bf35e6cc>] (vb2_core_reqbufs [videobuf2_common]) from [<bf389f84>] (vb2_ioctl_reqbufs+0x80/0xa8 [videobuf2_v4l2])
 r10:c0bb9028 r9:bf45147c r8:c3bc5b38 r7:c4ec2300 r6:c3bc5b48 r5:00000000
 r4:c3837e3c
[<bf389f04>] (vb2_ioctl_reqbufs [videobuf2_v4l2]) from [<bf267690>] (v4l_reqbufs+0x50/0x54 [videodev])
 r7:bf45147c r6:c4ddb360 r5:c4ec2300 r4:c3837e3c
[<bf267640>] (v4l_reqbufs [videodev]) from [<bf267e6c>] (__video_do_ioctl+0x234/0x464 [videodev])
 r7:00000003 r6:bf267640 r5:c0145608 r4:c3bc5b48
[<bf267c38>] (__video_do_ioctl [videodev]) from [<bf2688a0>] (video_usercopy+0x118/0x558 [videodev])
 r10:00000000 r9:c0bb9028 r8:c3837e3c r7:00000000 r6:00000000 r5:c0145608
 r4:c0145608
[<bf268788>] (video_usercopy [videodev]) from [<bf268cfc>] (video_ioctl2+0x1c/0x24 [videodev])
 r10:0000000e r9:a86fe3a4 r8:c0bb9028 r7:00000000 r6:c4ec2300 r5:c4ec2301
 r4:bf268ce0
[<bf268ce0>] (video_ioctl2 [videodev]) from [<bf260148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[<bf2600fc>] (v4l2_ioctl [videodev]) from [<c020c1d8>] (sys_ioctl+0x1ac/0x88c)
 r5:c4ec2301 r4:c0145608
[<c020c02c>] (sys_ioctl) from [<c0008240>] (ret_fast_syscall+0x0/0x28)
```

Might be triggered by this kernel code, if I found correct repo and revision:

```
/* Check that driver has set sane values */
if (WARN_ON(!num_planes))
    return -EINVAL;

for (i = 0; i < num_planes; i++)
    if (WARN_ON(!plane_sizes[i]))
        return -EINVAL;
```

https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/media/common/videobuf2/videobuf2-core.c#L806

4)
free memory space seems sufficient:
```
$ free -m
               total        used        free      shared  buff/cache   available
Mem:             429         182          77           6         169         191
Swap:              0           0           0
```

5)
Full [bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/7872517/bug-report.txt)
Note: `libcamera-hello` did not end by itself for some reason, I had to kill it by `killall libcamera-hello` to get the report.

naushir, 2022-01-15:
```
> [0:11:36.844093549] [958]  INFO RPI raspberrypi.cpp:620 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected mode: 2028x1520-pgCC
> [0:11:36.853795437] [958] ERROR IPARPI raspberrypi.cpp:539 Unable to find sensor control 0x009e0903
> [0:11:36.854002435] [958] ERROR IPARPI raspberrypi.cpp:351 Sensor control validation failed.
> [0:11:36.854132433] [958] ERROR RPI raspberrypi.cpp:1306 IPA configuration failed!
> [0:11:36.854289432] [958] ERROR RPI raspberrypi.cpp:757 Failed to configure the IPA: -32
> [0:11:37.540281542] [958] ERROR V4L2 v4l2_videodevice.cpp:1158 /dev/video1[14:cap]: Unable to request 4 buffers: Invalid argument
```
The above log suggests there is still a mismatch between the kernel and userland drivers.  Could you remove the ``camera_auto_detect=1`` line from you config.txt, and ensure you only have ``dtoverlay=imx477,media-controller=0`` and try again?

ohhai, 2022-01-15:
@naushir  thank you, now it works!
This issue can be closed, I think.

Streaming to `vlc` definitely has problems, there are visual artefacts, big delay and errors from `cvlc`.
Should not be performance issue, tested w/o xserver (after `systemctl isolate multi-user.target`) and checked CPU usage, it's not full.

Tested by this on RPi:
`libcamera-vid -n --framerate 10 --width 1332 --height 990 --denoise off --profile baseline --flush -t 0 --inline --listen -o tcp://0.0.0.0:6123`
And this on PC (ignore hw decode errors, I disabled it because of this: https://bugzilla.redhat.com/show_bug.cgi?id=1950692):
```
$ cvlc tcp/h264://192.168.0.55:6123
VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
[0000555838465a40] dummy interface: using the dummy interface module...
[00007f5f30006320] gl gl: Initialized libplacebo v4.157.0 (API v157)
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib64/dri/r600_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
[00007f5f680cc8b0] vaapi generic error: profile(7) is not supported
[00007f5f30006320] gl gl: Initialized libplacebo v4.157.0 (API v157)
[00007f5f480090f0] avcodec decoder: Using G3DVL VDPAU Driver Shared Library version 1.0 for hardware decoding
[h264 @ 0x7f5f4805ccc0] Failed setup for format vdpau: hwaccel initialisation returned error.
[00007f5f480090f0] avcodec decoder error: existing hardware acceleration cannot be reused
[00007f5f30006320] gl gl: Initialized libplacebo v4.157.0 (API v157)
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib64/dri/r600_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
[00007f5f68160040] vaapi generic error: profile(7) is not supported
[00007f5f30006320] gl gl: Initialized libplacebo v4.157.0 (API v157)
[00007f5f44000c80] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1966 ms)
[00007f5f480090f0] main decoder error: Timestamp conversion failed for 2833320: no reference clock
[00007f5f480090f0] main decoder error: Could not convert timestamp 0 for FFmpeg
[00007f5f44000c80] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 3920 ms)
[00007f5f480090f0] main decoder error: Timestamp conversion failed for 4966663: no reference clock
[00007f5f480090f0] main decoder error: Could not convert timestamp 0 for FFmpeg
[h264 @ 0x7f5f480a5b80] get_buffer() failed
[h264 @ 0x7f5f480a5b80] thread_get_buffer() failed
[h264 @ 0x7f5f480a5b80] decode_slice_header error
[h264 @ 0x7f5f480a5b80] no frame!
[00007f5f44000c80] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (jitter of 6786 ms ignored)
[00007f5f480090f0] main decoder error: Timestamp conversion failed for 10899996: no reference clock
[00007f5f480090f0] main decoder error: Could not get display date for timestamp 0
[00007f5f480090f0] main decoder error: Timestamp conversion failed for 10899996: no reference clock
[00007f5f480090f0] main decoder error: Could not convert timestamp 0 for FFmpeg
```

`ffplay` is much better
using example from here: https://www.raspberrypi.com/documentation/accessories/camera.html

Here are no clear errors in kernel log, only this line repeating 20-30 times per second:
```
[  436.999440] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8d000000
```
(address varies)

Also, `libcamera-vid` prints this for all cases:
```
[0:32:39.106154000] [983]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 0
[0:32:39.112035000] [979]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:32:39.685666000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (6 left)
[0:32:39.776535000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (5 left)
[0:32:39.875450000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (4 left)
[0:32:39.974926000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (3 left)
[0:32:40.074953000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (2 left)
[0:32:40.186030000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (1 left)
[0:32:40.282094000] [979]  INFO RPI raspberrypi.cpp:1636 Dropping frame at the request of the IPA (0 left)
```

davidplowman, 2022-01-15:
Thanks for the update. I don't know what's happened to recent versions of vlc. The same problems occur simply saving an h264 file and trying to replay it, when older versions, and other applications like ffplay, have no trouble. But I'll close this issue, please feel free to post again if anything else comes up.

closed 2022-01-15 by davidplowman