raspberrypi/rpicam-apps #225: Long repeated capture hangs eventually at app.Wait()

dustinkerstein, 2022-01-14:
_This could definitely be more of a `libcamera` issue, so please let me know if I should instead post over there._

I noticed that some of my captures when using my custom `libcamera-app` (combined still and video capture with a in-memory circular buffer that isn't deallocated/re-allocatd on each capture) sometimes just stops capturing. So I modified `libcamera-vid` with a simple while loop test (though I recognize this isn't exactly a supported / useful design):

```
int main(int argc, char *argv[])
{
	try
	{
		unsigned int counter = 0;
		while (true) {
			LibcameraEncoder app;
			VideoOptions *options = app.GetOptions();
			if (options->Parse(argc, argv))
			{
				if (options->verbose)
					options->Print();

				event_loop(app);
			}
			counter++;
			std::cerr << "COUNTER: " << counter << std::endl;
		}
	}
	catch (std::exception const &e)
	{
		std::cerr << "ERROR: *** " << e.what() << " ***" << std::endl;
		return -1;
	}
	return 0;
}
```

And around the 1582nd loop when running the following command: `LIBCAMERA_LOG_LEVELS=0 ./libcamera-vid --frames 1 --framerate 10 --width 4056 --height 3040 --shutter 100000 --awbgains 1,1 --gain 1 --codec yuv420 -o /dev/null` it just stops. The line where it seems to hang is: `LibcameraEncoder::Msg msg = app.Wait()`

Here are logs during the capture: [Logs.zip](https://github.com/raspberrypi/libcamera-apps/files/7871950/Logs.zip)

I am currently on an `libcamera` commit 8ff5a8d548eee7c2cb14b355867debdb29e75419 plus a patch from here - https://github.com/raspberrypi/libcamera-apps/issues/217#issuecomment-1007605812 - I will give the above test another try with the latest `libcamera` code and report back.

```
$ uname -a
Linux raspberrypi 5.10.87-v8+ #1 SMP PREEMPT Sat Dec 18 19:47:37 GMT 2021 aarch64 GNU/Linux
./libcamera-hello --version
libcamera-apps build: fe1f849bfb49-intree-dirty 14-01-2022 (15:56:40)
libcamera build: v0.0.0+3369-8ff5a8d5-dirty (2022-01-07T17:53:56+00:00)
$ vcgencmd version
Nov 18 2021 16:16:49
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
```

dustinkerstein, 2022-01-14:
Quick update with latest `libcamera` code (with no patches) - Test is still running through 2379 loops. If I can't replicate on the latest `libcamera` code I will go ahead and close this issue.

dustinkerstein, 2022-01-14:
Good through 3672 loops. Closing this issue for now. I'll reopen if I run across this issue again.

closed 2022-01-14 by dustinkerstein

davidplowman, 2022-01-14:
Thanks. One thing I;m curious about is whether it keeps making new preview windows (or were you using `-n`)? I don't think I ever put any code in to kill the previews, not really thinking that anyone would ever run it like that. Strictly speaking I suppose that wants sorting out, though it doesn't seem top priority...!

dustinkerstein, 2022-01-24:
Hey David, sorry for the long delay - I've been traveling a bit.

In the test above, I was not using `-n`, but I have since repeated the test on the latest `libcamera` source (with and without `-n`) and I haven't been able to get it to hang. However, I have still seen an occasional hang in my actual app... So I will keep trying to isolate it and will post back here if I do.