raspberrypi/rpicam-apps #226: Libcamera-apps linking fails

AndreGuimRua, 2022-01-19:
Hi, 

I'm trying just the libcamera-apps. I have updated the system with `sudo apt dist-upgrade` and got the last version of the git files, but whenever I try to build the project the following happens:

`[ 13%] Built target images
[ 25%] Built target preview
[ 37%] Built target outputs
[ 48%] Built target encoders
[ 65%] Built target post_processing_stages
Generating version string: 14e32b75ebba-intree 19-01-2022 (10:20:37)
[ 65%] Built target VersionCpp
Scanning dependencies of target libcamera_app
[ 67%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
[ 69%] Linking CXX shared library libcamera_app.so
[ 76%] Built target libcamera_app
[ 79%] Linking CXX executable ../libcamera-hello
[ 81%] Linking CXX executable ../libcamera-jpeg
[ 83%] Linking CXX executable ../libcamera-raw
[ 86%] Linking CXX executable ../libcamera-still
[ 88%] Linking CXX executable ../libcamera-vid
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
/usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
/usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/libcamera-jpeg.dir/build.make:110: libcamera-jpeg] Error 1
make[1]: *** [CMakeFiles/Makefile2:464: apps/CMakeFiles/libcamera-jpeg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
/usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
/usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
collect2: error: ld returned 1 exit status
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
/usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
make[2]: *** [apps/CMakeFiles/libcamera-raw.dir/build.make:112: libcamera-raw] Error 1
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
/usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
make[1]: *** [CMakeFiles/Makefile2:496: apps/CMakeFiles/libcamera-raw.dir/all] Error 2
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/libcamera-vid.dir/build.make:112: libcamera-vid] Error 1
make[1]: *** [CMakeFiles/Makefile2:558: apps/CMakeFiles/libcamera-vid.dir/all] Error 2
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
/usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
/usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
/usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
collect2: error: ld returned 1 exit status
/usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
make[2]: *** [apps/CMakeFiles/libcamera-still.dir/build.make:110: libcamera-still] Error 1
make[1]: *** [CMakeFiles/Makefile2:588: apps/CMakeFiles/libcamera-still.dir/all] Error 2
/usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/libcamera-hello.dir/build.make:110: libcamera-hello] Error 1
make[1]: *** [CMakeFiles/Makefile2:526: apps/CMakeFiles/libcamera-hello.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
`

I used the following for CMake

`cmake .. -DENABLE_DRM=1 -DENABLE_X11=1 -DENABLE_QT=1 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0`

Also the instructions over [https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps](url)

Any help is appreciated

AndreGuimRua, 2022-01-19:
Hi, thanks for the quick answer
I'll rebuild libcamera and update the kernel too. I wasn't doing it because when I did it the libcamera-apps stopped finding any camera (maybe because my kernel was not updated).

I'll let you know asap the results

Best regards and thank you

AndreGuimRua, 2022-01-19:
Hi @davidplowman I've just rebuilded libcamera and updated with sudo rpi-update

When I try to run the older libcamera-hello now it's broken

> pi@raspberrypi:~/6_SW_Architecture $ libcamera-hello
> [0:22:20.095450769] [5994]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3396-6b160d91
> [0:22:20.139596121] [5997]  WARN RPI raspberrypi.cpp:1208 Mismatch between Unicam and CamHelper for embedded data usage!
> [0:22:20.140766581] [5997]  INFO RPI raspberrypi.cpp:1328 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
> [0:22:20.142001540] [5994]  INFO Camera camera.cpp:1028 configuring streams: (0) 1640x1232-YUV420
> [0:22:20.142588409] [5997]  INFO RPI raspberrypi.cpp:744 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
> [0:22:20.144886069] [5994] ERROR Allocator framebuffer_allocator.cpp:98 Stream is not part of /base/soc/i2c0mux/i2c@1/imx219@10 active configuration
> ERROR: *** failed to allocate capture buffers ***

Also the build of the libcamera-apps keeps failing:

> pi@raspberrypi:~/6_SW_Architecture/libcamera-apps/build $ make -j4 
> [ 13%] Built target images
> [ 25%] Built target encoders
> [ 38%] Built target preview
> [ 50%] Built target outputs
> [ 65%] Built target post_processing_stages
> Generating version string: 14e32b75ebba-intree 19-01-2022 (11:09:06)
> [ 65%] Built target VersionCpp
> Scanning dependencies of target libcamera_app
> [ 68%] Building CXX object core/CMakeFiles/libcamera_app.dir/version.cpp.o
> [ 70%] Linking CXX shared library libcamera_app.so
> [ 77%] Built target libcamera_app
> [ 79%] Linking CXX executable ../libcamera-jpeg
> [ 81%] Linking CXX executable ../libcamera-vid
> [ 84%] Linking CXX executable ../libcamera-hello
> [ 86%] Linking CXX executable ../libcamera-raw
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
> /usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
> /usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
> /usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
> /usr/bin/ld: ../preview/libpreview.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::Request::addBuffer(libcamera::Stream const*, libcamera::FrameBuffer*, std::unique_ptr<libcamera::Fence, std::default_delete<libcamera::Fence> >)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Rec709'
> /usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::ColorSpace::toString[abi:cxx11](std::optional<libcamera::ColorSpace> const&)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::UniqueFD::reset(int)'
> collect2: error: ld returned 1 exit status
> make[2]: *** [apps/CMakeFiles/libcamera-hello.dir/build.make:110: libcamera-hello] Error 1
> make[1]: *** [CMakeFiles/Makefile2:526: apps/CMakeFiles/libcamera-hello.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
> /usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
> collect2: error: ld returned 1 exit status
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
> make[2]: *** [apps/CMakeFiles/libcamera-jpeg.dir/build.make:110: libcamera-jpeg] Error 1
> make[1]: *** [CMakeFiles/Makefile2:464: apps/CMakeFiles/libcamera-jpeg.dir/all] Error 2
> collect2: error: ld returned 1 exit status
> make[2]: *** [apps/CMakeFiles/libcamera-vid.dir/build.make:112: libcamera-vid] Error 1
> make[1]: *** [CMakeFiles/Makefile2:558: apps/CMakeFiles/libcamera-vid.dir/all] Error 2
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Jpeg'
[... cut ...]
> /usr/bin/ld: ../encoder/libencoders.so: undefined reference to `libcamera::operator==(libcamera::ColorSpace const&, libcamera::ColorSpace const&)'
> /usr/bin/ld: ../core/libcamera_app.so: undefined reference to `libcamera::ColorSpace::Smpte170m'
> collect2: error: ld returned 1 exit status
> make[2]: *** [apps/CMakeFiles/libcamera-raw.dir/build.make:112: libcamera-raw] Error 1
> make[1]: *** [CMakeFiles/Makefile2:496: apps/CMakeFiles/libcamera-raw.dir/all] Error 2
> make: *** [Makefile:171: all] Error 2
> 

If you have any info that can be helpfull I would really appreciate that

Best regards

davidplowman, 2022-01-19:
Hi again, are you sure you've updated and installed libcamera as well as libcamera-apps? Can you report the output of `libcamera-hello --version`, that should output the top git commits of both repos. Those missing ColorSpace symbols have all been in libcamera since mid-December, but it seems you're linking against something older.

AndreGuimRua, 2022-01-19:
Hi, just to ensure that I was doing correctly I've formatted a new sd cards with the latest Bullseye and repeated the process and everything went fine this time

Thanks for your help @davidplowman : )

closed 2022-01-19 by AndreGuimRua