raspberrypi/rpicam-apps #227: How to achieve consistent capture timing

Schaggo, 2022-01-21:
Hi, 

I am trying to achieve consistent capture timing in libcamera-still so that every frame is taken exactly x microseconds after request.

So far I am using the signal mode to "prime" the cameras to let exposure settle and then send an OS kill-command with SIGURS1 to the thread the cam is running on.  

so far the execution time of the kill-command is 2.2x seconds +- 300ms. 

python pseudo: 

```
subprocess of libcamera-still -s
usleep (2 seconds)
start = time.time()
os.kill(libcamera-still, SIGURS1)
```

Is python maybe too slow?

Do you have a hint, where I might look in order to achieve a near-absolute capture timing?

naushir, 2022-01-21:
> I am trying to achieve consistent capture timing in libcamera-still so that every frame is taken consistently x microseconds after request.

Could you clarify on this further please?  Is your intention to receive a JPEG capture output `x` microseconds after you have sent a request through a signal? And you need `x` to be a fixed non-changing value between runs?

Schaggo, 2022-01-21:
Exactly.

I want to time/ synch an array of flashes to the image capture.

I already started with a custom libcamera app that halt's until an epoch time is met, but if there is be a way to do this in Python, it would save me some work.

kbingham, 2022-01-21:
This is an interesting topic/use-case.

We have plans for handling flash control in libcamera, but we haven't got there yet or got any specific devices with a flash to develop this on.

In libcamera world, we would expect to set a control on the Request (that holds the buffer that will be captured at that point in time) to enable the flash so that it gets synchronized by the pipeline handler internally in libcamera.

Then user space should be able to say "At this frame I want the flash to be on" (where maybe things get more complicated like, the flash needs to be on the frame before or something with a delay).

It will be /very/ difficult to match the flash timing precisely for when the capture occurs from when a request is queued otherwise, as it may have variable queue lengths or depths..

Schaggo, 2022-01-21:
Thank you both for chiming in!

I was able to achieve this before in raspistill - but the other way around. I started the capture of a frame with a slow shutter and then kind of timed the flash, based on min and  max delays. So synch wasn't _really_ necessary. 

I haven't read through the documentation yet, but my rough idea would be:
-  to synch the clocks of the flash trigger and the camera via PTP and then 
-  pass an epoch time (or an abbreviation thereof) to the camera app (and the flash trigger with a delay) and 
-  simply halt capture/flash until the time is met with nanosleep or some other monotonic timer.

Do you think this is doable? 
I am probably thinking too easy here.

naushir, 2022-01-21:
Is your flash LED based? Do you intend on keeping it on for the duration of the exposure, or just strobe it for part of the exposure?

Schaggo, 2022-01-21:
Sadly no. 
Sorry, English isn't my first language... I hope the following makes sense: 
I would like to build a device that captures 8 images with 8 different lighting conditions in order to create ["normal maps from multi-angle lighting"](https://substance3d.adobe.com/documentation/sddoc/multi-angle-to-normal-159451093.html)

The main problem is that I am trying to achieve this for animated objects (faces, hands, hairless animals, whatever fits in the booth)

Schaggo, 2022-01-21:
To be clear: 
The series of rapid images would be captured with a video application. Preferably RAW, bug mjpg should do.

kbingham, 2022-01-21:
libcamera can already let you register an event on the completion of a raw buffer, so you could get a pretty close approximation to events when the frame completes, but it might be racy for timing on the start of the next frame. Perhaps it could be calibrated to identify any delays incurred to some level, but it may not be fully real-time.

It could be used as a mechainsm to synchronise when to change your lighting though. Otherwise I wonder if it would be interesting or useful to bring the startFrame event all the way up to the Camera so applications could know when a frame starts. Again, a bit racy, but it might let you prepare something for the 'next' frame ... and would need plenty of investigation and measurements ... ?

Schaggo, 2022-01-21:
Alright. That seems like a framework within which to investigate. I will write my version of libccamera-still, libcamera-vid and bug you with the results.

Thank you so far! 
Always a pleasure to talk to you guys.

Schaggo, 2022-01-21:
Sorry, I had a quick thought: 

What is providing the base clock for hardware and software in libcamera? Does libcamera adjust itself to the sensor or deliver a start  (...) to the camera, or is it just asking for the next frame that can be delivered by the device?

closed 2022-01-22 by Schaggo

kbingham, 2022-01-22:
The sensor should be configured to the desired frame rate, and that will control the pacing and clocking. It's the responsibility of the application to queue requests with buffers fast enough to be able to capture into a buffer each time the sensor completes through the unicam. If the application is too slow, then a frame would get dropped.

Generally, it would be expected that you would have multiple requests prepared, and when once completes, you manage the data from that as quickly as possible (ideally in a separate thread) and then return the buffers back to be refilled.

If you need specific help designing your application around libcamera, please find us on the libcamera-devel mailinglist or IRC channels (this github really relates directly to the Raspberry Pi applications themselves)

naushir, 2022-01-24:
Trying to achieve precise timing from applications is likely to be very error prone and painful (but it is possible).  Even though you can control framing timing (through framerate control), you also have to account for things that are entirely out of the application scope, e.g. pipeline depth in the kernel/libcamera implementation.

For me the simplest approach would be to rely on a hardware sync pulse generated by the camera itself to trigger an external flash.  The imx477 does have support for such a strobe signal that can be connected externally to a flash device.  That way, you do not have to rely on timing information at all.

Schaggo, 2022-01-24:
Thank you, naushir.
While I agree with your approach as a possibility for a general cam-flash implementation, it wouldn't work for my use case. 

Im using arducams IMX519 and am not sure if they have a synch pin. 

I would later also like to try to synch multiple cameras via ptp (ppl). So the only option I am seeing is to synch the all clocks and frame rates.