raspberrypi/rpicam-apps #228: ERROR: the system appears to be configured for the legacy camera stack

sooyoungmoon, 2022-01-21:
Hi, I'm trying to do video streaming from Raspberry Pi Camera v2 to my laptop PC.

The camera is connected to my raspberry Pi 3 Model B+ where Ubuntu 20.04 (64bits) is running.

I installed & built libcamera and libcamera-apps from source, and then tried to run example app by executing the command

`$ libcamera-hello`

However, my bash shell prints out the error message as follows:

`ERROR: the system appears to be configured for the legacy camera stack`

As you might know, there are two types of SW stack (libcamera and raspicam) to access cameras from raspberry pi boards.

When I execute raspistill or raspivid, it works without errors. So it seems that my raspi system have been configured to use raspicam, not libcamera. 

I struggled to do video streaming using raspivid, but for me it is quite difficult due to technical issues.

So I decided to use libcamera instead.

I executed raspi-config and disabled the legacy camera setting, but same result when I call libcamera-hello app.

Would you please help me to configure my system to use libcamera and execute libcamera-apps?

Thank you and have a wonderful day.

davidplowman, 2022-01-21:
It sounds like you might need to add `dtoverlay=imx219` to your /boot/config.txt file. At least, this is what you would do on Raspberry Pi OS, on Ubuntu I guess the file might be somewhere else?

sooyoungmoon, 2022-01-22:
I have two config files in my system, /boot/config.txt and /boot/firmware/config.txt. I don't know difference between them, but anyway I added the statement `dtoverlay=imx219' to ` /boot/firmware/config.txt and the error disappeared. Thanks!
Instead, another error occurred, though. 
Because the original problem was solved, would close this issue and create another one when it is necessary.
I appreciate for your help. Have a nice day.

closed 2022-01-22 by sooyoungmoon

Commander-bao, 2022-08-01:
I failed again and again... Is it because I use the 64bit Raspberry Pi os?

davidplowman, 2022-08-01:
> I failed again and again... Is it because I use the 64bit Raspberry Pi os?

Hi, it's probably best if you create a new issue that describes the problem you're having, what hardware and software you're using, and gives all details about the error you are encountering. Thanks.