raspberrypi/rpicam-apps #231: libcamera-apps weird segmentation fault error

lanewala2019, 2022-01-25:
I'm using a NoIR camera with a RPi Zero 2 W, all recently purchased (within the last month). I was getting some "fd not found";"authentication failed" errors but it seems like after turning on Glamor in raspi-config, those errors have disappeared.

However, I am noticing a consistent pattern of segmentation faults when using libcamera-jpeg.  Here are some results I am seeing, and would be grateful for someone helping me understand...

`libcamera-hello --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json`
works perfectly fine

`libcamera-vid -t 10000 -o test.h264 --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json`
works perfectly fine

`libcamera-jpeg -o test1.jpg -t 2000 --shutter 20000 --gain 1.5 --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json`
fails with segmentation fault with or without the shutter/gain options.. Using the '-n' option results in no segmentation faults.

My /boot/config.txt has the following:

`dtoverlay=vc4-kms-v3d`
`dtoverlay=imx219`
`max_framebuffers=2`

I am fine with using the "-n" option if that's what it takes, but am curious why capturing a jpeg causes that issue whereas the h264 does not. It might be a memory fault, or an issue with the SD card (I'm doubtful about the latter since it's a new card that I just initialized a couple of hours ago). Interestingly, the reason for using a new SD card is that my previous one used probably for a month suddenly gave out and I am unable to reinitialize it on any computer.  In order to minimize issues, I tend to power off (using shutdown commands) every other day or so; not sure if doing so is actually damaging the card although I kinda doubt it...

naushir, 2022-01-26:
I've just tried the following command on my system which has up-to-date libcamera and libcamera-apps libraries, and everything seems to work as expected:
```
libcamera-jpeg -o test1.jpg -t 2000 --shutter 20000 --gain 1.5 --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json
```

Can you confirm if you have rebuilt any libraries from source, or did you rebuild at any point?  Could you also provide the output of ``libcamera-jpeg --version`` please?  Finally, does ``libcamera-still`` work with the same arguments?

lanewala2019, 2022-01-29:
Hi @naushir, I used whatever was in the Bullseye build, followed by whatever update/upgrade was available.
`
pi@raspberrypi:~/StreamCamera $ libcamera-jpeg --version
libcamera-apps build: b9a692385d32 07-01-2022 (16:10:12)
libcamera build: v0.0.0+3384-44d59841
pi@raspberrypi:~/StreamCamera $ 

`

Using libcamera-still produces the following (with no image file output, even though the preview shows):
`
pi@raspberrypi:~/StreamCamera $ libcamera-still -o test1.jpg -t 2000 --shutter 20000 --gain 1.5 --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json
[68:12:25.747140978] [7091]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3384-44d59841
[68:12:25.895381109] [7092]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!
[68:12:25.896890228] [7092]  INFO RPI raspberrypi.cpp:1313 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[68:12:25.901645449] [7091]  INFO Camera camera.cpp:1028 configuring streams: (0) 1640x1232-YUV420
[68:12:25.902371961] [7092]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[68:12:25.985234264] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[68:12:25.986667913] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[68:12:25.986804112] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[68:12:25.986863747] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[68:12:25.986921091] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[68:12:25.986981196] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[68:12:25.987066612] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[68:12:25.987123331] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AnalogueGain = 1.500000
[68:12:25.987182602] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureTime = 20000
[68:12:25.989840734] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[68:12:25.990148131] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[68:12:25.990278392] [7097]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[68:12:25.991429176] [7092]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[68:12:25.991521937] [7092]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[68:12:25.991573447] [7092]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[68:12:29.027213853] [7091]  INFO Camera camera.cpp:1028 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SRGGB10_CSI2P
[68:12:29.028249012] [7092]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
Segmentation fault
pi@raspberrypi:~/StreamCamera $ 

`

Running libcamera-still with '-n' added works fine:
`
pi@raspberrypi:~/StreamCamera $ libcamera-still -o test1.jpg -t 2000 -n --shutter 20000 --gain 1.5 --tuning-file /usr/share/libcamera/ipa/raspberrypi/imx219_noir.json
[68:14:16.999152172] [7102]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3384-44d59841
[68:14:17.092341702] [7103]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!
[68:14:17.093646914] [7103]  INFO RPI raspberrypi.cpp:1313 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[68:14:17.097092442] [7102]  INFO Camera camera.cpp:1028 configuring streams: (0) 1640x1232-YUV420
[68:14:17.097843902] [7103]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[68:14:17.136321332] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[68:14:17.136523519] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[68:14:17.136591436] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[68:14:17.136653051] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[68:14:17.136712062] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[68:14:17.136773989] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[68:14:17.136836437] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[68:14:17.136893052] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AnalogueGain = 1.500000
[68:14:17.136953416] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureTime = 20000
[68:14:17.137009614] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[68:14:17.137075760] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[68:14:17.137132792] [7108]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[68:14:17.138086856] [7103]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[68:14:17.138186440] [7103]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[68:14:17.138237273] [7103]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[... cut ...]
[68:14:19.393443856] [7112]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AnalogueGain = 1.500000
[68:14:19.393505627] [7112]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureTime = 20000
[68:14:19.393564116] [7112]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 100, 1000000000 ]
[68:14:19.393635887] [7112]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[68:14:19.393692919] [7112]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 2
Still capture image received
pi@raspberrypi:~/StreamCamera $ 

`

naushir, 2022-02-01:
@lanewala2019 our Bullsye update has now been published.  Would you be able to try and update your libraries with the following commands and try again?
```
sudo apt update
sudo apt upgarde
sudo ldconfig
```
I suspect you may be hitting a mismatch between libcamera/libcamera-apps libraries that were included in the distribution vs you custom builds.  Hopefully and update should clear this out.

naushir, 2022-02-09:
If there is nothing else needed for this issue, I will close it down shortly.

closed 2022-02-09 by naushir