raspberrypi/rpicam-apps #232: Storing frames in a vector and saving them records the same image

AndreGuimRua, 2022-01-26:
Hi,

I was trying to getting confortable modifying the libcamera apps so I wanted to save X number of frame, storing each one on a vector and saving them after. For that purpose I've modified the libcamera-jpeg app as follow:

New class and vector to store data:
```
class ImageData {
public:
	std::vector<libcamera::Span<uint8_t>> mem;
	StreamInfo info;
	libcamera::ControlList metadata;
	std::string filename;
	std::string cam_name;
	
	ImageData(std::vector<libcamera::Span<uint8_t>> m, StreamInfo i, libcamera::ControlList met, std::string fil, std::string cam)
	{
		mem = m;
		info = i;
		metadata = met;
		filename = fil;
		cam_name = cam;
	}
};

std::vector<ImageData> imageBuffer;
```

Modification to event_loop():
```
else if (app.StillStream())
{				
	Stream *stream = app.StillStream();
	StreamInfo info = app.GetStreamInfo(stream);
	frameinfo = app.GetStreamInfo(stream);
	CompletedRequestPtr &payload = std::get<CompletedRequestPtr>(msg.payload);
	const std::vector<libcamera::Span<uint8_t>> mem = app.Mmap(payload->buffers[stream]);
				
	//Store in buffer
	ImageData d = ImageData(mem, info, payload->metadata, std::to_string(a)+options->output, app.CameraId());
	imageBuffer.push_back(d);

	a++;

	if(a>9)
	{
		app.StopCamera();
		return;
	}
}
```

After 9 frames, return to main() and:

```
int aj = 0;
for(std::size_t i = 0; i < imageBuffer.size(); ++i) 
{
	jpeg_save(imageBuffer[i].mem, imageBuffer[i].info, imageBuffer[i].metadata, "OUT_"+std::to_string(i)+".jpg", imageBuffer[i].cam_name, options);
}
```

The issue is that all the images stored in the vector are the same, even if the metadata and info (exposure for example) are different.
Any help would be appreciated, maybe I'm forgetting cleaning a buffer or something.

If I run `jpeg_save()` in event loop, all the images are saved correctly

Best regards

davidplowman, 2022-01-26:
Hi, I think the problem is that libcamera uses a limited number of buffers, and they get re-filled from the camera over and over. When you copy the span, that's basically just a pointer to the start of the buffer and a length, you're not actually copying the images. Does that make sense?

AndreGuimRua, 2022-01-26:
Hi @davidplowman it make sense to me, I tought that `std::vector<libcamera::Span<uint8_t>> const &mem` was the buffer and not the pointer, I'm going to try and make a copy of the buffer memory and see if that works. I'll let you kow, thanks for the help : )

naushir, 2022-01-26:
It sounds like the circular buffer mechanism provided in libcamera-apps might be very close to what you need.  See the ``--circular`` command line option and how it is used in ``libcamera-vid``.

AndreGuimRua, 2022-01-26:
@davidplowman @naushir I found my desired solution copying the memory using `mem[0].data()` as the buffer start.
I've just added a `uint8_t buffer[imageWidth*imageHeight*3/2];`to my `ImageData` class and made the copy on the `event_loop` using `memcpy(buffer, mem[0].data(), imageWidth*imageHeight*3/2);`
I've also had to modify the `jpeg_save` method so it works with this new buffer.

This way each frame on the buffer vector is unique. Thank both for your help : )

closed 2022-01-26 by AndreGuimRua