raspberrypi/rpicam-apps #233: Starting EGL preview window from Raspberry Pi OS Bulleye lite

diruuu, 2022-01-27:
Need some help.
I tried to use `libcamera-apps` in Raspberry Pi OS lite, it works on DRM mode. But after I installed X server and a window manager, I can't open the preview window from xterm. It says "Preview window unavailable".

What is the minimal requirements to be able to open the preview window on x server?

davidplowman, 2022-01-27:
Hi, I'm guessing that Raspberry Pi OS Lite has "libcamera-apps-lite" installed via apt. Can you install "libcamera-apps" instead?

diruuu, 2022-01-27:
> Hi, I'm guessing that Raspberry Pi OS Lite has "libcamera-apps-lite" installed via apt. Can you install "libcamera-apps" instead?

Thank you! this solved the issue.

closed 2022-01-27 by diruuu