raspberrypi/rpicam-apps #236: RPI stream error: Failed to queue buffer for ISP Input

Sharsie, 2022-01-29:
I'm having an issue with libcamera-vid after it runs for a while. I tried several output targets as I started with udp/tcp streaming, saving to nfs and finally tried saving the files locally to eliminate any issues with networking.

I tried multiple resolutions, ended up running it at half of the camera, but it made no difference (camera is Waveshare RPi IR-CUT Camera (B)). It's running without a timeout and at some point (usually within 1-2 minutes), I receive an error `rpi_stream.cpp:242 Failed to queue buffer for ISP Input` at which point it just stops outputting.

However, the libcamera-vid command keeps running and I have to kill it manually. I was initially blaming the memory, but it never peaked over a 100mb threshold of 400mb available, so I kinda ruled that out and I'm out of ideas. I can't really find any relevant information, so any suggestions on where to start or how to help debugging this would be very appreciated.

I also tried running the command with `LIBCAMERA_LOG_LEVELS=0`, however it started producing about 4-5 times smaller files  that were all about 2.5MB (10s segments at 1292x972) and could not be opened, however I never received the error. So I am unsure of what happened there.

**Bug report**

[camera-bug-report](https://gist.github.com/Sharsie/633c141681ec792de98a7d3af4c296af)

**Additional context**

I am running a clean install of Raspberry Pi OS Lite that only includes ssh access and wifi connection

naushir, 2022-01-31:
```
** uname -a **
Linux kingston21348751933 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux
```
This seems to be a custom kernel you are running.  Can you switch to the Raspberry Pi downstream kernel and try this again?

naushir, 2022-01-31:
I can reproduce this on the RPi kernel as well.  Will investigate to see what could be the cause.

Sharsie, 2022-01-31:
Thank you, I am unaware of having a custom kernel though, I tried motioneyeOS previously, but it would be very silly of me if I flashed wrong file instead of RPi OS (though I have not ruled that out, will investigate once I am back home).
As far as I know, I flashed RPi OS lite that I downloaded on 30th of October 2021, updated with apt during this weekend and that's pretty much it.
 I'll start by reflashing a clean version of latest rpi os to make sure i'm not running anything weird. I'll be happy to help with any debugging if needed from tomorrow onwards.
Thanks a lot for looking into it.

naushir, 2022-02-01:
We have identified the cause of this problem, and a fix has been submitted for review/merge [here](https://patchwork.libcamera.org/patch/15316/).  Hopefully this should be merged very soon.

Sharsie, 2022-02-01:
That's great news, thank you very much!

naushir, 2022-02-02:
The fixes for this issue have now been merged.  You can get the latest libraries by doing:
```
sudo apt update
sudo apt upgrade
```
I'll close this issue down now, please re-open if you still have problems.

closed 2022-02-02 by naushir