raspberrypi/rpicam-apps #237: [BUG] camera-bug-report croaks on "-c" value: --sout

jlpoolen, 2022-01-29:
**Describe the bug**

I am trying to create an RTSP server on a Pi Zero 2 W.  I followed the directionst at:
https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-libcamera-apps
which states to run a server, use this command line:

    libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264

Instead, I used this modified version which includes the IP address of the Pi server running this command:

    libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h264

The server runs for about 3 minutes and then crashes.  I will be filling a separate bug about, though in preparation for that bug, I attempted to create a bug report using your tool and came across a bug with camera-bug-report.  This bug concerns the ability to generate a bug report using "camera-bug-report " with the option to execute the command using "-c".  Here is a command line I attempted:

    pi@raspberrypi:~ $ camera-bug-report -c "libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264"
    ERROR: *** unrecognised option '--sout' ***

    Bug report generated to bug-report.txt
    Please upload this file when you create a new bug report at:
    https://github.com/raspberrypi/libcamera-apps/issues/
    pi@raspberrypi:~ $

It appears there is a problem with the parsing of the value provided to the "-c" option.

Attached is the report generated bug-report_jlpoole_202201291042.zip
[bug-report_jlpoole_202201291042.zip](https://github.com/raspberrypi/libcamera-apps/files/7964369/bug-report_jlpoole_202201291042.zip)

**Bug report**
N/A

The bug report tool uses the following syntax:

```
camera-bug-report -t <timeout in seconds> -o <output file> -c "<libcamera-app command>"
```

For example,

```
 camera-bug-report -c "libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264"
```
fails with:
 ERROR: *** unrecognised option '--sout' ***

You can also run without a timeout:

```
 See above.
```

naushir, 2022-02-01:
This is a bit of a tricky one.  I think if you use the following syntax it should work:
```
`--sout #rtp{sdp=rtsp://:8554/stream1}'
```
(i.e. single quotes across the whole argument).  I am not sure if there is any easier way to fix this. 

Even with this workaround, the camera-bug-report tool did complain about the invalid utf-8 characters from the output of cvlc.  This has been fixed in https://github.com/raspberrypi/libcamera-apps/pull/239.

jlpoolen, 2022-02-02:
You have a back tic in the indented sample above as the first character, yet you reference "single quote" so I assumed the back tic, aka "grave" ASCII decimal 96, was an error and placed the singled quote, ASCII decimal 39,  as you suggested (see _infra_).  

Here's what happened when I tried your suggested command line:

    @raspberrypi:~ $ libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin '--sout #rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h26
    VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
    vlc: unknown option or missing mandatory argument `--sout #rtp{sdp=rtsp://192.168.1.12:8554/stream1}'
    Try `vlc --help' for more information.
    [35:29:31.312867184] [262877]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3384-44d59841
    [35:29:31.386097858] [262885]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
    [35:29:31.386355818] [262885]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
    [35:29:31.386499459] [262885] ERROR CameraSensor camera_sensor.cpp:535 'imx477 10-001a': Camera sensor does not support test pattern modes.
    [35:29:31.472652806] [262885]  INFO RPI raspberrypi.cpp:1313 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
    [35:29:31.474593051] [262877]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420
    [35:29:31.475304016] [262885]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
    [35:29:31.685161812] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
    [35:29:31.685480863] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
    [35:29:31.685565756] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
    [35:29:31.685631796] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
    [35:29:31.685696689] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
    [35:29:31.685760072] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
    [35:29:31.740975716] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
    [35:29:31.741338203] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
    [35:29:31.741424398] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
    [35:29:31.741486427] [262892]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 1
    [35:29:31.743135225] [262885]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [35:29:31.743300479] [262885]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [35:29:31.743454745] [262885]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [35:29:31.743556512] [262885]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [35:29:31.743654790] [262885]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [35:29:31.743750412] [262885]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    pi@raspberrypi:~ $ date
    Tue  1 Feb 19:38:46 PST 2022
    pi@raspberrypi:~ $ 

I realize the above stopped due to something else dealing with buffer allocations, but the operative line that is disconcerting is this one 

     vlc: **unknown option or missing mandatory argument** `--sout #rtp{sdp=rtsp://192.168.1.12:8554/stream1}'

(Note: I will confirm that what I executed in a shell had the character between "argument" and "--sout" as a single quote, ASCII 39, not a back tick, ASCII 96, yet when I pasted my output into this GitHub window, something trying to be clever in Git Hub is replacing the ASCII 39 with ASCII 96.  The is unfortunate.  I confirmed the character was switched by GitHub's JavaScript because I then copied what I pasted here in the editing box of GitHub back into NotePad++ and the characters pasted into NotePad++ is ASCII 96.)   This GitHub smart-quote "feature" could be very troublesome when trying to share command lines with single quotation marks (ASCII 39). )

Did your attempt also cause the "unknown option" message from vlc?

jlpoolen, 2022-02-02:
You have a back tic in the indented sample above as the first character, yet you reference "single quote" so I assumed the back tic, aka "grave" ASCII decimal 96, was an error and placed the singled quote, ASCII decimal 39, as you suggested (see _infra_).

Here's what happened when I tried your suggested command line:

    pi@raspberrypi:~ $ date
    Tue  1 Feb 20:26:44 PST 2022
    pi@raspberrypi:~ $ camera-bug-report -c "libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin '--sout #rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h264"
    [36:18:11.525691674] [268909]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3384-44d59841
    [36:18:11.582444098] [268914]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
    [36:18:11.582533270] [268914]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
    [36:18:11.582588273] [268914] ERROR CameraSensor camera_sensor.cpp:535 'imx477 10-001a': Camera sensor does not support test pattern modes.
    [36:18:11.652502535] [268914]  INFO RPI raspberrypi.cpp:1313 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
    [36:18:11.654433008] [268909]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420
    [36:18:11.655362853] [268914]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
    [36:18:11.770562016] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
    [36:18:11.770777028] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
    [36:18:11.770831146] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
    [36:18:11.770886201] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
    [36:18:11.770937975] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
    [36:18:11.770992509] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
    [36:18:11.771104442] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
    [36:18:11.771152205] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
    [36:18:11.771208979] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
    [36:18:11.771250440] [268921]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 1
    [36:18:11.772796777] [268914]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [36:18:11.772879907] [268914]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [36:18:11.772916003] [268914]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [36:18:11.772948609] [268914]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [36:18:11.772980538] [268914]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    [36:18:11.773012258] [268914]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
    Traceback (most recent call last):
      File "/usr/bin/camera-bug-report", line 140, in <module>
        cmd_out = run_prog(args.c, args.t)
      File "/usr/bin/camera-bug-report", line 57, in run_prog
        line = p.stdout.readline()
      File "/usr/lib/python3.9/codecs.py", line 322, in decode
        (result, consumed) = self._buffer_decode(data, self.errors, final)
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0xac in position 8: invalid start byte
    pi@raspberrypi:~ $ 

So the movement of the single quote to be placed before --sout appears to have worked. 

(Note: I will confirm that what I executed in a shell had the character between "argument" and "--sout" as a single quote, ASCII 39, not a back tick, ASCII 96, yet when I pasted my output into this GitHub window, something trying to be clever in Git Hub is replacing the ASCII 39 with ASCII 96. The is unfortunate. I confirmed the character was switched by GitHub's JavaScript because I then copied what I pasted here in the editing box of GitHub back into NotePad++ and the characters pasted into NotePad++ is ASCII 96.) This GitHub smart-quote "feature" could be very troublesome when trying to share command lines with single quotation marks (ASCII 39). )

I'll await the patches to become live and downloadable (I'm not up for compiling this project on a Zero).

 Thank you.

naushir, 2022-02-02:
> You have a back tic in the indented sample above as the first character, yet you reference "single quote" so I assumed the back tic, aka "grave" ASCII decimal 96, was an error and placed the singled quote, ASCII decimal 39, as you suggested (see _infra_).

Yes, sorry that was a typo on my part.

> 
> Here's what happened when I tried your suggested command line:
> 
> ```
>   File "/usr/bin/camera-bug-report", line 140, in <module>
>     cmd_out = run_prog(args.c, args.t)
>   File "/usr/bin/camera-bug-report", line 57, in run_prog
>     line = p.stdout.readline()
>   File "/usr/lib/python3.9/codecs.py", line 322, in decode
>     (result, consumed) = self._buffer_decode(data, self.errors, final)
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xac in position 8: invalid start byte
> pi@raspberrypi:~ $ 
> ```

This is the issue I got as well, and is fixed by https://github.com/raspberrypi/libcamera-apps/pull/239.

> I'll await the patches to become live and downloadable (I'm not up for compiling this project on a Zero).

I think this will happen in the next day or so!

naushir, 2022-02-02:
> This is the issue I got as well, and is fixed by #239.
> 
> > I'll await the patches to become live and downloadable (I'm not up for compiling this project on a Zero).
> 
> I think this will happen in the next day or so!

The fix is now available:
```
sudo apt update
sudo apt upgrade
```

naushir, 2022-02-07:
@jlpoolen if you are happy with the resolution of this issue, feel free to close this.

jlpoolen, 2022-02-08:
Here is the command as I understand @naushir suggested I modify (moving the single quote before the "--sout"):

      camera-bug-report -c "libcamera-vid -t 0 --inline -n -o - | cvlc stream:///dev/stdin '--sout #rtp{sdp=rtsp://:8554/stream1}' :demux=h264"

When I run this, binary data starts flowing to the console after line 1204.  Attached is the bug report up to the point where binary data starts flooding the console, i.e. line 1204.
[partial_debug_through_1204.txt](https://github.com/raspberrypi/libcamera-apps/files/8020383/partial_debug_through_1204.txt)

I do want to be able to use camera-bug-report for various command lines, so I'd like to have established how to hand single quotes within a double-quoted command line.  I'm going to be trying some other test in the next hour using gstreamer so I'll see how that goes.  I've been finding I'm running something, then it crashes minutes later, so this bug capture tool will prove very helpful.  I'll update further if I encounter other line commands where I am having problems wrapping the command with the **camera-bug-report -c " [command] "** syntax.

jlpoolen, 2022-02-08:
So, using the single command for cvlc, there is no problem with the following two runs which I terminated only to demonstrate that the runs did not have errors:

    pi@raspberrypi:/tmp date; cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264;date
    Mon  7 Feb 19:17:13 PST 2022
    VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
    ^CMon  7 Feb 19:17:25 PST 2022
    pi@raspberrypi:/tmp $ date; cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://:8554/stream1}' :demux=h264;date
    Mon  7 Feb 19:17:30 PST 2022
    VLC media player 3.0.16 Vetinari (revision 3.0.13-8-g41878ff4f2)
    ^CMon  7 Feb 19:19:16 PST 2022
    pi@raspberrypi:/tmp $

And, if I run the command of libcamera-vid... [first part of the pipe] plus the clvc comand [2nd part of the pipe], the server works (I can view the streams on multiple clients!).

I did have some problem getting this command line to work, at first:
 
     date;libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h264;date

I thought perhaps having a "date;" preceding the libcamera-vid was somehow affecting the pipe, but I could not repeat the error so I moved on.  I do, if you are interested, have the dmesg output attached here and I direct your attention to the last 50 lines or so which seem to have some sort of kernel back trace.

[dmesg_through_Mon_Feb_7_192812.txt](https://github.com/raspberrypi/libcamera-apps/files/8020556/dmesg_through_Mon_Feb_7_192812.txt)

So, at this point, I need a syntax that allows me to wrap the libcamera-vid piped into clvc with the camera-bug-report.  The one you had suggested caused the streamer or libcamera-vid to output to the console.

naushir, 2022-02-08:
Unfortunately this is not at all straightforward with the python subprocess module that I am using in the camera-bug-report tool.  I will have a quick play, but I doubt this will be possible with my existing knowledge.

jlpoolen, 2022-02-08:
How about having a procedure, or command line flag, where I call the camera-bug-report tool on its own to generate a snapshot of things you would want to know about.  Then I'd submit the generated file along with the command line that generates the problem?  In other words, not try to wrap the problem command line.  And supplement with some documentation on what to run with a command line or concurrently.  I regret I am not familiar with debugging processes, I always get lost when I try them, especially when each system, e.g. java, Perl, has their own variety of tools.

naushir, 2022-02-08:
> How about having a procedure, or command line flag, where I call the camera-bug-report tool on its own to generate a snapshot of things you would want to know about. Then I'd submit the generated [...]

This can already be done, simply remove the ``-c`` argument from the command line, and it will gather all information about the system.  The only think that will not be available is the command output.

jlpoolen, 2022-02-08:
Thank you, I lost sight of that fact, mainly because I came to learn of
this while trying to wrap a problem command line.

naushir, 2022-02-09:
I'll close this one done now, if there are any further issue, please feel free to reopen or create a new report.

closed 2022-02-09 by naushir