raspberrypi/rpicam-apps #240: [BUG] UDP Network Stream crashed after 14 hours

lazarusCantini, 2022-01-31:
**Describe the bug**
ERROR RPISTREAM rpi_stream.cpp:242 
I have tried a several Times to hold a stream for  continues surveilance via the libcamera app on a headless Raspberry without Preview Window. No try has worked for more than one day. I Shortened the Verbose Bug Report.

**Bug report**
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 28-01-2022 (16:58:03)
Command: /usr/local/bin/camera-bug-report -o bug_report_long_stream_4.txt -c libcamera-vid -t 0 -v --width 1080 --height 720 -n --codec mjpeg -o udp://192.168.178.2:8672

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03111
Serial		: 1000000048be2eba
Model		: Raspberry Pi 4 Model B Rev 1.1

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=a8cc43ed-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

** cat /boot/config.txt **
[... 1080 lines cut ...]
NetOutput: output buffer 0xb17149a8 size 64302
Viewfinder frame 85904
Viewfinder frame 85905
Viewfinder frame 85906
NetOutput: output buffer 0xaf914860 size 64298
NetOutput: output buffer 0xb0514a60 size 64419
NetOutput: output buffer 0xafb14660 size 64287
Viewfinder frame 85907
Viewfinder frame 85908
NetOutput: output buffer 0xb17149a8 size 64443
NetOutput: output buffer 0xaf914860 size 64404
Viewfinder frame 85909
NetOutput: output buffer 0xb0514a60 size 64418
Viewfinder frame 85910
NetOutput: output buffer 0xb17149a8 size 64182
Viewfinder frame 85911
Viewfinder frame 85912
NetOutput: output buffer 0xb17149a8 size 64375
Viewfinder frame 85913
NetOutput: output buffer 0xb0514a60 size 64275
NetOutput: output buffer 0xb1716360 size 64173
Viewfinder frame 85914
[8:32:07.920215288] [4602] ERROR RPISTREAM rpi_stream.cpp:242 
NetOutput: output buffer 0xb0514a60 size 64422
Error: ***** INTERRUPT *****

davidplowman, 2022-01-31:
Hi, thanks for reporting this. Actually I think it looks the same as some of the other recent reports, and we're investigating right now.

naushir, 2022-02-01:
We have identified the cause of this problem, and a fix has been submitted for review/merge [here](https://patchwork.libcamera.org/patch/15316/).  Hopefully this should be merged very soon.

naushir, 2022-02-02:
The fixes for this issue have now been merged.  You can get the latest libraries by doing:
```
sudo apt update
sudo apt upgrade
```
I'll close this issue down now, please re-open if you still have problems.

closed 2022-02-02 by naushir