raspberrypi/rpicam-apps #241: Exposure doesn't adjust when output to STDOUT for extended periods

knutegit, 2022-01-31:
I wrote a Java program that collects images every minute from libcamera-still streaming the image data on STDOUT.  I started it last night at about 2100 and ran it until almost 1000 this morning.  During the period from 2100 to 0957 about 148000 images had been streamed from libcamera-still and I captured one a minute.  Then I stopped and restarted the program and captured one more image.  All of the images from 2100 until 0957 have an exposure of 1/1127.  After restarting the program the next captured image had an exposure of 1/192.  This may partly explain why I was finding libcamera-still images to be darker than raspistill images.  You can see the images here:  http://knutejohnson.com/cameratest6.  The libcamera-still command line: libcamera-still --framerate 0 --exposure long --thumb none --timeout 0 --timelapse 1 --width 1920 --height 1080 --quality 80 --output -

What I was actually looking for was different however.  I have a motion detection program that collects images in the same manner from libcamera-still.  In the middle of the night the images were washing out to all white.  Unfortunately that program doesn't capture images unless it finds a difference from the image before it.  I'm guessing the change was slow and therefore nothing was captured.  I didn't find a white image last night but I did find the exposure appearing to be stuck.

Raspberypi 3B  Linux camerapi 5.10.92-v7+ #1517 SMP Thu Jan 27 13:52:04 GMT 2022 armv7l
libcamera, libepoxy and libcamera-apps recompiled last week.

davidplowman, 2022-02-01:
Hi, thanks for re-posting this. It finally dawned on me what's going on, and why other sensors weren't showing this problem.

There's a little complication with the v2 sensor in that it returns "embedded data" (telling us the exposure and gain for each frame), but we found it to be unreliable on the first frame after the camera starts. Hence [this line](https://git.linuxtv.org/libcamera.git/tree/src/ipa/raspberrypi/cam_helper_imx219.cpp#n84) which causes the AGC/AEC not to run on the very first frame. Unfortunately, because the timelapse interval is so short, it never gets beyond the first frame so the AGC/AEC is, as you've discovered, stuck.

There are a couple of fixes. The first is that, ironically, we decided not to use the embedded data for this sensor so replacing that line by `return ENABLE_EMBEDDED_DATA;` would make it work. We should make that change, however, I'm inclined to do fix 2 anyway. You can try fix 2 by checking out out the "timelapse-fix" branch from this repository. This forces a few frames of preview between the captures regardless how short the interval is, and means generally that the captures are always likely to be better exposed, whatever the properties of the sensor.

So anyway, please give that branch a try, and I'll push that change onto our main branch too.

knutegit, 2022-02-02:
I installed the timelapse fix and ran my motion detection program all day yesterday and last night.  No change in the exposure on restarts.  And no washed out images either.  I'll watch it closely for a couple more days but I think you've got the fix for that issue.  I may try taking the --framerate 0 and --exposure long out to see if that makes any difference.

knutegit, 2022-02-03:
Another 24 hours and no problems at all.  I think we can close this one.

There was an update yesterday to libcamera and libcamera-apps.  I'm assuming the timelapse fix is not in those but everything else is?

davidplowman, 2022-02-03:
I think that's right, the timelapse fix will be in the next update (to happen soon-ish). I'll go ahead and close this issue - thanks for all your help in trying things out!

closed 2022-02-03 by davidplowman