raspberrypi/rpicam-apps #243: Post processing to combine the camera stream with a PNG image.

diruuu, 2022-02-03:
Not sure where to ask this. 

I want to add a PNG photo frame [like this](https://cdn.kibrispdr.org/data/frame-foto-polaroid-png-4.png) to the camera feed using post-processing, so basically stacking up the photo frame on top of the camera feed. I think using OpenCV would be overkill for this, so I supposed something like `CImg` or `imagemagick` could work for the post processing stage. 

My problem is I don't know how to feed the image buffer to either `CImg` or `imageMagic` and modify the camera stream by combining it to the PNG photo frame. Most of examples on the internet don't work with an image stream like what libcamera use for post processing.

Can someone guide me on this? Been stuck for hours.

davidplowman, 2022-02-03:
Hi, can I start by asking a little more about what you want to implement?

* I understand you want to superimpose an image such as a photoframe on top of the camera feed. Are you wanting to use libcamera-apps to do this?
* Do you want the frame to be added across all the applications, in preview, video and still capture mode?

You could certainly do something like this with a libcamera-apps post-processing stage. I guess you'd load the PNG when it starts and then you could simply draw it on top of the camera image, it's a simply laid out pixel buffer after all. Some things you'd have to watch out for:

* Pixel formats. libcamera-apps quite often work with YUV420 buffers. You might need to support that format. Perhaps you'd convert your image to YUV420 as soon as you've loaded it, then you'd have both an RGB and YUV420 version available.
* Resizing. Presumably you will have to resize the frame to match the images delivered by the camera. You could create resized versions once the streams have been configured and you know what size you need, obviously you'd want to avoid repeating the resize operation for every camera image!

diruuu, 2022-02-03:
Hi, thanks.

Yes, the plan is to use libcamera-apps's post processing to show the photoframe on preview and still capture mode. 

I understand, I should make a custom post-processing stage and use `bool Process(CompletedRequest &completed_request)` member function to modify the buffer, but I have a little experience with C++ and not sure if there is a library out there that can combine/stack the RGB image on top of the YUV420 buffer.

diruuu, 2022-02-08:
I was able to get it working in some degree with OpenCV. Using the code below:

```C++
libcamera::Span<uint8_t> buffer = app_->Mmap(completed_request->buffers[stream_])[0];
FrameInfo info(completed_request->metadata);
info.sequence = completed_request->sequence;

Mat channel[4];
// Loading PNG image
Mat png_image = imread("/home/pi/image.png", IMREAD_UNCHANGED);
// Split channels
split(png_image, channel);
// Get camera feed buffer
uint8_t *ptr = (uint8_t *)buffer.data();
Mat mYUV(info_.height * 1.5, info_.width, CV_8UC1, ptr, info_.stride);
Mat mYUVClone = mYUV.clone();
// Construct RGB channel for the PNG image
Mat rgb[3] = {channel[0], channel[1], channel[2]};
Mat pngClone;
merge(rgb, 3, pngClone);
// Convert camera feed from YUV 420 into BGR color space
Mat mRGB(info_.height, info_.width, CV_8UC3);
cvtColor(mYUVClone, mRGB, COLOR_YUV2BGR_I420, 3);
// Copy PNG image into the camera feed
pngClone.copyTo(mRGB(Rect(100, 100, png_image.cols, png_image.rows)), channel[3]);
// Convert back to YUV 420 color space
cvtColor(mRGB, mYUV, COLOR_BGR2YUV_I420, 1);
```

The only problem is, the result doesn't look right. Looks like some RGB channels on the PNG image got overlapped or get duplicated. Really appreciate if someone can figure out if this is a problem with OpenCV or with the YUV 420 in general.
[image]

davidplowman, 2022-02-08:
Hi, I'm not really clear exactly what YUV formats OpenCV understands. For example, this line
```
Mat mYUV(info_.height * 1.5, info_.width, CV_8UC1, ptr, info_.stride);
```
looks to me like it's treating the YUV420 as a single plane of size "width * (1.5 * height)", which doesn't feel right. In reality, a YUV420 buffer contains stride*height Y values, followed by stride/2 * height/2 U values, then stride/2 * height/2 V values. (When stride and width do not match, then you have "padding bytes" at the end of each row.)

I was trying to find an online description of what `COLOR_YUV2BGR_I420` really means, but don't seem to have stumbled into anything straightforward.

naushir, 2022-02-25:
Is there any further updates on this issue?  If not, I will close it down shortly.

closed 2022-03-04 by naushir