raspberrypi/rpicam-apps #244: [BUG]

Abdelmuttalib, 2022-02-06:
**Describe the bug**
A clear and concise description of what the bug is.
I have a raspberry pi  model b that i tried m#connect with a v2.1 camera module but the raspberry pi seems not to recognize the camera module, and when i try to run the command `libcamera-hello`. i get the following error 'INFO Camera camera_manager.cpp:293 libcamera v0.0.0
ERROR: *** no cameras available ***'. Although while the command is running, looks like something is trying to flash on the screen but it instantly disappears and this error message is show. I have tried everything and enabled camera many times, before updating the system i would get camera interface as Pi Camera in raspi-config, after updating the system and firmware, it is shown as 'Legacy camera support', Not sure if that has anything to do with the problem. I will provide the output of some commands to make thing easier.

command `uname -a`, 
output `Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux`

command `vcgencmd version`
output `Jan 20 2022 13:57:40 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start_x)`

command `vcgencmd get_camera`
output `supported=1 detected=0`

the content of config.txt
`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
start_x=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
dtoverlay=vc4-fkms-v3d
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtoverlay=imx519
gpu_mem=128
enable_uart=1
`

**Bug report**
Please use the ``camera-bug-report`` tool to create a bug report, and upload it here.

The bug report tool uses the following syntax:

```
camera-bug-report -t <timeout in seconds> -o <output file> -c "<libcamera-app command>"
```

For example,

```
camera-bug-report -t 5 -o bug.txt -c "libcamera-still -t 1000 -o test.jpg"
```
will attempt to run libcamera-still and timeout after 5 seconds if the application has stalled. The script will generate a ``bug.txt`` file that captures all the output logs and system state to help us debug the issue.

You can also run without a timeout:

```
camera-bug-report -o bug.txt -c "libcamera-vid -t 0 -o test.264"
```
This will run ``libcamera-vid`` indefinitely until either you press ``Ctrl+C`` or the application terminates, after which the necessary output logs and system state will be captured.

If you cannot run your application through the ``camera-bug-report`` tool, run it without the ``-c`` command line argument **after running the camera application.**  In these cases, please also provide the command line used to run the application, as well as any output generated during the run.

**Additional context**
Add any other context about the problem here.

davidplowman, 2022-02-06:
Hi
> dtoverlay=imx519

The v2 camera is imx219, not imx519 (perhaps just a typo?). Though generally we recommend not listing the camera at all but using `camera_auto_detect=1` which will detect all 3 official Raspberry Pi cameras automatically.

6by9, 2022-02-06:
Having ```start_x=1```, ```gpu_mem=128```, and an absence of ```camera_auto_detect=1``` implies you've enabled the legacy camera stack through raspi-config. libcamera therefore won't find any cameras.

Abdelmuttalib, 2022-02-10:
Hello, after adding 'camera_auto_detect=1' to confix.txt, the camera not available seem to be gone. But again when i try to run 'libcamera-hello', i get a new error message of 'ERROR: *** failed to configure streams *** '.

Abdelmuttalib, 2022-02-10:
`pi@raspberrypi:~ $ libcamera-hello
[0:03:41.818067236] [1757]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:03:42.116665421] [1759] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/video0
[0:03:42.116772292] [1759] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0901 but control is not exposed by device /dev/video0
[0:03:42.116837162] [1759] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x00980911 but control is not exposed by device /dev/video0
[0:03:42.116901125] [1759] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0903 but control is not exposed by device /dev/video0
Stream configuration adjusted
[0:03:42.122061625] [1757]  INFO Camera camera.cpp:937 configuring streams: (0) 1640x1232-YUV420
[0:03:42.123539458] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y10  Score: 500 (best 500)
[0:03:42.123933440] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y12  Score: 500 (best 500)
[0:03:42.124241662] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y14  Score: 500 (best 500)
[0:03:42.124528569] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA10 Score: 1500 (best 500)
[0:03:42.124813644] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GB10 Score: 1500 (best 500)
[0:03:42.125097792] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG10 Score: 1500 (best 500)
[0:03:42.125380681] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG10 Score: 1500 (best 500)
[0:03:42.125663569] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA81 Score: 2500 (best 500)
[0:03:42.125945810] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BA12 Score: 500 (best 500)
[0:03:42.126228032] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GB12 Score: 500 (best 500)
[0:03:42.126510514] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG12 Score: 500 (best 500)
[0:03:42.126800051] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG12 Score: 500 (best 500)
[0:03:42.130131495] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGB3 Score: 500 (best 500)
[0:03:42.130287606] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BGR3 Score: 500 (best 500)
[0:03:42.130375106] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GB14 Score: 500 (best 500)
[0:03:42.130456458] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt BG14 Score: 500 (best 500)
[0:03:42.130535903] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RG14 Score: 500 (best 500)
[0:03:42.130615181] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GR14 Score: 500 (best 500)
[0:03:42.130694106] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGB4 Score: 500 (best 500)
[0:03:42.130765069] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBAA Score: 1000 (best 500)
[0:03:42.130835884] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pGAA Score: 1000 (best 500)
[0:03:42.130907755] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRAA Score: 1000 (best 500)
[0:03:42.130977088] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pgAA Score: 1000 (best 500)
[0:03:42.131054310] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGGB Score: 2500 (best 500)
[0:03:42.131127569] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBCC Score: 0 (best 0)
[0:03:42.131202421] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pGCC Score: 0 (best 0)
[0:03:42.131276644] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pRCC Score: 0 (best 0)
[0:03:42.131348921] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pgCC Score: 0 (best 0)
[0:03:42.131426940] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pBEE Score: 500 (best 0)
[0:03:42.131504958] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pGEE Score: 500 (best 0)
[0:03:42.131582292] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pREE Score: 500 (best 0)
[0:03:42.131659940] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt pgEE Score: 500 (best 0)
[0:03:42.131735569] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GRBG Score: 2500 (best 0)
[0:03:42.131813144] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GBRG Score: 2500 (best 0)
[0:03:42.131891588] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBO Score: 500 (best 0)
[0:03:42.131968773] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y10P Score: 500 (best 0)
[0:03:42.132044792] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y12P Score: 500 (best 0)
[0:03:42.132124884] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt Y14P Score: 500 (best 0)
[0:03:42.132202755] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBP Score: 500 (best 0)
[0:03:42.132282514] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBQ Score: 500 (best 0)
[0:03:42.132361218] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt RGBR Score: 500 (best 0)
[0:03:42.132440051] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt YVYU Score: 500 (best 0)
[0:03:42.132521569] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt YUYV Score: 500 (best 0)
[0:03:42.132588681] [1759]  INFO RPI raspberrypi.cpp:122 Mode: 1640x1232 fmt GREY Score: 2500 (best 0)
[... cut ...]
[0:03:42.133195181] [1759]  INFO RPI raspberrypi.cpp:624 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected mode: 1640x1232-pgCC
[0:03:42.135722292] [1759] ERROR IPARPI raspberrypi.cpp:567 Unable to find sensor control 0x009e0903
[0:03:42.135847236] [1759] ERROR IPARPI raspberrypi.cpp:376 Sensor control validation failed.
[0:03:42.135904014] [1759] ERROR RPI raspberrypi.cpp:1353 IPA configuration failed!
[0:03:42.136061181] [1759] ERROR RPI raspberrypi.cpp:761 Failed to configure the IPA: -32
ERROR: *** failed to configure streams ***`

naushir, 2022-02-10:
You seem to be running with an old version of libcamera/libcamera-apps that is incompatible with your kernel version.  Please update with
```
sudo apt update
sudo apt upgrade
```
and the libraries should be back in sync. This ought to fix your issue.

Abdelmuttalib, 2022-02-12:
Hello, thank you for the response. 

Running the two commands does some updating and upgrading, but it did not fix the problem for me, I still get the same error above when running 'libcamera-hello'

Abdelmuttalib, 2022-02-12:
Also when trying to run this code 
`from picamera import PiCamera
import time

camera = PiCamera()

camera.start_preview()
time.sleep(2)

camera.capture('one.jpg')
`

I get the following error everytime, even when i enable the legacy camera in raspi-config and reboot.

`mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 456, in _init_camera
    self._camera = mo.MMALCamera()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2279, in __init__
    super(MMALCamera, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 631, in __init__
    mmal_check(
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.ril.camera': Out of memory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Desktop/new.py", line 4, in <module>
    camera = PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 459, in _init_camera
    raise PiCameraError(
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.`

Abdelmuttalib, 2022-02-12:
I think I have done `rpi-update `, back in time when i first started using the raspberry pi, which i know now is not recommended. Could this be the thing that caused the problem ?

Abdelmuttalib, 2022-02-12:
I try also with raspistill, running `raspistill -v -o test.jpg` or `raspivid -o vid.h264`, generates the following error message

'ERROR: the system should be configured for the legacy camera stack'

naushir, 2022-02-25:
Are there any further updates on this issue?  If not, I will close it down shortly.

Abdelmuttalib, 2022-02-25:
How can i do install a new SD card image on the same SD card currently installed on the raspberry pi. Thank you.

naushir, 2022-02-25:
If you don't mind overwriting the contents of the existing SD card (and losing everything on it), then you are use the Raspberry Pi Imager tool obtained from https://www.raspberrypi.com/software/ to overwrite your existing card.

If you want to keep the contents of your SD card, please use a new SD card with the Imager tool.

Abdelmuttalib, 2022-02-28:
I have done a clean install of Raspberry pi OS using the imager on the same SD card, then did the installation and connected the camera to the rasspberry pi and enabled the legacy camera from the 'sudo raspi-config'. Then i did the `libcamera-hello` and i get this error `Error: the system appears to be configured for the legacy camera stack`.

output of command `cat /etc/debian_version` is `11.2`

naushir, 2022-02-28:
> I have done a clean install of Raspberry pi OS using the imager on the same SD card, then did the installation and connected the camera to the rasspberry pi and enabled the legacy camera from the 'sudo raspi-config'.

Please do not enable the legacy stack.  You can run libcamera simply by flashing a fresh image without any configuring required.

Abdelmuttalib, 2022-02-28:
I have just added `dtoverlay=imx219` to the config.txt, and running the command `libcamera-hello` works just fine and the camera seems to be working fine with libcamera commands.

although when i try to do so with python as to the following code `from picamera import PiCamera
import time

camera = PiCamera()
time.sleep(2)

camera.capture('One.jpg')`.

I get this error 
`mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 456, in _init_camera
    self._camera = mo.MMALCamera()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2279, in _init_
    super(MMALCamera, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 631, in _init_
    mmal_check(
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.ril.camera': Out of memory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Desktop/New.py", line 4, in <module>
    camera = PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 431, in _init_
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 459, in _init_camera
    raise PiCameraError(
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.`

Note: I did a restart after adding `dtoverlay=imx219` to the config.txt

davidplowman, 2022-02-28:
Hi, the Picamera library is based on the legacy camera stack and therefore will not work if you enable libcamera. There's a new version of Picamera in development, called Picamera2, and you can find an early preview version [here](https://github.com/raspberrypi/picamera2). You will need to decide whether you wish to use the new libcamera-based code or the legacy code as you cannot run both at once.

Abdelmuttalib, 2022-02-28:
Hello, thank you for the responses.

How can i use the camera module with python, as the main objective of my project is to use the raspberry pi with the camera module to develop an AI powered face recognition door lock, I have seen examples that use usb webcam, how can i use and control the camera module with python to do video streaming for image processing(face recognition) and video recording and saving the video recording for example. Thank you.

davidplowman, 2022-03-01:
Hi, you either need to enable the legacy camera stack and use the old Picamera library, or stay with the new libcamera-based stack and install Picamera2, for which I gave a link previously. I also noted that Picamera2 is very much still in development, though it is perfectly capable of delivering images to OpenCV or TensorFlow Lite (check out the examples folder).

naushir, 2022-03-01:
Given the original issue is now not relevant, I'll close this issue down.  @Abdelmuttalib please feel free to look at our camera forums for the on-going discussions on picamera2.

closed 2022-03-01 by naushir