raspberrypi/rpicam-apps #245: [BUG] Libcamera-vid: dropped frames on RPiZero2W

jlpoolen, 2022-02-08:
I'm piping libcamera-vid into cvlc per the recommended commands in the documentation.  On a Raspberry Pi Zero 2 W ("RPiZ2"), I'm running:

      date;libcamera-vid -t 0 --inline -o - | cvlc stream:///dev/stdin --sout '#rtp{sdp=rtsp://192.168.1.12:8554/stream1}' :demux=h264;date

The server runs.  I connected 2 VLC clients from a Windows desktop connected through the same switch where my Windows desktop is about 4 feet away from the table where the Raspberry Pi Zero 2 W sits.  I noticed a dropped frame, e.g. 25-30%.

[image]

I then stopped the two Windows clients, and ran a VLC client on the RPiZ2 against the same running server and witnessed a dropped frame rate of 27%.  I believe the dropping of frames are occurring before entry into cVLC, but that's just a belief.  I am going to be running gstreamer, as well, but I thought I submit this issue using your recommended commands and software.

[image]

**Bug report**
I am unable to use``camera-bug-report`` , I have an issue pending as #237

Here is the initial bug portion I was able to capture before bits started to flood the console:  [partial_debug_through_1204.txt](https://github.com/raspberrypi/libcamera-apps/files/8020383/partial_debug_through_1204.txt)

**Additional context**
After 50 minutes, my ssh shell where I launched the streaming server has error messages such as these at about 2 per second:

     [121:26:11.216412470] [889408]  WARN RPI raspberrypi.cpp:2118 Dropping unmatched input frame in stream Unicam Embedded

In previous runs, I noticed that the WARNs would start appearing after say, 40 minutes.  

I look forward to having the camera-bug-report working so I can provide you a complete picture in one neat bundle.

The dropped frames seems to be always occurring.  Lastly, when I was running the server without a VLC session displaying anything, in other words sitting idle, the CPU rate was pretty low.  The 2nd connection from the remote windows desktop did not seem to impact.  When I ran VLC on the RPiZ2, then, the CPU bumped up which I'm sure is to be expected.  I did not at any time see any red-lining on the CPU.

naushir, 2022-02-08:
Bit of fortunate timing for this patch.  I have submitted some patches (https://patchwork.libcamera.org/project/libcamera/list/?series=2921) that should fix these drop frame occurrences.  Hopefully this should fix your issue above as well.

The changes are under review, but should be merged soon hopefully.

davidplowman, 2022-02-08:
Another thing to be aware of is that we're having a lot of trouble with the most recent versions of `vlc` (certainly some older versions work much better). We're finding that it can't even play h264 _files_ any more, let alone network streams. In fact, we've been through the documentation recently, explicitly trying to "de-vlc" it.  In practice that means that bug reports are much easier to investigate if they can be presented without the use of vlc. For example, use `libcamera-vid` to stream to a udp address, and have the client play that using (for example) `ffplay`. 

Also, just another small note, slightly lesser powered devices will run better without showing an X Windows preview, I don't know if you've tried that. Either quit (or suspend) X Windows and you'll get the DRM/KMS window which is much cheaper, or use `-n` to suppress the preview window altogether.

jlpoolen, 2022-02-08:
Thank you for the insight @davidplowman .  I believe VLC integrates the "open sourced" project LIVE555.  I've been trying to find an alternative to LIVE555 and hope to become proficient with gstreamer.  Since the Raspberry documentation used VLC, I thought perhaps you might know something I don't and figured it was fair game to use VLC.  

Also, I ran the server from an ssh console within which libcamera determined it would not try to display the preview window (I tried setting DISPLAY to the machine I'm accessing the RPiZ2w from, but no luck with libcamera (I do run Emacs via an X-window, though)).  I"ll try to use the  no preview option in the future, if there is much future for using cvlc here.  

I've been witnessing frame drops in the commercial product Reolink surveillance cameras which uses LIVE555 and I don't know if it is my network, Reolink's hardware, and/or LIVE555.  The problems I encountered has brought me to the Raspberry Pi platform so I can get a handle on what's causing the problem.  My goal is to see if I can stream high quality and consistent 24/7 video to [Moonfire-nvr](https://github.com/scottlamb/moonfire-nvr) which captures, preserves, and organizes the tremendously unwieldy videos streams.  I'm also considering whether to explore the Raspberry Pi Compute Module 4, though I think that is more horsepower than is warranted and raises the price/stakes.  So, right now the main focus of my inquiry is how well does the Raspberry Pi hardware and libcamera work and whether there are real world limitations for a 24/7 operation.

jlpoolen, 2022-02-08:
After consulting with Scott Lamb of the [Moonfire-nvr](https://github.com/scottlamb/moonfire-nvr) project, I concur that any time spent addressing problems arising from a tool chain involving the LIVE555 server is not merited.  Therefore, I am okay with having this issue closed.  I will focus my attention now on gstreamer.

davidplowman, 2022-02-09:
Thanks for the update!

closed 2022-02-09 by davidplowman