raspberrypi/rpicam-apps #246: libcamera-vid crash (Kernel Oops) on older RPi zero W and camera V2.1

j-lag, 2022-02-08:
Hello folks!

**Describe the bug**

"libcamera-vid -g 1 -b 500000 --framerate 30  --width=1920 --height=1080 -t 0" produce a crash and a kernel Oops on RPI zero W and camera V2.1.
The crash was detected on an older buster image but is still present on a freashly installed Bullseye lite.

Note: 
- Similar command work with legacy raspivid but libcamera is required to fine tune camera parameter using the json tunning file.
- No problem on newer RPI zero W2.
- system need a reboot after crash to recover the camera.

**Bug report**
camera-bug-report -t 0 -o bug.txt -c "libcamera-vid -g 1 -b 500000 --framerate 30  --width=1920 --height=1080 -t 0"
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 08-02-2022 (11:44:58)
Command: /usr/bin/camera-bug-report -t 0 -o bug.txt -c libcamera-vid -g 1 -b 500000 --framerate 30  --width=1920 --height=1080 -t 0

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
raspberrypi

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 997.08
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000b7ad03cf
Model		: Raspberry Pi Zero W Rev 1.1

--------------------------------------------------------------------------------
Configuration
--------------------------------------------------------------------------------
** cat /boot/cmdline.txt **
console=serial0,115200 console=tty1 root=PARTUUID=6409d7a3-02 rootfstype=ext4 fsck.repair=yes rootwait

** cat /boot/config.txt **
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
[... 1802 lines cut ...]
[0:03:51.394534248] [582]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!
[0:03:51.398445618] [582]  INFO RPI raspberrypi.cpp:1313 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:03:51.401884943] [581]  INFO Camera camera.cpp:1028 configuring streams: (0) 1920x1080-YUV420
[0:03:51.415846264] [582]  INFO RPI raspberrypi.cpp:738 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SBGGR10_1X10 - Selected unicam format: 1920x1080-pBAA
[0:03:51.640433517] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:03:51.640787550] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:03:51.640986569] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:03:51.641181587] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:03:51.641352603] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:03:51.641542621] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:03:51.641729639] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:03:51.641898655] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:03:51.642084673] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:03:51.642255689] [589]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 1
[0:03:51.645307978] [582]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:03:51.645495996] [582]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:03:51.645633009] [582]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:03:51.645767021] [582]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:03:51.645898034] [582]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
[0:03:51.646030046] [582]  INFO RPISTREAM rpi_stream.cpp:133 No buffers available for ISP Output0
Error: ***** INTERRUPT *****
```

**Additional context**
https://forums.raspberrypi.com/viewtopic.php?p=1970733#p1970733

davidplowman, 2022-02-08:
Hi, I've just installed a fresh Bullseye on a Pi Zero and tried exactly the same thing. I take it you're seeing it fail pretty quickly? Mine's been running for about an hour without any complaints. Here are the details of my system:
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 997.08
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000fb436ec1
Model		: Raspberry Pi Zero W Rev 1.1
```
```
pi@raspberrypi:~ $ cat /proc/meminfo
MemTotal:         439944 kB
MemFree:          213136 kB
MemAvailable:     298988 kB
Buffers:           16048 kB
Cached:           110532 kB
SwapCached:            0 kB
Active:            56892 kB
Inactive:          90920 kB
Active(anon):        524 kB
Inactive(anon):    21464 kB
Active(file):      56368 kB
Inactive(file):    69456 kB
Unevictable:          16 kB
Mlocked:              16 kB
SwapTotal:        102396 kB
SwapFree:         102396 kB
Dirty:              1352 kB
Writeback:             0 kB
AnonPages:         21260 kB
Mapped:            50344 kB
Shmem:               756 kB
KReclaimable:      10168 kB
Slab:              20628 kB
SReclaimable:      10168 kB
SUnreclaim:        10460 kB
KernelStack:         872 kB
PageTables:         1280 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      322368 kB
Committed_AS:     266932 kB
VmallocTotal:     573440 kB
VmallocUsed:        5528 kB
VmallocChunk:          0 kB
Percpu:               96 kB
CmaTotal:         262144 kB
CmaFree:          106560 kB
```
```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux
```
```
pi@raspberrypi:~ $ vcgencmd version
Jan 20 2022 13:58:22 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
```
```
pi@raspberrypi:~ $ libcamera-vid --version
libcamera-apps build: 7ac5197216ea 01-02-2022 (14:49:18)
libcamera build: v0.0.0+3406-e96d0201
```
So I think that's the same as yours, just slightly newer libcamera and libcamera-apps, though not that there's any reason to suspect that of making a difference. And finally my command line too:
```
libcamera-vid --width 1920 --height 1080 --framerate 30 -g 1 -b 500000 -t 0
```
I do see all those same "Expacting an uncached alias" warnings in dmesg, but that doesn't seem to affect how it runs. So very puzzling.

PinkFreud, 2022-03-02:
I just received the above oops after a few (~ 10 min) of running libcamera-vid on a single core Zero W.

This was working pretty reliably for me before I applied updates today.  Packages updated, from /var/log/apt/history.log:

`
Start-Date: 2022-03-01  20:16:56
Commandline: apt upgrade
Requested-By: pi (1000)
Upgrade: rpi-eeprom:armhf (13.8-1, 13.10-1), libsasl2-2:armhf (2.1.27+dfsg-2.1, 2.1.27+dfsg-2.1+deb11u1), raspi-config:armhf (20220126, 20220301), libexpat1:armhf (2.2.10-2, 2.2.10-2+deb11u2), libwbclient0:armhf (2:4.13.13+dfsg-1~deb11u2, 2:4.13.13+dfsg-1~deb11u3), libsasl2-modules-db:armhf (2.1.27+dfsg-2.1, 2.1.27+dfsg-2.1+deb11u1), libssl1.1:armhf (1.1.1k-1+deb11u1, 1.1.1k-1+deb11u1+rpt1), libexpat1-dev:armhf (2.2.10-2, 2.2.10-2+deb11u2), openssl:armhf (1.1.1k-1+deb11u1, 1.1.1k-1+deb11u1+rpt1)
End-Date: 2022-03-01  20:17:58
`

Nothing in this list immediately jumps out at me as potentially being the culprit, but I can't figure out what else would have changed.

Like above, I'm seeing spurious uncached alias messages spammed to the kernel log:
`
...
[ 1424.656550] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1424.710183] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1424.811670] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1424.854064] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1424.889578] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1424.983571] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1425.017373] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
[ 1425.093826] vc_sm_cma_import_dmabuf_internal: Expecting an uncached alias for dma_addr 0x8b108000
...
`
Looking back through kern.log, I see these messages being repeatedly spammed on previous boots, so it appears it may just not have resulted in a crash until now.

PinkFreud, 2022-03-02:
uname -a:
`Linux garage 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux`

vcgencmd version:
```
Jan 20 2022 13:58:22 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)
```

libcamera-hello --version:
```
libcamera-apps build: 7ac5197216ea 01-02-2022 (14:49:18)
libcamera build: v0.0.0+3406-e96d0201
```

commandline:
`libcamera-vid -t 0 --inline --listen --vflip 1 --hflip 1 --width 1920 --height 1080 -o tcp://0.0.0.0:8888`

Console output:
I didn't notice if libcamera-vid output anything related to the crash, as I had it running in screen at the time.  The first indication I had of an issue was the stream disappearing, and when I looked at my ssh session to the Zero W, I saw the oops:
```
Mar  2 00:28:14 garage kernel: [ 1426.757423] 8<--- cut here ---
Mar  2 00:28:14 garage kernel: [ 1426.757496] Unable to handle kernel paging request at virtual address bfa30b98
Mar  2 00:28:14 garage kernel: [ 1426.757532] pgd = ed98d9b0
Mar  2 00:28:14 garage kernel: [ 1426.757551] [bfa30b98] *pgd=00000000
Mar  2 00:28:14 garage kernel: [ 1426.757597] Internal error: Oops: 80000005 [#1] ARM
Mar  2 00:28:14 garage kernel: [ 1426.757620] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep ov5647 snd_soc_hdmi_codec hci_uart btbcm bluetoot
h ecdh_generic ecc libaes spidev 8021q garp stp llc brcmfmac vc4 brcmutil cec sha256_generic libsha256 drm_kms_helper cfg80211 i2c_mux_pinctrl i2c_mux snd_soc_core raspberrypi_hwmon
 snd_compress snd_pcm_dmaengine syscopyarea rfkill sysfillrect sysimgblt fb_sys_fops bcm2835_unicam v4l2_dv_timings v4l2_fwnode bcm2835_codec(C) i2c_bcm2835 bcm2835_isp(C) bcm2835_v
4l2(C) v4l2_mem2mem snd_bcm2835(C) bcm2835_mmal_vchiq(C) spi_bcm2835 videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd vi
deodev mc vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Mar  2 00:28:14 garage kernel: [ 1426.758039] CPU: 0 PID: 787 Comm: libcamera-vid Tainted: G         C        5.10.92+ #1514
Mar  2 00:28:14 garage kernel: [ 1426.758063] Hardware name: BCM2835
Mar  2 00:28:14 garage kernel: [ 1426.758091] PC is at 0xbfa30b98
Mar  2 00:28:14 garage kernel: [ 1426.758594] LR is at video_usercopy+0x404/0x558 [videodev]
Mar  2 00:28:14 garage kernel: [ 1426.758625] pc : [<bfa30b98>]    lr : [<bf1f4b8c>]    psr: 60000013
Mar  2 00:28:14 garage kernel: [ 1426.758646] sp : c36cfe18  ip : c36cfe18  fp : c36cfeec
Mar  2 00:28:14 garage kernel: [ 1426.758665] r10: 00000000  r9 : c0bb9028  r8 : c36cfe3c
Mar  2 00:28:14 garage kernel: [ 1426.758685] r7 : 00000000  r6 : 0000003c  r5 : c0505611  r4 : c0445611
Mar  2 00:28:14 garage kernel: [ 1426.758705] r3 : c29a82c0  r2 : bf216a60  r1 : 341b2665  r0 : c2894c78
Mar  2 00:28:14 garage kernel: [ 1426.758749] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Mar  2 00:28:14 garage kernel: [ 1426.758777] Control: 00c5387d  Table: 03368008  DAC: 00000055
Mar  2 00:28:14 garage kernel: [ 1426.758810] Process libcamera-vid (pid: 787, stack limit = 0x475d0712)
Mar  2 00:28:14 garage kernel: [ 1426.758840] Stack: (0xc36cfe18 to 0xc36d0000)
Mar  2 00:28:14 garage kernel: [ 1426.758866] fe00:                                                       c0050864 c29a82c0
Mar  2 00:28:14 garage kernel: [ 1426.758900] fe20: 00000001 c36cfe7c b14b79f4 c29a82c0 c36f0300 b14b79b0 00097104 00000002
Mar  2 00:28:14 garage kernel: [ 1426.758934] fe40: 0000000a 00000000 00004000 00000001 00000000 00000592 00000000 000731b3
Mar  2 00:28:14 garage kernel: [ 1426.758967] fe60: 00000000 00000000 00000000 00000000 00000000 00000000 00000004 c29a82c0
Mar  2 00:28:14 garage kernel: [ 1426.759001] fe80: 00000001 00000000 00000000 c36cfe98 c0098c04 c0096fe8 00000000 c00850a8
Mar  2 00:28:14 garage kernel: [ 1426.759035] fea0: c36cfefc c36cfeb0 c00850a8 c06c2280 00000592 00000000 336a9641 341b2665
Mar  2 00:28:14 garage kernel: [ 1426.759069] fec0: 00000592 bf1f4ce0 c36f0301 c36f0300 00000000 c0bb9028 b14b79b0 00000019
Mar  2 00:28:14 garage kernel: [ 1426.759104] fee0: c36cfefc c36cfef0 bf1f4cfc bf1f4794 c36cff14 c36cff00 bf1ec148 bf1f4cec
Mar  2 00:28:14 garage kernel: [ 1426.759138] ff00: c0445611 c36f0301 c36cffa4 c36cff18 c020c1f8 bf1ec108 00000000 00000000
Mar  2 00:28:14 garage kernel: [ 1426.759172] ff20: 00000592 00000000 36726c6a 00e99960 00000000 341b2665 0bebc200 c0bb9028
Mar  2 00:28:14 garage kernel: [ 1426.759207] ff40: 00000001 b14b7970 000000a8 c00083e4 c36ce000 00000000 c36cffa4 c36cff68
Mar  2 00:28:14 garage kernel: [ 1426.759241] ff60: c020ed14 c020dde8 0bebc200 341b2665 00000592 341b2665 36726c6a 00e99820
Mar  2 00:28:14 garage kernel: [ 1426.759276] ff80: 0000000b b14b79b0 00000036 c00083e4 c36ce000 00000000 00000000 c36cffa8
Mar  2 00:28:14 garage kernel: [ 1426.759310] ffa0: c0008240 c020c058 00e99820 0000000b 00000019 c0445611 b14b79b0 b14b79f4
Mar  2 00:28:14 garage kernel: [ 1426.759345] ffc0: 00e99820 0000000b b14b79b0 00000036 b14b7970 00e998b8 b6b50124 00000019
Mar  2 00:28:14 garage kernel: [ 1426.759380] ffe0: b6f2e144 b14b7964 b6f15414 b6c421ac 40000010 00000019 00000000 00000000
Mar  2 00:28:14 garage kernel: [ 1426.759405] Backtrace: 
Mar  2 00:28:14 garage kernel: [ 1426.759853] [<bf1f4788>] (video_usercopy [videodev]) from [<bf1f4cfc>] (video_ioctl2+0x1c/0x24 [videodev])
[... cut ...]
This is a freshly installed image of Bullseye (Raspberry Pi OS lite) in mid-Janurary.  The only software on here besides the base OS are to run a few Pimoroni boards (Enviro pHAT, Scroll pHAT, LED Shim).  X is not installed, and the Pi runs headless, with me using ssh to manage it.

As for whether this is repeatable: As I mentioned above, this is the first time I noticed the crash, though I only added the camera to this system in Jan.  I hadn't noticed the spurious dma messages before, but it looks like they've been spamming the kernel log for some time.

PinkFreud, 2022-04-06:
This crash has become far easier to reproduce with raspberrypi-kernel 1:1.20220328-1.  It typically occurs within a few minutes, if not seconds, of starting libcamera-vid.  Suffice it to say, streaming on the Pi Zero W with libcamera-vid is now useless, as this requires a reboot each time to make the camera usable again.

uname -a
```
Linux garage 5.15.30+ #1536 Mon Mar 28 13:31:21 BST 2022 armv6l GNU/Linux
```

vcgencmd version
```
Mar 24 2022 13:20:54 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
```

libcamera-hello --version
```
libcamera-apps build: 0dc5ea038c79 04-03-2022 (10:29:30)
libcamera build: v0.0.0+3424-e68e0f1e
```

dmesg
```
[Tue Apr  5 22:50:51 2022] 8<--- cut here ---
[Tue Apr  5 22:50:51 2022] Unable to handle kernel paging request at virtual address bfa89c9c
[Tue Apr  5 22:50:51 2022] pgd = 5350144d
[Tue Apr  5 22:50:51 2022] [bfa89c9c] *pgd=00000000
[Tue Apr  5 22:50:51 2022] Internal error: Oops: 80000005 [#1] ARM
[Tue Apr  5 22:50:51 2022] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep ov5647 snd_soc_hdmi_codec hci_uart btbcm bluetooth ecdh_generic ecc libaes spidev 8021q garp stp llc vc4 cec brcmfmac brcmutil sha256_generic libsha256 drm_kms_helper snd_soc_core cfg80211 snd_compress snd_pcm_dmaengine syscopyarea sysfillrect i2c_mux_pinctrl sysimgblt i2c_mux fb_sys_fops raspberrypi_hwmon rfkill bcm2835_unicam v4l2_dv_timings v4l2_fwnode v4l2_async bcm2835_codec(C) bcm2835_isp(C) i2c_bcm2835 bcm2835_v4l2(C) snd_bcm2835(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig spi_bcm2835 videobuf2_memops snd_pcm videobuf2_v4l2 videobuf2_common snd_timer snd videodev vc_sm_cma(C) mc fixed uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[Tue Apr  5 22:50:51 2022] CPU: 0 PID: 1327 Comm: libcamera-vid Tainted: G         C        5.15.30+ #1536
[Tue Apr  5 22:50:51 2022] Hardware name: BCM2835
[Tue Apr  5 22:50:51 2022] PC is at 0xbfa89c9c
[Tue Apr  5 22:50:51 2022] LR is at video_usercopy+0x4e8/0x5a8 [videodev]
[Tue Apr  5 22:50:51 2022] pc : [<bfa89c9c>]    lr : [<bf24dc90>]    psr: 60000013
[Tue Apr  5 22:50:51 2022] sp : c21a7e10  ip : c21a7e10  fp : c21a7eec
[Tue Apr  5 22:50:51 2022] r10: c0c1f028  r9 : 00000000  r8 : 00000000
[Tue Apr  5 22:50:51 2022] r7 : c21a7e3c  r6 : 00000000  r5 : c044560f  r4 : c050560f
[Tue Apr  5 22:50:51 2022] r3 : 00000000  r2 : bf26fa40  r1 : c7072cd2  r0 : c251c7c0
[Tue Apr  5 22:50:51 2022] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[Tue Apr  5 22:50:51 2022] Control: 00c5387d  Table: 02548008  DAC: 00000055
[Tue Apr  5 22:50:51 2022] Register r0 information: slab kmalloc-8k start c251c000 pointer offset 1984 size 8192
[Tue Apr  5 22:50:51 2022] Register r1 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Register r2 information: 58-page vmalloc region starting at 0xbf245000 allocated at load_module+0xa74/0x25f8
[Tue Apr  5 22:50:51 2022] Register r3 information: NULL pointer
[Tue Apr  5 22:50:51 2022] Register r4 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Register r5 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Register r6 information: NULL pointer
[Tue Apr  5 22:50:51 2022] Register r7 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Register r8 information: NULL pointer
[Tue Apr  5 22:50:51 2022] Register r9 information: NULL pointer
[Tue Apr  5 22:50:51 2022] Register r10 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Register r11 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Register r12 information: non-slab/vmalloc memory
[Tue Apr  5 22:50:51 2022] Process libcamera-vid (pid: 1327, stack limit = 0xb2c69be7)
[Tue Apr  5 22:50:51 2022] Stack: (0xc21a7e10 to 0xc21a8000)
[Tue Apr  5 22:50:51 2022] 7e00:                                     c21a7e3b c04d67a0 c0014720 00000000
[Tue Apr  5 22:50:51 2022] 7e20: 00000000 00000000 00000000 00000000 c3268000 b58da094 002642a8 00000001
[Tue Apr  5 22:50:51 2022] 7e40: 00000001 002f7600 00004002 00000000 00000000 0000254b 00000000 0005528b
[Tue Apr  5 22:50:51 2022] 7e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000004 00000013
[Tue Apr  5 22:50:51 2022] 7e80: 002f8000 00000000 00000000 c1bb0000 c00526cc 00000000 00000000 c7072cd2
[Tue Apr  5 22:50:51 2022] 7ea0: c32680a0 00000008 00000000 00000000 c0c1f028 c21a7f60 c21a7f5c c7072cd2
[Tue Apr  5 22:50:51 2022] 7ec0: c0209de8 bf24dd50 c3268001 00000000 c3268000 c0c1f028 b58da094 0000000f
[Tue Apr  5 22:50:51 2022] 7ee0: c21a7efc c21a7ef0 bf24dd6c bf24d7b4 c21a7f14 c21a7f00 bf245148 bf24dd5c
[Tue Apr  5 22:50:51 2022] 7f00: c044560f c3268001 c21a7fa4 c21a7f18 c021fd78 bf245108 00000010 00000000
[Tue Apr  5 22:50:51 2022] 7f20: 0000254b c04d67b4 c21a7f74 c21a7f38 c007f00c c04d67a0 0000254b 00000000
[Tue Apr  5 22:50:51 2022] 7f40: 20069136 00000000 00000193 c7072cd2 c21a6000 b58d9e98 c0c1f028 b6f42880
[Tue Apr  5 22:50:51 2022] 7f60: 00000193 c00083e4 c21a7fa4 c21a7f78 c0090458 c7072cd2 0000254b b58da094
[Tue Apr  5 22:50:51 2022] 7f80: b4f11850 b4f09560 00000036 c00083e4 c21a6000 00000000 00000000 c21a7fa8
[Tue Apr  5 22:50:51 2022] 7fa0: c0008260 c021fc70 b58da094 b4f11850 0000000f c044560f b58da094 b6de4788
[Tue Apr  5 22:50:51 2022] 7fc0: b58da094 b4f11850 b4f09560 00000036 fffffff8 b58da1cc b6ac2124 00eb0b58
[... cut ...]
[Tue Apr  5 22:50:51 2022] 7fc0: b58da094 b4f11850 b4f09560 00000036 fffffff8 b58da1cc b6ac2124 00eb0b58
[Tue Apr  5 22:50:51 2022] 7fe0: b6a96f18 b58da06c b6a22da0 b6bb41ac
[Tue Apr  5 22:50:51 2022]  r10:00000000 r9:c21a6000 r8:c00083e4 r7:00000036 r6:b4f09560 r5:b4f11850
[Tue Apr  5 22:50:51 2022]  r4:b58da094
[Tue Apr  5 22:50:51 2022] Code: bad PC value
[Tue Apr  5 22:50:51 2022] ---[ end trace 6d128aa3735ea8ae ]---
```

The rest is pretty much the same - this is the same Pi Zero W as before, and memory usage is similar as well.  I'm running libcamera-vid with the same arguments.

davidplowman, 2022-04-08:
I've had another go at trying to reproduce this, using a fresh install of yesterday's new OS Lite (`uname -a` = `Linux raspberrypi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux` and `vcgencmd version` and `libcamera-hello --version` reporting the same as you). I've installed nothing else and changed no configuration anywhere.

My command line is `libcamera-vid -t 0 -n --inline --listen --vflip --hflip --width 1920 --height 1080 -o tcp://0.0.0.0:8888` and I'm dumping the output on my PC with `netcat`.

I just can't get this thing to fail, and it's been going for hours. But at least the spam in `dmesg` has disappeared. I'm starting to wonder if we should be having a go with another device, would you happen to have another Pi Zero that you could try it on? Out of interest, does it also fail for you if you send the output straight to `/dev/null`, avoiding the network altogether (just trying to simplify the test case)?

PinkFreud, 2022-04-10:
> I've had another go at trying to reproduce this, using a fresh install of yesterday's new OS Lite (`uname -a` = `Linux raspberrypi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux` and [...]

Hmmmm.  I'll mention that this isn't a standard RasPi camera module, but rather a third party camera (I was interested in using a wide-angle lens: https://www.waveshare.com/product/rpi-camera-h.htm ), but I had this camera previously connected to a RPi 3 running buster w/ the old Raspi camera utils, and it worked without issue.

I have a few options available to me here: I can swap the Zero W, as I have some spares.  I can also try a spare RasPi camera module (I have some spare v1s, and perhaps a v2 as well).

I also have some hw stacked on the gpio pins on here (Pimoroni LED Shim, Pimoroni Enviro pHAT, Pimoroni Scroll pHAT HD - all use i2C via gpio pins 2+3, with LED control for the enviro pHat on additional gpio pin 4), though I think it's unlikely to be interfering with the camera module.  I can try removing the hw stack to see if the issue continues to occur, especially since you're unable to reproduce this issue.

naushir, 2022-04-11:
> I also have some hw stacked on the gpio pins on here (Pimoroni LED Shim, Pimoroni Enviro pHAT, Pimoroni Scroll pHAT HD - all use i2C via gpio pins 2+3, with LED control for the enviro pHat on [...]

@PinkFreud, that would be useful.  There is an off-chance that you are seeing issue because of additional power draw or some other hardware interaction that we may not be able to reproduce on a standard platform.

PinkFreud, 2022-04-12:
I removed the hw stack tonight and tried again.  Kernel oops within a minute or two of streaming, so I don't think it's related to power draw.  On the off chance that the cable was seated incorrectly, I also reseated it, to no avail.

I'll try swapping out both the camera and Pi one at a time tomorrow night, if I have time.

PinkFreud, 2022-04-15:
I haven't had a chance to get back to this until now.  I just swapped the camera for an official RasPi v1 camera (ov5647) and a v2.1 (imx219).  Same error appeared with both.

I'll try swapping the Pi next, either tonight or tomorrow night.

PinkFreud, 2022-04-15:
And no go on the Pi.  Swapping the Pi had no effect - the crash still occurs.

```
[  472.341754] 8<--- cut here ---
[  472.341841] Unable to handle kernel paging request at virtual address bfa40c9c
[  472.341876] pgd = 0499dfb2
[  472.341906] [bfa40c9c] *pgd=00000000
[  472.341948] Internal error: Oops: 80000005 [#1] ARM
[  472.341977] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep imx219 snd_soc_hdmi_codec hci_uart btbcm bluetooth ecdh_generic ecc libaes spidev 8021q garp stp llc vc4 cec brcmfmac brcmutil drm_kms_helper sha256_generic libsha256 i2c_mux_pinctrl i2c_mux snd_soc_core snd_compress snd_pcm_dmaengine cfg80211 syscopyarea sysfillrect sysimgblt fb_sys_fops raspberrypi_hwmon rfkill bcm2835_unicam v4l2_dv_timings v4l2_fwnode v4l2_async bcm2835_codec(C) snd_bcm2835(C) i2c_bcm2835 spi_bcm2835 v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) snd_pcm videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_timer snd vc_sm_cma(C) videodev mc fixed uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[  472.342461] CPU: 0 PID: 637 Comm: libcamera-vid Tainted: G         C        5.15.30+ #1536
[  472.342496] Hardware name: BCM2835
[  472.342514] PC is at 0xbfa40c9c
[  472.342541] LR is at video_usercopy+0x4e8/0x5a8 [videodev]
[  472.343181] pc : [<bfa40c9c>]    lr : [<bf204c90>]    psr: 60000013
[  472.343206] sp : c2677e10  ip : c2677e10  fp : c2677eec
[  472.343227] r10: c0c1f028  r9 : 00000000  r8 : 00000000
[  472.343246] r7 : c2677e3c  r6 : 00000000  r5 : c044560f  r4 : c050560f
[  472.343268] r3 : 00000000  r2 : bf226a40  r1 : d404a0e3  r0 : c246a7c0
[  472.343311] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  472.343342] Control: 00c5387d  Table: 026f0008  DAC: 00000055
[  472.343364] Register r0 information: slab kmalloc-8k start c246a000 pointer offset 1984 size 8192
[  472.343434] Register r1 information: 0-page vmalloc region starting at 0xcb000000 allocated at iotable_init+0x0/0x110
[  472.343517] Register r2 information: 58-page vmalloc region starting at 0xbf1fc000 allocated at load_module+0xa74/0x25f8
[  472.343582] Register r3 information: NULL pointer
[  472.343610] Register r4 information: non-slab/vmalloc memory
[  472.343640] Register r5 information: non-slab/vmalloc memory
[  472.343668] Register r6 information: NULL pointer
[  472.343693] Register r7 information: non-slab/vmalloc memory
[  472.343720] Register r8 information: NULL pointer
[  472.343745] Register r9 information: NULL pointer
[  472.343770] Register r10 information: non-slab/vmalloc memory
[  472.343797] Register r11 information: non-slab/vmalloc memory
[  472.343825] Register r12 information: non-slab/vmalloc memory
[  472.343851] Process libcamera-vid (pid: 637, stack limit = 0xa3b4d373)
[  472.343886] Stack: (0xc2677e10 to 0xc2678000)
[  472.343918] 7e00:                                     c2677e3b c288ba80 c2677e5c 00000000
[  472.343955] 7e20: 00000000 00000000 00000000 00000000 c28ebbe0 b5952094 00c9c018 00000006
[  472.343991] 7e40: 00000001 002f7600 00004002 00000000 00000000 000001d8 00000000 000239a1
[  472.344028] 7e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000004 00000018
[  472.344066] 7e80: 002f8000 00000000 00000000 c2677e98 c0046c80 c000b100 c1214014 c288ba80
[  472.344103] 7ea0: c0c2b2a0 c12d0000 00000000 c19a7540 c2677edc c2677ec0 c0013798 d404a0e3
[  472.344142] 7ec0: 00000000 bf204d50 c28ebbe1 00000000 c28ebbe0 c0c1f028 b5952094 0000000f
[  472.344179] 7ee0: c2677efc c2677ef0 bf204d6c bf2047b4 c2677f14 c2677f00 bf1fc148 bf204d5c
[  472.344218] 7f00: c044560f c28ebbe1 c2677fa4 c2677f18 c021fd78 bf1fc108 00000010 00000000
[  472.344255] 7f20: 000001d8 c04d67b4 c2677f74 c2677f38 c007f00c c04d67a0 000001d8 00000000
[  472.344291] 7f40: 13b84d8b 00000000 00000193 d404a0e3 c2676000 b5951e98 c0c1f028 b6fba880
[  472.344330] 7f60: 00000193 c00083e4 c2677fa4 c2677f78 c0090458 d404a0e3 000001d8 b5952094
[  472.344368] 7f80: b50180d8 b5009858 00000036 c00083e4 c2676000 00000000 00000000 c2677fa8
[  472.344406] 7fa0: c0008260 c021fc70 b5952094 b50180d8 0000000f c044560f b5952094 b6e5c788
[  472.344442] 7fc0: b5952094 b50180d8 b5009858 00000036 fffffff8 b59521cc b6b3a124 00528b58
[  472.344479] 7fe0: b6b0ef18 b595206c b6a9ada0 b6c2c1ac 60000010 0000000f 00000000 00000000
[  472.344509] Backtrace: 
[  472.344533] [<bf2047a8>] (video_usercopy [videodev]) from [<bf204d6c>] (video_ioctl2+0x1c/0x24 [videodev])
[  472.345394]  r10:0000000f r9:b5952094 r8:c0c1f028 r7:c28ebbe0 r6:00000000 r5:c28ebbe1
[  472.345431]  r4:bf204d50
[  472.345452] [<bf204d50>] (video_ioctl2 [videodev]) from [<bf1fc148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[  472.346204] [<bf1fc0fc>] (v4l2_ioctl [videodev]) from [<c021fd78>] (sys_ioctl+0x114/0x9b0)
[  472.346624]  r5:c28ebbe1 r4:c044560f
[  472.346647] [<c021fc64>] (sys_ioctl) from [<c0008260>] (ret_fast_syscall+0x0/0x1c)
[  472.346696] Exception stack(0xc2677fa8 to 0xc2677ff0)
[  472.346730] 7fa0:                   b5952094 b50180d8 0000000f c044560f b5952094 b6e5c788
[  472.346769] 7fc0: b5952094 b50180d8 b5009858 00000036 fffffff8 b59521cc b6b3a124 00528b58
[  472.346802] 7fe0: b6b0ef18 b595206c b6a9ada0 b6c2c1ac
[  472.346834]  r10:00000000 r9:c2676000 r8:c00083e4 r7:00000036 r6:b5009858 r5:b50180d8
[  472.346863]  r4:b5952094
[  472.346891] Code: bad PC value
[  472.346920] ---[ end trace 6d128aa3735ea8ae ]---
```

davidplowman, 2022-04-15:
I'm wondering if you might be able to make an image - tested and known to exhibit the problem - that you could post somewhere for us to download and try? I'm running out of ideas a little bit here...

PinkFreud, 2022-04-15:
@davidplowman: That shouldn't be much of a problem to do.  It's just the base Raspberry Pi OS image w/ some Pimoroni software installed on it.

I'm finding myself wondering, though - the indications are that this isn't a hardware issue, but rather a software issue.  Considering this from a software standpoint, how are you playing the stream when you test?

I have zoneminder set up here, using ffmpeg to pull the stream in.  When ZM connects, I'll get the stream for a few mins before the crash occurs.

If you let me know how you're testing, I can try that on my end, too, to see if it occurs with whatever client you're using.  Perhaps ZM and/or ffmpeg is doing something specific to cause the issue?

davidplowman, 2022-04-16:
On the receiving end I'm just reading the bytes with netcat and dumping them. You'd have thought what happens on the client end can't affect the server too much, though with a TCP connection I suppose you can get back-pressure if you don't read stuff fast enough. I don't have ZoneMinder, but could you share the ffmpeg command that you use?

Also, out of interest, does the crash happen if you (a) stream to a UDP socket and (b) if you don't stream the output at all (e.g. use `-o /dev/null`). Maybe you could try the netcat experiment too (I think the command on the client was simply `netcat <ip-addr> <port> >/dev/null`).

PinkFreud, 2022-04-16:
Unfortunately, I can't tell you what ffmpeg settings are in use by zm - it's merely an 'ffmpeg' option.  I can check the process list the next time I try to stream to it, to see if it's calling the external ffmpeg utility and which args it's using.

If it helps, I'm using the standard ZM package included in debian bullseye - it looks like current raspberry pi os has the same release, which isn't terribly surprising.

The relevant settings pages in ZM (the buffers page is using default settings, but I've included it anyway):
[image]
[image]
[image]

I think your suggested experiments with are worthwhile.  I'll give them a try and let you know.

PinkFreud, 2022-04-16:
Both -o /dev/null (~ 8 minutes) and using netcat locally on the Pi to connect to tcp 8888 and write to /dev/null (~ 2 minutes) trigger the crash for me.  I'm not sure it's worth trying udp at this point, given the previous two methods failed.

What's the next step - creating an image to send to you?

PinkFreud, 2022-04-17:
@davidplowman: Sanitized and shrunk image: https://nextcloud.mirkwood.net/nextcloud/index.php/s/s2Nk6AmJzj9XLpb

davidplowman, 2022-04-19:
I think it's good that we seem to be able to rule out networking issues. Thanks for that image - I'll give it a try and report back what happens.

davidplowman, 2022-04-19:
Hi again, could you double-check that link or perhaps the permissions associated with it, I'm getting a "File not found" error when I click on it. Thanks.

PinkFreud, 2022-04-19:
@davidplowman: Ah, looks like the link expired.  I've recreated the share: https://nextcloud.mirkwood.net/nextcloud/index.php/s/5QQ45TSW7TeG8y2

davidplowman, 2022-04-20:
I set the image going yesterday and it's still running quite happily this morning over 10 hours later - nothing in dmesg, and the libcamera-vid process seems to be using about 55% CPU. I'm running out of things to suggest. You've swapped out most of your hardware, did you try swapping out your power supply too? (Is it an official Raspberry Pi one?)

PinkFreud, 2022-04-20:
Huh, no I did not.  I'll try a different power supply tonight.

PinkFreud, 2022-04-21:
@davidplowman: I pulled a new 2.4A power supply (not offical RPi, unfortunately) out of it's box - brand-new, unused, and a brand purchased from Element14 with good results on other Pis.  I was able to repeat the same crash on this psu.

I can try a couple of other power supplies, but I'm not convinced this is the root cause, especially as I haven't seen other power-related issues on this Pi, nor did removing the additional hw (and therefore, lowering the total power draw) affect the issue.

The only other thing I haven't tried swapping is the camera cable - while I have plenty of cables for standard Pis, I don't have many spares for the Zero.  I'll see what I can dig up.

PinkFreud, 2022-05-07:
I haven't had a chance to get back to this until now - this project has been demanding far more of my time than I can really afford to spend on it!

After noticing that new libcamera0 and libcamera-apps-lite were out, I've upgraded to them.  I'm currently running:
`libcamera-apps-lite 0~git20220228+0dc5ea0-1`
`libcamera0:armhf    0~git20220426+18e68a9b-1`

The crash still occurs, though the output is a bit more interesting now.  I've tried with the Zero camera cable I was using, and a spare CM camera adapter (see https://github.com/tback/raspberrypi_documentation/blob/master/hardware/computemodule/cmio-camera.md ) that I had lying around - these adapters are compatible with the Zero connector, as the CM board can use the adapter or a Zero camera cable.

I'll note that I'm now trying this on two RPis - one is my old Zero W, and the other is a Zero 2 W I managed to pick up before the shortage hit.  With the new output from the libcamera apps, I see on the Zero W (where the crash occurs):
```
Preview window unavailable
[0:04:39.807773419] [604]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3548-a11d63f9
[0:04:40.121831230] [605]  WARN RPI raspberrypi.cpp:1237 Mismatch between Unicam and CamHelper for embedded data usage!
[0:04:40.128000858] [605]  INFO RPI raspberrypi.cpp:1352 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[0:04:40.133332538] [604]  INFO Camera camera.cpp:1029 configuring streams: (0) 1920x1080-YUV420
[0:04:40.137956259] [605]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected unicam format: 1920x1080-pRAA
[0:04:40.266752510] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Saturation = 1.000000
[0:04:40.268930379] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Contrast = 1.000000
[0:04:40.271723211] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Brightness = 0.000000
[0:04:40.274493045] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AwbMode = 0
[0:04:40.277206881] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: Sharpness = 1.000000
[0:04:40.280561680] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: ExposureValue = 0.000000
[0:04:40.283129525] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeExposureMode = 0
[0:04:40.284941416] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:04:40.287846241] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: AeMeteringMode = 0
[0:04:40.291057048] [611]  INFO IPARPI raspberrypi.cpp:635 Request ctrl: NoiseReductionMode = 1
[0:34:56.089090512] [605]  WARN V4L2 v4l2_videodevice.cpp:1930 /dev/video0[13:cap]: Dequeue timer of 2476000.00us has expired!
[0:34:56.089540489] [605] ERROR RPI raspberrypi.cpp:1790 Unicam has timed out!
[0:34:56.089740479] [605] ERROR RPI raspberrypi.cpp:1791 Please check that your camera sensor connector is attached securely.
[0:34:56.089930470] [605] ERROR RPI raspberrypi.cpp:1792 Alternatively, try another cable and/or sensor.

Message from syslogd@garage at May  7 13:11:55 ...
 kernel:[ 3143.047411] Internal error: Oops: 80000005 [#1] ARM

Message from syslogd@garage at May  7 13:11:55 ...
 kernel:[ 3143.049247] Process libcamera-vid (pid: 605, stack limit = 0x21dc7fca)
...
```

Interestingly, on the Zero 2 W, running the same package versions and 32-bit userland, I see:
```
...
[0:50:10.010704705] [732]  WARN V4L2 v4l2_videodevice.cpp:1930 /dev/video0[13:cap]: Dequeue timer of 1938000.00us has expired!
[0:50:10.010905038] [732] ERROR RPI raspberrypi.cpp:1790 Unicam has timed out!
[0:50:10.011410454] [732] ERROR RPI raspberrypi.cpp:1791 Please check that your camera sensor connector is attached securely.
[0:50:10.011764345] [732] ERROR RPI raspberrypi.cpp:1792 Alternatively, try another cable and/or sensor.
[0:51:29.074321300] [732]  WARN V4L2 v4l2_videodevice.cpp:1930 /dev/video0[13:cap]: Dequeue timer of 1938000.00us has expired!
[0:51:29.074520242] [732] ERROR RPI raspberrypi.cpp:1790 Unicam has timed out!
[0:51:29.074603430] [732] ERROR RPI raspberrypi.cpp:1791 Please check that your camera sensor connector is attached securely.
[0:51:29.074685679] [732] ERROR RPI raspberrypi.cpp:1792 Alternatively, try another cable and/or sensor.
[2:04:59.941273083] [732]  WARN V4L2 v4l2_videodevice.cpp:1930 /dev/video0[13:cap]: Dequeue timer of 1938000.00us has expired!
[2:04:59.941471482] [732] ERROR RPI raspberrypi.cpp:1790 Unicam has timed out!
[2:04:59.941822767] [732] ERROR RPI raspberrypi.cpp:1791 Please check that your camera sensor connector is attached securely.
[2:04:59.942043922] [732] ERROR RPI raspberrypi.cpp:1792 Alternatively, try another cable and/or sensor.
```

What's most interesting about this output, though, is that despite these warnings on the Z2W, the stream is still running.  Every so often, I see it pause in ZoneMinder, and then it refreshes and continues.  For the stream from the ZW, though, it died with the kernel oops.

I can't definitively say that the timeout warning is tied to the Oops on the single-core ZW, but there may be a correlation here.

PinkFreud, 2022-05-07:
I've followed the instructions on https://forums.raspberrypi.com/viewtopic.php?t=323390 to reinstall the legacy camera stack on this ZW, without making any further hardware swaps.  So far, the old raspivid is streaming without warning or error to ZoneMinder.  I'll leave this running for some time to see if it crashes, and will update this bug report accordingly.

PinkFreud, 2022-05-07:
It's been a couple of hours, and so far, no crash on the ZW with the old MMAL-based camera stack.

The only issue I see in the kernel log (I'm using tcp port 8888 for streaming):
`[  419.241556] TCP: request_sock_TCP: Possible SYN flooding on port 8888. Sending cookies.  Check SNMP counters.`

At this point, I really don't think my hardware is to blame, given that the old camera stack is working on the ZW.  It's interesting that I'm able to reproduce the crash on two different Zero Ws, but other than some timeout complaints on the Zero 2 W, the stream is stable with libcamera-vid.

danieltwagner, 2022-05-07:
I've also had this happen after installing a fresh copy of the newest Raspberry Pi OS on my Pi Zero W.
Taking pictures using `libcamera-jpeg` seems to work fine but streaming video crashes.

Debug info:
```
$ uname -a
Linux pi-camera 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux

$ grep -v "^#" /boot/config.txt | grep .
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
disable_overscan=1
[cm4]
otg_mode=1
[all]
[pi4]
arm_boost=1
[all]

$ vcgencmd version
Mar 24 2022 13:20:54
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

$ libcamera-hello --version
libcamera-apps build: 0dc5ea038c79 04-03-2022 (10:29:30)
libcamera build: v0.0.0+3548-a11d63f9

$ cat /proc/meminfo
MemTotal:         439812 kB
MemFree:          269612 kB
MemAvailable:     310276 kB
Buffers:           12924 kB
Cached:            70204 kB
SwapCached:            0 kB
Active:            46088 kB
Inactive:          54768 kB
Active(anon):        444 kB
Inactive(anon):    19460 kB
Active(file):      45644 kB
Inactive(file):    35308 kB
Unevictable:        1892 kB
Mlocked:            1892 kB
SwapTotal:        102396 kB
SwapFree:         102396 kB
Dirty:               156 kB
Writeback:             0 kB
AnonPages:         19632 kB
Mapped:            31100 kB
Shmem:               668 kB
KReclaimable:       9536 kB
Slab:              19324 kB
SReclaimable:       9536 kB
SUnreclaim:         9788 kB
KernelStack:         776 kB
PageTables:         1040 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      322300 kB
Committed_AS:     131460 kB
VmallocTotal:     573440 kB
VmallocUsed:        5644 kB
VmallocChunk:          0 kB
Percpu:               64 kB
CmaTotal:         262144 kB
CmaFree:          193644 kB
[... 91 lines cut ...]
[  119.383049] ff00: c044560f c378c001 c342ffa4 c342ff18 c021fda0 bf1eb108 00000010 00000000
[  119.383086] ff20: 00000077 c04d67e4 c342ff74 c342ff38 c007f00c c04d67d0 00000077 00000000
[  119.383123] ff40: 16012387 00000000 00000193 434f642b c342e000 b58bfe98 c0c1f028 b6f2a880
[  119.383162] ff60: 00000193 c00083e4 c342ffa4 c342ff78 c0090458 434f642b 00000077 b58c0094
[  119.383202] ff80: b4f17958 b4f0a590 00000036 c00083e4 c342e000 00000000 00000000 c342ffa8
[  119.383238] ffa0: c0008260 c021fc98 b58c0094 b4f17958 0000000f c044560f b58c0094 b6dcc788
[  119.383275] ffc0: b58c0094 b4f17958 b4f0a590 00000036 fffffff8 b58c01cc b6aaa124 01e949c0
[  119.383316] ffe0: b6a7eeec b58c006c b6a0a020 b6b9c1ac 60000010 0000000f 00000000 00000000
[  119.383347] Backtrace:
[  119.383371] [<bf1f37a8>] (video_usercopy [videodev]) from [<bf1f3d6c>] (video_ioctl2+0x1c/0x24 [videodev])
[  119.384250]  r10:0000000f r9:b58c0094 r8:c0c1f028 r7:c378c000 r6:00000000 r5:c378c001
[  119.384286]  r4:bf1f3d50
[  119.384305] [<bf1f3d50>] (video_ioctl2 [videodev]) from [<bf1eb148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[  119.385070] [<bf1eb0fc>] (v4l2_ioctl [videodev]) from [<c021fda0>] (sys_ioctl+0x114/0x9b0)
[  119.385496]  r5:c378c001 r4:c044560f
[  119.385518] [<c021fc8c>] (sys_ioctl) from [<c0008260>] (ret_fast_syscall+0x0/0x1c)
[  119.385566] Exception stack(0xc342ffa8 to 0xc342fff0)
[  119.385599] ffa0:                   b58c0094 b4f17958 0000000f c044560f b58c0094 b6dcc788
[  119.385637] ffc0: b58c0094 b4f17958 b4f0a590 00000036 fffffff8 b58c01cc b6aaa124 01e949c0
[  119.385672] ffe0: b6a7eeec b58c006c b6a0a020 b6b9c1ac
[  119.385703]  r10:00000000 r9:c342e000 r8:c00083e4 r7:00000036 r6:b4f0a590 r5:b4f17958
[  119.385733]  r4:b58c0094
[  119.385764] Code: bad PC value
[  119.385794] ---[ end trace 573ed3b4c982bd9e ]---
````

danieltwagner, 2022-05-07:
It may be worth mentioning that my video doesn't arrive in one piece when I open vlc with `udp/h264://:1234` which I was independently going to look at; VLC has lots of errors about failures to decode, but this may be a red herring.

Example screenshot of what arrives on my target machine. I started VLC before the stream such that it should receive the I-frame. I also see periodic changes in color when I reduce the GOP Interval (e.g. `-g 10`)
[image]

PinkFreud, 2022-05-07:
As an update to my test with the legacy camera stack above, it appears that the crash does indeed occur on the Zero W - but it took several (> 3) hours before it happened.

It appears there's likely an unresolved issue with Bullseye's kernel here, which libcamera may have a higher chance of triggering.

```
[11829.983160] 8<--- cut here ---
[11829.983261] Unable to handle kernel paging request at virtual address e320f000
[11829.983298] pgd = 31a87762
[11829.983332] [e320f000] *pgd=00000000
[11829.983373] Internal error: Oops: 5 [#1] ARM
[11829.983401] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep snd_soc_hdmi_codec spidev hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc vc4 cec brcmfmac brcmutil drm_kms_helper sha256_generic libsha256 snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea raspberrypi_hwmon cfg80211 sysfillrect sysimgblt fb_sys_fops rfkill i2c_bcm2835 spi_bcm2835 snd_bcm2835(C) bcm2835_codec(C) bcm2835_isp(C) snd_pcm v4l2_mem2mem bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig snd_timer videobuf2_memops videobuf2_v4l2 videobuf2_common snd vc_sm_cma(C) videodev mc fixed uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[11829.983850] CPU: 0 PID: 685 Comm: VCHIQ completio Tainted: G         C        5.15.32+ #1538
[11829.983886] Hardware name: BCM2835
[11829.983903] PC is at vector_swi+0x30/0x58
[11829.983947] LR is at 0xb6d6d1ac
[11829.983969] pc : [<c0008330>]    lr : [<b6d6d1ac>]    psr: 20000093
[11829.983991] sp : c2e5dfb0  ip : b6e7205c  fp : 00000000
[11829.984010] r10: e320f000  r9 : b6d6d1ac  r8 : 20000010
[11829.984030] r7 : 00000036  r6 : b6e72358  r5 : 00000001  r4 : b6c94d90
[11829.984051] r3 : 00000008  r2 : b6c94d48  r1 : c014c407  r0 : 00000003
[11829.984075] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[11829.984127] Control: 00c5387d  Table: 03940008  DAC: 00000055
[11829.984151] Register r0 information: non-paged memory
[11829.984185] Register r1 information: non-slab/vmalloc memory
[11829.984220] Register r2 information: non-paged memory
[11829.984247] Register r3 information: non-paged memory
[11829.984273] Register r4 information: non-paged memory
[11829.984297] Register r5 information: non-paged memory
[11829.984322] Register r6 information: non-paged memory
[11829.984347] Register r7 information: non-paged memory
[11829.984372] Register r8 information: non-paged memory
[11829.984397] Register r9 information: non-paged memory
[11829.984422] Register r10 information: non-paged memory
[11829.984449] Register r11 information: NULL pointer
[11829.984475] Register r12 information: non-paged memory
[11829.984501] Process VCHIQ completio (pid: 685, stack limit = 0x0d91a970)
[11829.984535] Stack: (0xc2e5dfb0 to 0xc2e5e000)
[11829.984565] dfa0:                                     00000003 c014c407 b6c94d48 00000008
[11829.984604] dfc0: b6c94d90 00000001 b6e72358 00000036 b6f24b54 b6e72128 b6c94d48 c014c407
[11829.984640] dfe0: b6e7205c b6c94d1c b6e5dd48 b6d6d1ac 20000010 00000003 00000000 00000000
[11829.984669] Backtrace: no frame pointer
[11829.984705] Code: e58d0044 e3a0b000 ee11cf10 e59fa0ac (e59aa000) 
[11829.984735] ---[ end trace 573ed3b4c982bd9e ]---
```

naushir, 2022-05-12:
I'm not really sure what more to suggest here.  I've tried over and over again, but I cannot reproduce this problem.  The log points:

```
[0:34:56.089540489] [605] ERROR RPI raspberrypi.cpp:1790 Unicam has timed out!
[0:34:56.089740479] [605] ERROR RPI raspberrypi.cpp:1791 Please check that your camera sensor connector is attached securely.
[0:34:56.089930470] [605] ERROR RPI raspberrypi.cpp:1792 Alternatively, try another cable and/or sensor.
```
imply that there is a hardware or connectivity issue with the sensor board.  However, this does not explain the actual kernel crash.

Could it be some kind of environmental issue?  How is the device housed?  Is there any electrical activity in the near vicinity of the device?

PinkFreud, 2022-05-15:
> Could it be some kind of environmental issue? How is the device housed? Is there any electrical activity in the near vicinity of the device?

Right now, it's just on a workbench.  The Pi isn't in a case - it's just the bare board, and the hardware stack I've mentioned.  I've also tried a brand new Pi Zero W that I had hanging around in a drawer in a sealed bag - it, too, seems to exhibit the kernel crash.

As this is the same hardware stack I was successfully using on Buster, I suspect if I were to go back to that, it would work without issue.  In fact, as I've invested so much time into this already, I think that's the route I'll be taking.

I plan, tomorrow, to ensure the newly-unpacked Pi Zero W still crashes with both camera stacks (libcamera, and legacy) on Bullseye, and then I'll be rolling the Pi w/ the hardware stack back to Buster so that I can get it reinstalled and functional.  I can continue testing on the standalone ZW with Bullseye, but I really need to get this project running at this point - I'm months behind schedule.

naushir, 2022-05-16:
@PinkFreud sorry for the frustration by the lack of progress - and thank you for all your testing efforts!

Equidamoid, 2022-05-23:
Getting same `Oops` as in https://github.com/raspberrypi/libcamera-apps/issues/246#issuecomment-1120312444 (is it the same as https://github.com/raspberrypi/linux/issues/4223 btw?). The usecase is encoding videostream with ffmpeg (`h264_v4l2m2m`). 
Usually the oops causes the application to crash (every 2-3 hrs), sometimes it locks up the system and requires powercycle to recover. 

Kernel `5.10.92+`. 

Let me know if I can help with testing stuff.

devfuzz, 2022-05-26:
Same Oops.

- Pi Zero W, Pi OS Lite, no X
- Seems to happen less frequently on Pi Zero 2.
- Occurs whether started via SSH or rc.local.
- Sporadic repeatability. (Can run for seconds to hours before Oops.)
- Was running with gpu_mem=128, but have tried others.
- Running with overlay FS enabled might exacerbate the problem. (free -m shows plenty available, however.) 

`$ libcamera-vid --width 1280 --height 720 --saturation 1.0 --gain 25 --awbgains 1.0,1.0 --metering average --segment 60000 --intra 30 --inline -t 0 -o REC%0004d.h264`

```
[ 2441.091854] 8<--- cut here ---
[ 2441.094987] Unable to handle kernel paging request at virtual address bfa96c9c
[ 2441.099580] pgd = d0eefa8d
[ 2441.103760] [bfa96c9c] *pgd=00000000
[ 2441.106628] Internal error: Oops: 80000005 [#1] ARM
[ 2441.109033] Modules linked in: cmac algif_hash aes_arm aes_generic ecb algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc snd_soc_hdmi_codec vc4 cec brcmfmac ov5647 brcmutil drm_kms_helper sha256_generic libsha256 snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect raspberrypi_hwmon sysimgblt fb_sys_fops cfg80211 rfkill bcm2835_unicam v4l2_dv_timings v4l2_fwnode v4l2_async bcm2835_codec(C) snd_bcm2835(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops snd_pcm videobuf2_v4l2 videobuf2_common snd_timer snd videodev mc vc_sm_cma(C) uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 overlay i2c_mux_pinctrl i2c_mux i2c_bcm2835 fixed
[ 2441.132864] CPU: 0 PID: 800 Comm: libcamera-vid Tainted: G         C        5.15.32+ #1538
[ 2441.137928] Hardware name: BCM2835
[ 2441.140477] PC is at 0xbfa96c9c
[ 2441.143018] LR is at video_usercopy+0x4e8/0x5a8 [videodev]
[ 2441.146184] pc : [<bfa96c9c>]    lr : [<bf25ac90>]    psr: 60000013
[ 2441.148740] sp : db917e10  ip : db917e10  fp : db917eec
[ 2441.151243] r10: c0c1f028  r9 : 00000000  r8 : 00000000
[ 2441.153749] r7 : db917e3c  r6 : 00000000  r5 : c044560f  r4 : c050560f
[ 2441.156257] r3 : 00000000  r2 : bf27ca40  r1 : 9adca48b  r0 : c13b42e8
[ 2441.158741] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 2441.161266] Control: 00c5387d  Table: 02ef0008  DAC: 00000055
[ 2441.163764] Register r0 information: slab kmalloc-8k start c13b4000 pointer offset 744 size 8192
[ 2441.168759] Register r1 information: non-paged memory
[ 2441.171296] Register r2 information: 58-page vmalloc region starting at 0xbf252000 allocated at load_module+0xa74/0x25f8
[ 2441.176343] Register r3 information: NULL pointer
[ 2441.178894] Register r4 information: non-slab/vmalloc memory
[ 2441.181411] Register r5 information: non-slab/vmalloc memory
[ 2441.183861] Register r6 information: NULL pointer
[ 2441.186242] Register r7 information: non-slab/vmalloc memory
[ 2441.188586] Register r8 information: NULL pointer
[ 2441.190853] Register r9 information: NULL pointer
[ 2441.193035] Register r10 information: non-slab/vmalloc memory
[ 2441.195169] Register r11 information: non-slab/vmalloc memory
[ 2441.197227] Register r12 information: non-slab/vmalloc memory
[ 2441.199171] Process libcamera-vid (pid: 800, stack limit = 0xa8d5e942)
[ 2441.201106] Stack: (0xdb917e10 to 0xdb918000)
[ 2441.203004] 7e00:                                     db917e3b c04d67d0 c0014720 00000000
[ 2441.206717] 7e20: 00000000 00000000 00000000 00000000 c307c8c0 b595b074 002642d0 00000003
[ 2441.210597] 7e40: 00000002 00278d00 00004002 00000001 00000000 00000988 00000000 000f22bf
[ 2441.214656] 7e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000004 0000001d
[ 2441.218863] 7e80: 00278d00 00000000 00000000 c33f0000 c00526cc 00000000 00000000 9adca48b
[ 2441.223297] 7ea0: c307c1e0 00000008 00000000 00000000 c0c1f028 db917f60 db917f5c 9adca48b
[ 2441.227831] 7ec0: c0209e10 bf25ad50 c307c8c1 00000000 c307c8c0 c0c1f028 b595b074 0000000e
[ 2441.232408] 7ee0: db917efc db917ef0 bf25ad6c bf25a7b4 db917f14 db917f00 bf252148 bf25ad5c
[ 2441.237071] 7f00: c044560f c307c8c1 db917fa4 db917f18 c021fda0 bf252108 00000010 00000000
[ 2441.241855] 7f20: 00000989 c04d67e4 db917f74 db917f38 c007f00c c04d67d0 00000989 00000000
[ 2441.246897] 7f40: 054ae646 00000000 00000193 9adca48b db916000 b595ae78 c0c1f028 b6fd4880
[ 2441.252183] 7f60: 00000193 c00083e4 db917fa4 db917f78 c0090458 9adca48b 00000989 b595b074
[ 2441.257806] 7f80: b50193b0 b500b108 00000036 c00083e4 db916000 00000000 00000000 db917fa8
[ 2441.263720] 7fa0: c0008260 c021fc98 b595b074 b50193b0 0000000e c044560f b595b074 b6e5d788
[ 2441.269890] 7fc0: b595b074 b50193b0 b500b108 00000036 56750a18 00000238 b6b3b124 01314768
[ 2441.276217] 7fe0: b6b0ff18 b595b04c b6a9bda0 b6c2d1ac 60000010 0000000e 00000000 00000000
[ 2441.282609] Backtrace: 
[ 2441.285722] [<bf25a7a8>] (video_usercopy [videodev]) from [<bf25ad6c>] (video_ioctl2+0x1c/0x24 [videodev])
[ 2441.292885]  r10:0000000e r9:b595b074 r8:c0c1f028 r7:c307c8c0 r6:00000000 r5:c307c8c1
[ 2441.296079]  r4:bf25ad50
[ 2441.299162] [<bf25ad50>] (video_ioctl2 [videodev]) from [<bf252148>] (v4l2_ioctl+0x4c/0x64 [videodev])
[ 2441.306133] [<bf2520fc>] (v4l2_ioctl [videodev]) from [<c021fda0>] (sys_ioctl+0x114/0x9b0)
[ 2441.312779]  r5:c307c8c1 r4:c044560f
[ 2441.315805] [<c021fc8c>] (sys_ioctl) from [<c0008260>] (ret_fast_syscall+0x0/0x1c)
[ 2441.318882] Exception stack(0xdb917fa8 to 0xdb917ff0)
[... cut ...]
VmallocUsed:        5620 kB
VmallocChunk:          0 kB
Percpu:               64 kB
CmaTotal:         262144 kB
CmaFree:           98524 kB
```

```
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000d5356511
Model		: Raspberry Pi Zero W Rev 1.1
```

christydennison, 2022-06-11:
I'm also experiencing this issue. I have 32 Raspberry Pi Zero 2 Ws, all running libcamera-still --timelapse for 16 hours daily. About 1/3 of them have this crash every day and they don't come back from reboot and I have to manually power cycle them to get them to come back online. Some run for days without an issue, but others will stop after an hour. kern logs are inconsistently spammed and fill up a 32G microSD. I have code to delete the logs and reboot, but again, they don't usually come back after reboot.

command: 
```
libcamera-still -o /home/pi/stills/ -t $time_length --timelapse $time_interval --timestamp --height 480 --width 640 --immediate --nopreview
```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

```
pi@raspberrypi:~ $ vcgencmd version
Mar 24 2022 13:20:54
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
```

```
pi@raspberrypi:~ $ libcamera-hello
Preview window unavailable
[27:55:38.125769272] [125664]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3544-22656360
[27:55:38.273164935] [125666]  WARN RPI raspberrypi.cpp:1241 Mismatch between Unicam and CamHelper for embedded data usage!
[27:55:38.275693796] [125666]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[27:55:38.277441145] [125664]  INFO Camera camera.cpp:1029 configuring streams: (0) 1640x1232-YUV420
[27:55:38.278386304] [125666]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
```

```
pi@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: 0dc5ea038c79 04-03-2022 (10:35:06)
libcamera build: v0.0.0+3544-22656360
```

```
cat /proc/meminfo
MemTotal:         429552 kB
MemFree:          171676 kB
MemAvailable:     272892 kB
Buffers:           26192 kB
Cached:           107336 kB
SwapCached:         1248 kB
Active:           109120 kB
Inactive:          75228 kB
Active(anon):       2712 kB
Inactive(anon):    48208 kB
Active(file):     106408 kB
Inactive(file):    27020 kB
Unevictable:          16 kB
Mlocked:              16 kB
SwapTotal:        102396 kB
SwapFree:          85368 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:         50448 kB
Mapped:            65808 kB
Shmem:               100 kB
KReclaimable:      25688 kB
Slab:              48340 kB
SReclaimable:      25688 kB
SUnreclaim:        22652 kB
KernelStack:        2160 kB
PageTables:         1884 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      317172 kB
Committed_AS:     200756 kB
VmallocTotal:   259653632 kB
[... 156 lines cut ...]
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507016] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507043] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507070] x11: 0000000000000000 x10: 0000000000000000 x9 : ffffffc008299b30
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507096] x8 : 000000008040000c x7 : ffffff800182f3a0 x6 : 0000007f81440000
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507123] x5 : 0000000000000001 x4 : 0000000000000000 x3 : 0000000000000000
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507150] x2 : 0000000000000000 x1 : ffffff8006dcd010 x0 : ffffff8003df30b0
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507177] Call trace:
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507189]  unlink_anon_vmas+0x9c/0x228
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507206]  free_pgtables+0x8c/0x108
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507224]  exit_mmap+0x98/0x1f0
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507243]  mmput+0x64/0x118
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507262]  do_exit+0x348/0xab8
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507279]  do_group_exit+0x4c/0xa8
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507297]  __wake_up_parent+0x0/0x38
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507315]  invoke_syscall+0x4c/0x110
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507335]  el0_svc_common.constprop.3+0xfc/0x120
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507355]  do_el0_svc+0x2c/0x90
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507373]  el0_svc+0x24/0x60
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507392]  el0t_64_sync_handler+0x90/0xb8
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507411]  el0t_64_sync+0x1a0/0x1a4
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507433] Code: d10042b5 eb14003f 54000320 f9400676 (f94002d7)
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507452] ---[ end trace 87206984f008fa51 ]---
Jun 10 21:46:05 raspberrypi kernel: [ 2154.507469] Fixing recursive fault but reboot is needed!
Jun 11 00:00:02 raspberrypi kernel: [10191.613894] BUG: Bad rss-counter state mm:0000000031c39ea0 type:MM_FILEPAGES val:-1
```

naushir, 2022-06-13:
@christydennison thank you for the report.  Clearly there does seem to be an issue and from the evidence, it seems to be isolated to the Raspberry Pi Zero 2 W platform.  Sadly I still cannot seem to reproduce this.

Are you able to run your program on a Raspberry Pi 3B+ or 4 platform with the exact same sdcard and setup to see if it goes wrong with that as well?

naushir, 2022-06-13:
One thing that perhaps I have not tried replicating is the network activity during my tests.  Would you be able to give a brief description of what sort of networking activity might be going on during your libcamera-vid runs?

naushir, 2022-06-13:
Another suggestion to try would be to add ``overvoltage=2`` to ``/boot/config.txt`` on the devices that seem to go wrong.  There is a suspicion that some boards may need to draw more power in heavy loads.

christydennison, 2022-06-13:
Hi @naushir , I will try the `over_voltage=2` and the same on a Pi4 and get back to you. However, given the form factor, using a Pi4 won't work for my project. Every minute, all of the images taken from the timelapse from the last minute are processed and deleted, and the result is uploaded to a DB. In the meantime, it would be great to know what kind of information you need to debug this. Would having access to one of my boards help, or would having all of the code (setup scripts, services etc) work? If so, do you mind sharing your email/twitter handle/etc or letting me know how else I can contact you?

naushir, 2022-06-14:
@christydennison (and @PinkFreud) sending us one or two of the failing boards (together with the sdcard image you are using + power supply) would be extremely helpful to help debug this issue!  We will, of course, ship replacements over to you as well.

You can find my email in the various git commits on this repo - I'd rather not type it directly into the message to avoid scrapers.  Feel free to send me an email and we can arrange all of the shipping for you.

In the meantime, I have my board here running for approximately 24 hours without failing, so it would be useful if you can tell me the result of using the ``over_voltage`` setting.

naushir, 2022-06-14:
> `libcamera-still -o /home/pi/stills/ -t $time_length --timelapse $time_interval --timestamp --height 480 --width 640 --immediate --nopreview`

Can you provide the values for ``$time_length`` and ``$time_interval`` in your tests?

christydennison, 2022-06-14:
@naushir Sounds great! I just sent you an email. Thank you so much!

```
midnight=`date -d "tomorrow 00:00" +%s`
now=`date +%s`
secs_until_mid=$(($midnight - $now))

time_interval=$(($sec_interval * 1000))
time_length=$(($secs_until_mid * 1000))
```
so `time_interval` is usually 16 hours = 16*60*60*1000 milliseconds, but I recalculate it because I have a different script that restarts the service running this when it hangs/crashes. It's only supposed to run between 8am and midnight UTC.
`time_length` is 4*1000 milliseconds (`sec_interval` = 4).

christydennison, 2022-06-15:
`over_voltage=2` did not seem to fix anything, sorry. Still got 7/30 failures today after rebooting all of them. Should I try a higher number or does that not seem like it would help?

naushir, 2022-06-15:
> `over_voltage=2` did not seem to fix anything, sorry. Still got 7/30 failures today after rebooting all of them. Should I try a higher number or does that not seem like it would help?

A higher number will certainly be worth trying.  You can find more details on the values [here](https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking).  Did you manage to get a failure with a Pi4 board?  Another thing that you may attempt - try disabling all wifi/networking access during a test run to see if that helps.  

One last question, did you have anything plugged in to the USB ports that could be drawing power?

christydennison, 2022-06-15:
I'll try a higher value and get back to you. I have not yet tried on a Pi4 but I just got one from a friend so I'm going to try today.

Nothing is plugged into the USB ports. The only device attached is the camera.

naushir, 2022-06-21:
@Ataman thank you for your report.  I do agree that this does not seem to be anything specific to libcamera itself.  We are unable to reproduce this issue with our devices locally, so are in the process of getting some devices from our users to help reproduce this ourselves.  I will report back once we have an update.

PinkFreud, 2022-06-22:
> Welp, I just tried enabling Legacy Camera support via raspi-config and using the v4l2src plugin with GStreamer. I couldn't get a stream for more than a minute until the same kernel oops occured.
> 
> I was using a GStreamer factory this time with the following launch string:
> 
> ```
> ( v4l2src device=/dev/video0 ! video/x-h264, width=1280, height=720, framerate=30/1 ! h264parse ! rtph264pay name=pay0 pt=96 )
> ```
> 
> It looks like this issue might be older than libcamera?

I was able to reproduce the crash (though, for me, it took several hours of running for it to occur!) with the legacy stack on bullseye in https://github.com/raspberrypi/libcamera-apps/issues/246#issuecomment-1120313797.  The Pi never exhibited the issue on buster, so I suspect the core issue is with one of the kernel modules which talks to the camera hw on the kernel bullseye currently uses.

lauripaas, 2022-06-22:
I'm having the same issue with my RPi Zero W. The setup consists of Pi (using Buster), attached camera, and a mobile 4G router, which the Pi uses to stream video to Youtube. And I'm powering everything with a 12V marine battery.
I won't post the same error dumps here, as many have already done, but someone mentioned interference being a possible reason. Well, I had my setup working for 9 months in remote wooded areas, with no crashes whatsoever.

But when I put up the same setup next to my house in the city (I hadn't changed anything after bringing everything out from the woods, no updates to Buster even), then the crashes start at random times.

I hadn't had the time to set it up in the woods again to test the stability, but maybe this is something to think about...

Ataman, 2022-06-22:
Considering that [lauripaas](https://github.com/lauripaas) was able to reproduce the same issue with Buster, are there other alternatives than using libcamera/v4l2 or are projects using the Zero W pretty much busted now?

naushir, 2022-06-22:
We do not believe this goes wrong on every Zero W board - we have several boards in our test setup trying to reproduce this and they have been running for days now.  I have requested some users to send us their defective boards so we can more easily reproduce this to find a fix.

Additionally, given that @lauripaas has managed to make it fail in the legacy stack - it implies this is likely not related to libcamera/unicam at all.

In the meantime, are you able to use a Pi 3b+ or 4 for your usage?  If not, it might be worth trying to add ``over_voltage=4`` to your ``/boot/config.txt`` file to see if that helps at all.  This is based on the suspicion that some boards may need to draw more power under higher cpu load, and fail to do so.

Ataman, 2022-06-22:
Sadly, the Zero W is the only option due to space constraints in my case. Since several sources already suspect an issue with the 5.x kernels, I'll be going back to a 4.19 kernel & legacy camera. Probably shouldn't have tried running the latest software on older hardware in the first place.

naushir, 2022-06-23:
@[christydennison](https://github.com/christydennison), looking back at your logs, you seem to be using a 64-bit kernel.  This is not necessarily the cause of the issue since other reporters are using the 32-bit kernel.  Would you be able to switch one or two of your devices to use the 32-bit kernel to see the outcome?

naushir, 2022-06-23:
> @naushir I'll try that and get back to you!
> 
> Also, as you expected, I haven't been able to get a failure out of the RPi4 yet, but it has only been running for a couple of days. I tried 2, 4, and 6 for `over_voltage` for the entire cluster and saw about a 5% (30% -> 25%) reduction in failures for 6.
> 
> I switched up my code a bit and now am doing the timelapse for 10 minutes, then doing the processing after the timelapse has stopped (before I was doing the processing every minute and the timelapse was still going on in the background) and that has reduced the failures from 30% to 10%, so I think that trying to run anything at the same time as the timelapse is part of the issue.

That's interesting... both results so lend evidence to a possible issue with the board being undervolted, particularly on higher CPU loads.  It might be worth pushing up your ``over_voltage`` value even more (**disclaimer, this could damage your board if you go too high**) to see the results.  I might have a go at undervolting my board to "simulate" this and see if I can reproduce the same error condition.

christydennison, 2022-06-23:
@naushir I don't think I'll wager a board given how hard they are to get 😅 but let me know how the lower voltage experiments go.

I realized I didn't share this traceback from the kernel logs before. There are a bunch of these "__swap_info_get: Bad swap file entry" blocks before the "get_swap_device: Bad swap file entry" is spammed and fills up the logs.
```
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900580] __swap_info_get: Bad swap file entry 1400000000b51513
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900614] BUG: Bad page map in process libcamera-still  pte:b5151316 pmd:80000000708d003
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900654] addr:0000007faaa00000 vm_flags:040444fb anon_vma:0000000000000000 mapping:ffffff800925e8f8 index:5da
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900686] file:dmabuf fault:0x0 mmap:dma_buf_mmap_internal readpage:0x0
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900731] CPU: 1 PID: 9833 Comm: libcamera-still Tainted: G    B    C        5.15.32-v8+ #1538
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900742] Hardware name: Raspberry Pi Zero 2 W Rev 1.0 (DT)
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900749] Call trace:
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900753]  dump_backtrace+0x0/0x1b8
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900767]  show_stack+0x20/0x30
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900779]  dump_stack_lvl+0x8c/0xb8
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900789]  dump_stack+0x18/0x34
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900797]  print_bad_pte+0x168/0x1c0
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900808]  unmap_page_range+0x208/0x740
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900817]  unmap_single_vma+0x44/0x50
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900826]  unmap_vmas+0xe4/0x108
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900835]  unmap_region+0xbc/0x118
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900846]  __do_munmap+0x1b4/0x4e8
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900856]  __vm_munmap+0x88/0x138
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900866]  __arm64_sys_munmap+0x3c/0x50
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900877]  invoke_syscall+0x4c/0x110
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900888]  el0_svc_common.constprop.3+0xfc/0x120
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900899]  do_el0_svc+0x2c/0x90
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900909]  el0_svc+0x24/0x60
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900919]  el0t_64_sync_handler+0x90/0xb8
Jun 23 20:45:56 raspberrypi kernel: [ 5730.900929]  el0t_64_sync+0x1a0/0x1a4
Jun 23 20:49:50 raspberrypi kernel: [ 5965.587647] BUG: Bad rss-counter state mm:000000000f48182b type:MM_FILEPAGES val:-1
Jun 23 20:49:50 raspberrypi kernel: [ 5965.587711] BUG: Bad rss-counter state mm:000000000f48182b type:MM_SWAPENTS val:-1
Jun 23 20:50:01 raspberrypi kernel: [ 5975.982889] get_swap_device: Bad swap file entry 1400000000a31415
```

naushir, 2022-06-28:
>```
>  kernel:[ 5585.689400] Internal error: Oops: 80000005 [#1] ARM
>  kernel:[ 5585.895104] Process libcamera-still (pid: 1625, stack limit = 0xf572f199)
>  kernel:[ 5585.901913] Stack: (0xc1207e10 to 0xc1208000)
>  kernel:[ 5585.906291] 7e00:                                     c1207e3b c188d7c0 c1207e5c 00000000
>  kernel:[ 5585.914493] 7e20: 00000000 00000000 00000000 00000000 c37d1a00 b595c094 00c9c038 00000003
>  kernel:[ 5585.922694] 7e40: 00000001 002eec00 00004002 00000000 00000000 000015d1 00000000 0008d355
>  kernel:[ 5585.930893] 7e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000004 00000014
>  kernel:[ 5585.939095] 7e80: 002ef000 00000000 00000000 c1207e98 c0046c80 c000b100 c3bec014 c188d7c0
>  kernel:[ 5585.947297] 7ea0: c0c2b2c0 c1888000 c30138c0 c30138c0 c1207edc c1207ec0 c0013798 ef99baaa
>  kernel:[ 5585.955499] 7ec0: 00000000 bf1e8d50 c37d1a01 00000000 c37d1a00 c0c1f028 b595c094 0000000e
>  kernel:[ 5585.963701] 7ee0: c1207efc c1207ef0 bf1e8d6c bf1e87b4 c1207f14 c1207f00 bf1e0148 bf1e8d5c
>  kernel:[ 5585.971903] 7f00: c044560f c37d1a01 c1207fa4 c1207f18 c021fda0 bf1e0108 00000010 00000000
>  kernel:[ 5585.980105] 7f20: 000015d1 c04d67e4 c1207f74 c1207f38 c007f00c c04d67d0 000015d1 00000000
>  kernel:[ 5585.988306] 7f40: 257aa2c8 00000000 00000193 ef99baaa c1206000 b595be98 c0c1f028 b6f79880
>  kernel:[ 5585.996508] 7f60: 00000193 c00083e4 c1207fa4 c1207f78 c0090458 ef99baaa 000015d1 b595c094
>  kernel:[ 5586.004708] 7f80: b50180d8 b5009b38 00000036 c00083e4 c1206000 00000000 00000000 c1207fa8
>  kernel:[ 5586.012908] 7fa0: c0008260 c021fc98 b595c094 b50180d8 0000000e c044560f b595c094 b6e4c788
>  kernel:[ 5586.021111] 7fc0: b595c094 b50180d8 b5009b38 00000036 fffffff8 b595c1cc b6b2a124 011b76a8
>  kernel:[ 5586.029312] 7fe0: b6afef18 b595c06c b6a8ada0 b6c1c1ac 60000010 0000000e 00000000 00000000
>  kernel:[ 5586.037505] Backtrace:
>  kernel:[ 5586.039964] [<bf1e87a8>] (video_usercopy [videodev]) from [<bf1e8d6c>] (video_ioctl2+0x1c/0x24 [videodev])
>  kernel:[ 5586.050436]  r10:0000000e r9:b595c094 r8:c0c1f028 r7:c37d1a00 r6:00000000 r5:c37d1a01
>  kernel:[ 5586.058287]  r4:bf1e8d50
>  kernel:[ 5586.060829] [<bf1e8d50>] (video_ioctl2 [videodev]) from [<bf1e0148>] (v4l2_ioctl+0x4c/0x64 [videodev])
>  kernel:[ 5586.070885] [<bf1e00fc>] (v4l2_ioctl [videodev]) from [<c021fda0>] (sys_ioctl+0x114/0x9b0)
>  kernel:[ 5586.079568]  r5:c37d1a01 r4:c044560f
>  kernel:[ 5586.083156] [<c021fc8c>] (sys_ioctl) from [<c0008260>] (ret_fast_syscall+0x0/0x1c)
>  kernel:[ 5586.090764] Exception stack(0xc1207fa8 to 0xc1207ff0)
>  kernel:[ 5586.095838] 7fa0:                   b595c094 b50180d8 0000000e c044560f b595c094 b6e4c788
>  kernel:[ 5586.104041] 7fc0: b595c094 b50180d8 b5009b38 00000036 fffffff8 b595c1cc b6b2a124 011b76a8
>  kernel:[ 5586.112241] 7fe0: b6afef18 b595c06c b6a8ada0 b6c1c1ac
>  kernel:[ 5586.117312]  r10:00000000 r9:c1206000 r8:c00083e4 r7:00000036 r6:b5009b38 r5:b50180d8
>  kernel:[ 5586.125157]  r4:b595c094
>  kernel:[ 5586.127705] Code: bad PC value
> ```

I have also reproduced this exact kernel call stack on my Pi Zero platform - with ``over_voltage=-2`` parameter set (i.e. undervolting my board).  It would be useful if folks who have the crash with the above callstack try to set ``over_voltage=2`` in ``/boot/config.txt`` and see if it solves their problem.  You might also want to try setting the value to 4 or 6 if 2 does not work.

Note that if the issue is down to the power supply itself, ``over_voltage=x`` may not actually fix things.

danieltwagner, 2022-06-28:
I have access to a single Pi Zero only at the moment but `over_voltage=2` seems to have it running stable for a few minutes now, compared to erroring out within seconds before. `vcgencmd measure_volts` reports 1.4v now but I guess we're implying that maybe some of the boards can't be trusted to provide stable voltages?

naushir, 2022-06-28:
> I have access to a single Pi Zero only at the moment but `over_voltage=2` seems to have it running stable for a few minutes now, compared to erroring out within seconds before. `vcgencmd [...]

That's slightly encouraging!  I'm not sure if it's the boards themselves that fail to regulate a stable voltage, or the power supplies that fail to provide a stable voltage.  I'll talk to some hardware folks to get some more details.

danieltwagner, 2022-06-28:
When I have physical access to an affected board again I can run it off a bench supply to ensure stable voltage to help diagnose. Are there any markings or board revisions that we should start to collect?

naushir, 2022-06-28:
The device serial number obtained from ``cat /proc/cpuinfo`` ought to be all that's needed to start with.

christydennison, 2022-06-29:
Now that I've been running the new code (where libcamera runs by itself and nothing runs at the same time as it) for a few days, the failures have gone from 10% -> 3% -> 0% (before reorganization, 30% daily failures). I'm not sure if they needed a few days to adjust or what is going on. I also have a script to reboot if upload failures start happening (usually because of hours of clock drift; unfortunately, trying to reset the clock manually doesn't seem to always work without a reboot), but the boards are managing to come back up after reboot, unlike before.

For anyone following this thread, it might be worth trying to reorganize your code to
1) have libcamera run by itself (wait until libcamera is not running to do other tasks)
2) start libcamera 30 seconds after boot (if you're using a systemd service, you can use `ExecStartPre=/bin/sleep 30`)
3) set `over_voltage=4`
4) reboot if upload failures occur

Ataman, 2022-06-30:
Like others already have pointed out, this issue affects older 4.19 Kernels as well. I've done some testing with the [2020-05-27-raspios-buster-lite-armhf](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-05-28/) image and still get the same kernel failures using legacy camera drivers.

However, the following boot configs resulted in a stable video stream over 24 hours with the above 32-bit image.
- ```over_voltage=2```
- ```arm_freq=600```

The following GStreamer-Pipeline streamed for over 24 hours without issues:
```rpicamsrc ! video/x-h264, width=1920, height=1080, framerate=30/1 ! h264parse config-interval=1 ! rtph264pay name=pay0 pt=96```

Reducing the clock means losing processing power of course. The above uses about 85% of the CPU but that's fine for just streaming HD video.

CTRL+F: Workaround

[18 more replies]

closed 2024-02-29 by naushir