raspberrypi/rpicam-apps #248: [BUG] libcamera-vid adding "-t ##" causes tcp server to disconnect

jlpoolen, 2022-02-11:
Server: Running on a RPiZero2W:

     date; camera-bug-report -c "libcamera-vid  -t 30 -n --inline --listen -o tcp://192.168.1.12:5000"
     
Client: Running on a Dell Inspiron (over wifie) running Gentoo:

     date; ffplay tcp://192.168.1.12:5000  -vf "setpts=N/30" -fflags nobuffer -flags low_delay -framedrop -loglevel debug ; date;
     
I launch server and it waits for a connection. 
I launch client and it connections and then connection is terminated with:

    Invalid data found when processing input
    
Attached Server log: 1495 lines in bug-report_server_20220210202326.txt
[bug-report_server_20220210202326.txt](https://github.com/raspberrypi/libcamera-apps/files/8046000/bug-report_server_20220210202326.txt)

Attached Client log: 28 lines in client_taurus_ffmpeg_debug.txt
[client_taurus_ffmpeg_debug.txt](https://github.com/raspberrypi/libcamera-apps/files/8046001/client_taurus_ffmpeg_debug.txt)

If I omit the "-t 30", the client successfully connects and the stream successfully works until I terminate either client or server.  Note: i tried to get ffplay to work for a limited duration, but for some reason that was ignored and I have to Ctrl-C to terminate this client process.  As far as I can tell, I'm using the most verbose output for ffplay.  Please let me know if there is anything I can add to the command line.  This seems like a signal is being sent from one and not interpreted by the other.

jlpoolen, 2022-02-11:
update: 9:04 PM 2/10/2022 PST
I then tried to run the server with no time limit and client with no time limit and now the server closes down after a client connection.

Here's my server command line:

      clear;date; camera-bug-report;libcamera-vid   -n --inline --listen -o tcp://192.168.1.12:5000;date

and bug report as file bug-report_server_no_wrap_202202102102.txt

Client ffmepg log as file: client_taurus_ffmpeg_202202102102.txt

I'm wondering if someone at a low level, perhaps disclosed in demesg is afoot.
[client_taurus_ffmpeg_202202102102.txt](https://github.com/raspberrypi/libcamera-apps/files/8046075/client_taurus_ffmpeg_202202102102.txt)
[bug-report_server_no_wrap_202202102102.txt](https://github.com/raspberrypi/libcamera-apps/files/8046076/bug-report_server_no_wrap_202202102102.txt)
[server_run_console_log_202202102102.txt](https://github.com/raspberrypi/libcamera-apps/files/8046078/server_run_console_log_202202102102.txt)

naushir, 2022-02-11:
> ``-t 30``

The units for this parameter are in milliseconds.  This is why the ``libcamera-vid`` application shut down near instantaneously after the command was run.  If you are looking to run for 30s, you need ``-t 30000000``.

davidplowman, 2022-02-13:
Hi, thanks for the suggestion. I think we'd be happy for it to output something like that when libcamera-vid exits, Please be sure to run the style checker (in utils/checkstyle.py) before submitting a pull request. Thanks!

jlpoolen, 2022-02-17:
If I do so, it will be in the next two weekends.  Daytime job really drains me.  Thank you.

closed 2022-02-17 by jlpoolen