raspberrypi/rpicam-apps #249: [BUG] Getting ERROR "Camera sensor does not support test pattern modes"

EricClaeys, 2022-02-12:
**Describe the bug**
When I run libcamera-*  I get:
`[5:33:40.849786275] [10225] ERROR CameraSensor camera_sensor.cpp:551 'imx477 10-001a': Camera sensor does not support test pattern modes.`

**Bug report**
[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/8053024/bug-report.txt)

**Additional context**
I updated the libcamera apps today and got some errors so updated the firmware.  The camera now works despite the ERROR message above.  libcamera-still is used by Allsky software and the ERROR message is confusing to users.

davidplowman, 2022-02-12:
Hi, this error message is coming from libcamera. I agree it seems rather over-zealous to flag this as an error when it is in fact totally fine Perhaps we should open a discussion on that.

EricClaeys, 2022-02-12:
@davidplowman Perhaps make it an "INFO" instead of "ERROR"?  Our program ignores INFO and WARN messages so as not to confuse users.  The word "ERROR" raises a red flag.
Thanks for getting back quickly.

Question: We'd ideally like to not use the libcamera-still command but instead us the library libcamera-still uses, so we can open the camera once when our program starts, then take pictures for days, then close the camera when our program exits, how would we do that?  We do that with ZWO cameras which have a Linux library to open, close, take pictures, query the camera, etc.  Is there something similar for the Pi cameras?

davidplowman, 2022-02-12:
Re. the "ERROR", I'll propose a change there, and we can see where that takes us.

For your other question, you could use the libcamera-apps code as a "library" and incorporate it into your own code, but maybe that's more complicated than you want. Another idea would be to start libcamera-still in "signal" mode and leave it running. Then you send it a SIGUSR1 whenever you want to take a picture. In this scenario the camera is running the whole time, though there's no terribly easy way to "query" it. Finally, perhaps the most interesting idea might be to wait for the "early preview" of Picamera2 ("real soon now"). This would give you access to pretty much everything that libcamera can do, and you'd be able to script whatever behaviours you want.

EricClaeys, 2022-02-12:
@davidplowman Thanks for your support.  I really appreciate it.

A nice thing about ZWO's library is the ability to query things like exposure time and gain used in auto mode, sensor temperature (one of our developers said he planned to look at a kernel driver to get the temp), camera capabilities (resolution, bin modes supported, ...).  We currently only support the Pi HD camera, although I'm sure at some point in the future there will be other Pi models and I'd like to be able to support them with minimal work.  For ZWO, there's basically no additional work since we can query the camera's properties.

Where would I go to find out more about Picamera2 ?

kbingham, 2022-02-12:
Aha, I think this is a bit subtle, that the CameraSensor class is trying to initialise the test pattern mode to be 'off' to ensure a consistent state, and isn't checking to see if there is a test pattern mode supported in the first place.

It's currently an error, because the function shouldn't expect to be called if it's not possible. I'll send a patch now.

davidplowman, 2022-02-12:
@kbingham Hi Kieran, thanks for taking a look, much appreciated!

@EricClaeys Picamera2 is in development, but we're hoping to make a "preview release" shortly for folks to have a look at. Keep watching, news soon I hope!

closed 2022-02-21 by naushir

kbingham, 2022-02-21:
> I'll close this down as the change is under review in libcamera: https://patchwork.libcamera.org/patch/15362/

Reviews welcome ;-)

naushir, 2022-02-21:
> Reviews welcome ;-)

Done :)