raspberrypi/rpicam-apps #251: Segfault - running Valgrind on libcamera-still

fire-eggs, 2022-02-14:
**Describe the bug**

I'm running valgrind against a locally built, debug copy of libcamera-still. The code is from this repo at this [commit](https://github.com/raspberrypi/libcamera-apps/commit/2d1009e3badcc8047361ff81149ad6cba3b911b5).

Every fresh invocation (i.e. immediately after a build) of `valgrind` + `libcamera-still` gives me a segfault. Usually on the second attempt, the segfault does not occur.

Here is the code context from libcamera_app.cpp, as the line numbers reported by valgrind may be off due to my workaround attempt below. `LibcameraApp::previewThread()`
```
FrameBuffer *buffer = item.completed_request->buffers[item.stream];
if (!buffer) continue;             // my attempt to work around the segfault with no luck
libcamera::Span span = Mmap(buffer)[0];  // segfault occurs on this line
```

Any change to prevent or workaround the segfault would be appreciated.

**Bug report**

[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/8063611/bug.txt)

**Additional context**

Valgrind command line:
```
valgrind --trace-children=yes --leak-check=full --track-origins=yes --show-leak-kinds=all --log-file=still8.txt ./libcamera-still -t 60000 --timelapse=10000 --timestamp 2> still8.log
```

And the segfault output from valgrind:
```
==1895== Invalid read of size 8
==1895==    at 0x496A618: LibcameraApp::previewThread() (libcamera_app.cpp:791)
==1895==    by 0x49960EB: void std::__invoke_impl<void, void (LibcameraApp::*)(), LibcameraApp*>(std::__invoke_memfun_deref,
 void (LibcameraApp::*&&)(), LibcameraApp*&&) (invoke.h:73)
==1895==    by 0x4995717: std::__invoke_result<void (LibcameraApp::*)(), LibcameraApp*>::type std::__invoke<void (LibcameraA
pp::*)(), LibcameraApp*>(void (LibcameraApp::*&&)(), LibcameraApp*&&) (invoke.h:95)
==1895==    by 0x4994A27: void std::thread::_Invoker<std::tuple<void (LibcameraApp::*)(), LibcameraApp*> >::_M_invoke<0ul, 1
ul>(std::_Index_tuple<0ul, 1ul>) (thread:264)
==1895==    by 0x4993E3F: std::thread::_Invoker<std::tuple<void (LibcameraApp::*)(), LibcameraApp*> >::operator()() (thread:
271)
==1895==    by 0x49915FB: std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (LibcameraApp::*)(), LibcameraApp*>
 > >::_M_run() (thread:215)
==1895==    by 0x4BA0CAB: ??? (in /usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28)
==1895==    by 0x4E51627: start_thread (pthread_create.c:477)
==1895==    by 0x4DA901B: thread_start (clone.S:78)
==1895==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
...
==1895== Process terminating with default action of signal 11 (SIGSEGV)
==1895==  Access not within mapped region at address 0x8
==1895==    at 0x496A618: LibcameraApp::previewThread() (libcamera_app.cpp:791)
==1895==    by 0x49960EB: void std::__invoke_impl<void, void (LibcameraApp::*)(), LibcameraApp*>(std::__invoke_memfun_deref,
 void (LibcameraApp::*&&)(), LibcameraApp*&&) (invoke.h:73)
==1895==    by 0x4995717: std::__invoke_result<void (LibcameraApp::*)(), LibcameraApp*>::type std::__invoke<void (LibcameraA
pp::*)(), LibcameraApp*>(void (LibcameraApp::*&&)(), LibcameraApp*&&) (invoke.h:95)
==1895==    by 0x4994A27: void std::thread::_Invoker<std::tuple<void (LibcameraApp::*)(), LibcameraApp*> >::_M_invoke<0ul, 1
ul>(std::_Index_tuple<0ul, 1ul>) (thread:264)
==1895==    by 0x4993E3F: std::thread::_Invoker<std::tuple<void (LibcameraApp::*)(), LibcameraApp*> >::operator()() (thread:
271)
==1895==    by 0x49915FB: std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (LibcameraApp::*)(), LibcameraApp*>
 > >::_M_run() (thread:215)
==1895==    by 0x4BA0CAB: ??? (in /usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.28)
==1895==    by 0x4E51627: start_thread (pthread_create.c:477)
==1895==    by 0x4DA901B: thread_start (clone.S:78)
```

davidplowman, 2022-02-21:
Hi, thanks for reporting this. I've found it a pain to reproduce but have managed to see it a few times. Anyway, I think I can guess what the problem might be, and I've submitted a possible fix (https://github.com/raspberrypi/libcamera-apps/pull/255). Please feel free to try it if you're in a position to do so, otherwise we'll get round to merging it shortly if no problems show up. Thanks!

fire-eggs, 2022-02-23:
Thank you for the fix! Sorry to hear it was a pain - most memory leaks are!

The long-term timelapse test just finished as I was composing this: Success! AFAIC this issue can be closed!

naushir, 2022-02-23:
Thanks for confirming the fix! Closing this issue now.

closed 2022-02-23 by naushir