raspberrypi/rpicam-apps #254: Cannot save image but can save it via ssh

FlashGordon3140, 2022-02-20:
I have a Pi4 8Gb with an HQ camera which is freshly installed with bulleye 64 bits version. When I issue on console a command like

Libcamera-jpeg -o test.jpg

I see the preview, but nothing is saved and got a segmentation fault. This is with -hello or -still but with -raw it is working.  Got the following output

libcamera-still -o test.jpeg
Made X/EGL preview window
[7:27:30.404813112] [270605]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
[7:27:30.431437328] [270606]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[7:27:30.431503977] [270606]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[7:27:30.431537978] [270606] ERROR CameraSensor camera_sensor.cpp:551 'imx477 10-001a': Camera sensor does not support test pattern modes.
[7:27:30.455088523] [270606]  INFO RPI raspberrypi.cpp:1326 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[7:27:30.456139481] [270605]  INFO Camera camera.cpp:1028 configuring streams: (0) 2028x1520-YUV420
[7:27:30.456612580] [270606]  INFO RPI raspberrypi.cpp:747 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[7:27:30.519573424] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[7:27:30.519683166] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[7:27:30.519808575] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[7:27:30.519839742] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[7:27:30.519868372] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[7:27:30.519897928] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[7:27:30.519947985] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[7:27:30.519976596] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[7:27:30.520046004] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[7:27:30.520076635] [270611]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[7:27:37.008048879] [270605]  INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[7:27:37.008399217] [270606]  INFO RPI raspberrypi.cpp:747 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Segmentation fault

However, when I connect to the Pi via ssh, all is ok?!!? Really do not know what is going on!!!!

Version installed
Linux MotionEye 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux

libcamera-hello --version
libcamera-apps build: 7ac5197216ea 01-02-2022 (14:53:46)
libcamera build: v0.0.0+3406-e96d0201

davidplowman, 2022-02-20:
HI, thanks for the bug report. Are you able to use gdb to capture a stack trace of the crash? That might be helpful. In any case I see a couple of crash reports have appeared while I've been away so I'll try to look at them as soon as I can.

In the meantime one might guess that the crash could be to do with the preview window. Maybe try adding the `-n` ("no preview") option and see if the crash disappears. Obviously that's not ideal because you won't see a preview, but it would be interesting to know.

FlashGordon3140, 2022-02-20:
bingo! -n is working! I guess explain why with ssh or Libcamera-raw were working because on ssh or raw you do not have a preview!

FlashGordon3140, 2022-02-20:
just a remark. I noticed that sometimes you get one working without -n, but if you repeat after you got the bug  all the time. If you wait a couple of hour, again work once and then bugging....

davidplowman, 2022-03-06:
Hi, we've pusted a new version to our apt repo that fixes a segfault. Would you be able to give that a try? Thanks.

FlashGordon3140, 2022-03-08:
Sorry but do not understand what I shall do? Update my system will download the new version and implement it? Thanks if you can tell me what I need to do.

I will be happy to give a try but I need to access the PI and I am traveling currently. So be patient. The reason I need to physically access the PI is because I used an external PoE splitter so without the need to use a PoE HAT on the Pi itself.  That’s work fine. To test I will need to revert to the old setup.

Thanks for your assistance to that matter!

Kind regards,

=-= Jacques Masson =-=

davidplowman, 2022-03-08:
Yes, if you do a system update (`sudo apt update` and `sudo apt upgrade`) that should fetch all the new stuff.

FlashGordon3140, 2022-03-08:
👍😎

naushir, 2022-03-18:
I'll close this one down now.  If you are still having problems with the updated code, feel free to re-open.

closed 2022-03-18 by naushir

FlashGordon3140, 2022-03-18:
👍

=-= Jacques =-=

Proudly sent from my iPad Pro 11!