raspberrypi/rpicam-apps #257: [BUG] ERROR: *** no cameras available ***

bezineb5, 2022-02-21:
**Describe the bug**
I'm trying to use a camera with Raspberry Pi Zero 2W and the new stack doesn't work on Bullseye.
What I found works:
- Bullseye (64 bits) on a Pi 4 with the same camera
- Buster or bullseye with the old stack on the Pi 0 2W, same camera

What I tried with no success:
- Using 2 different Pi Zero 2W
- Using bullseye 32 and 64 bits
- Updating the kernel using rpi-update (that broke the wifi, so I couldn't try the camera)
- Using the pi camera 1.3 and 2.

So I'm out of hope to make it work, it seems that libcamera doesn't work on bullseye+ 
pi Zero 2W.

Any idea what could be wrong?

**Bug report**
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 21-02-2022 (22:07:59)
Command: /usr/bin/camera-bug-report -t 5 -o bug.txt -c libcamera-still -t 1000 -o test.jpg

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
piboat3

** cat /proc/cpuinfo **
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
[... 654 lines cut ...]
** libcamera-still --version **
libcamera-apps build: 7ac5197216ea 01-02-2022 (14:49:18)
libcamera build: v0.0.0+3406-e96d0201

--------------------------------------------------------------------------------
libcamera-still -t 1000 -o test.jpg
--------------------------------------------------------------------------------
Preview window unavailable
[0:05:09.580737248] [1090]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
ERROR: *** no cameras available ***
```

**Additional context**
```
 $ i2cdetect -y 10
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

6by9, 2022-02-22:
It works fine on an _unmodified_ Bullseye image.

```gpu_mem=16``` in config.txt switches to the cutdown firmware, so removes all codecs, ISP, and memory sharing functionality, all of which libcamera relies on.

bezineb5, 2022-02-22:
Thanks a lot!
I was not aware of that. It would be nice to have that in a "troubleshooting" section in the doc, maybe?

I confirm it works, thanks a lot!

closed 2022-02-22 by bezineb5

kbingham, 2022-02-22:
> It works fine on an _unmodified_ Bullseye image.
> 
> `gpu_mem=16` in config.txt switches to the cutdown firmware, so removes all codecs, ISP, and memory sharing functionality, all of which libcamera relies on.

@6by9 I've had a few people ask me what they should now set for gpu_mem. Is there a specific minimum or recommended setting now that buffers are allocated on the CPU side? I always see wildly different values there perhaps where people change it ad-hoc.

bezineb5, 2022-02-22:
> @6by9 I've had a few people ask me what they should now set for gpu_mem. Is there a specific minimum or recommended setting now that buffers are allocated on the CPU side? I always see wildly [...]

@6by9 In my specific case, I'm using the Pi0 headless, I'm just using GPU memory for DMA-based PWM (like in periph.io or pi-blaster). So it was kind of a no-brainer to reduce to the minimum. I'd be interested to know how much memory is required for libcamera, aas @kbingham mentionned.

6by9, 2022-02-22:
IIRC Raspberry Pi OS has left the default gpu_mem alone as things like the video codecs still require it for internal buffer pools.

If just wanting to run libcamera without H264 or MJPEG codecs, then gpu_mem=32 is probably sufficient.
If you want to use the video codecs, then leaving it at the default (64MB on Pi0-1, and 80MB on Pi2-4) would be recommended.
```sudo vcdbg reloc``` will tell you how much of the gpu_mem heap is in use at any instant in time.

bezineb5, 2022-02-22:
> 

Thanks a lot!