raspberrypi/rpicam-apps #258: pkg-config files required by build not in libcamera source

tewarid, 2022-02-22:
I'm trying to get libcamera-apps to build with libcamera for a custom linux distro. cmake uses pkg-config to resolve libcamera and libcamera-base dependency but I couldn't find pkg-config (.pc) files in the libcamera repo. They are distributed on Raspbian but I can't find their source. Any help is highly appreciated.

marv, 2022-02-22:
Hey @tewarid,
libcamera uses Meson's `pkgconfig` module[1] to generate its pkg-config files[2][3]

[1] https://mesonbuild.com/Pkgconfig-module.html
[2] https://git.linuxtv.org/libcamera.git/tree/src/libcamera/base/meson.build#n67
[3] https://git.linuxtv.org/libcamera.git/tree/src/libcamera/meson.build#n163

tewarid, 2022-02-22:
Thanks for the info! I'll close this issue and investigate meson.

closed 2022-02-22 by tewarid