raspberrypi/rpicam-apps #259: libcamera-still hangs on futex

EricClaeys, 2022-02-24:
Running libcamera-apps on Buster.  Have a C++ program that calls `libcamera-still` via system() every X seconds to obtain a picture from RPiHQ camera and save it.  The C++ program waits for libcamera-still to finish so there is only one occurrence running.  The Pi is dedicated to the Allsky application so nothing else is running on it.

Occasionally the libcamera-still process hangs.  Running "strace" on the pid give:
~~~
strace: Process 9262 attached
futex(0xbed9418c, FUTEX_WAIT_PRIVATE, 12, NULL
~~~

Another hang gave:
~~~
futex(0xbefa9b88, FUTEX_WAIT_PRIVATE, 0, NULL
~~~

The last process to hang had this command line:
~~~
libcamera-still --output '/home/pi/allsky/tmp/image-20220222003208.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 25398416 --gain 1 --exif IFD0.Artist=li_25398416_1 --awb auto --saturation 1 --brightness 0 --quality 95 2> /dev/null
~~~
The only things that change between most shots is the shutter and gain values.
I ran `sudo apt update` and `sudo apt upgrade` and libcamera-apps did not need updating.

~~~
pi:/var/www/html/allsky $ libcamera-still --version
libcamera-apps build: 7ac5197216ea 01-02-2022 (14:39:20)
libcamera build: v0.0.0+3406-e96d0201
~~~

Let me know if there's any debugging I can do to help find the issue.

davidplowman, 2022-02-24:
Hi, thanks for the report. I haven't managed to reproduce this myself yet. Do you know if it occurs if you run the libcamera-still alone from the command line (which is what I've been trying)?

If you can get it to hang perhaps you could run it under (or attach) gdb, and report the output of `thread apply all bt`.

The other thing would be to set the environment variable `export LIBCAMERA_LOG_LEVELS=RPI:0` and capture the output to a log file (rather than `/dev/null`). That might reveal what the code is waiting for. Thanks!

EricClaeys, 2022-02-24:
@davidplowman It can sometimes go days without any problem so I am not surprised you haven't reproduced it yet.
I have never tried to reproduce by running via the command line since I am also testing our Allsky software.  I assume you have it in a simple shell loop?

Thanks for the suggestions - I will try them.  I currently have `LIBCAMERA_LOG_LEVELS=ERROR,WARNING` to decrease the tons of output that are added to our log file since it only confuses users.  But for my own testing I can easily change it.

I will also keep track of the shutter ang gain settings when it hangs in case they are related.

EricClaeys, 2022-02-24:
@davidplowman It hung again this morning with `--shutter 1000 --gain 1`.
`gdb --pid=21457` output:
~~~
Thread 2 (Thread 0xaf9b6200 (LWP 21475)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbed6e1a4) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbed6e178) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbed6e178, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb5a09c94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6f0a3cc in PostProcessor::outputThread() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#5  0xb5a0f9b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0xb6eb2494 in start_thread (arg=0xaf9b6200) at pthread_create.c:486
#7  0xb584c568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 0xb6f8f540 (LWP 21455)):
#0  futex_wait (private=0, expected=12, futex_word=0xbed6e19c) at ../sysdeps/unix/sysv/linux/futex-internal.h:61
#1  futex_wait_simple (private=0, expected=12, futex_word=0xbed6e19c) at ../sysdeps/nptl/futex-internal.h:135
#2  __pthread_cond_destroy (cond=0xbed6e178) at pthread_cond_destroy.c:54
#3  0xb5a09c80 in std::condition_variable::~condition_variable() () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6f0913c in PostProcessor::~PostProcessor() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#5  0xb6ef602c in LibcameraApp::~LibcameraApp() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#6  0x0001e3a4 in ?? ()
#7  0xb578c718 in __libc_start_main (main=0xbed6e3c4, argc=-1249120256, argv=0xb578c718 <__libc_start_main+268>, init=<optimized out>, fini=0x2c04c, rtld_fini=0xb6f7f4f0 <_dl_fini>, stack_end=0xbed6e3c4)
    at libc-start.c:308
#8  0x0001e568 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

naushir, 2022-02-25:
@davidplowman, should we be calling PostProcessor::Stop() in the PostProcessor desctructor? Although, I do see it called from LibcameraApp::StopCamera(), so I assume applications must/will always call this on shutdown?

The funny thing is for this example we have no postprocessing stages running (according to the command line), so the postprocessing thread should be completely idle.

davidplowman, 2022-02-25:
I think we have to call PostProcessor::Stop() when we stop the camera because otherwise it could still asynchronously dump completed requests onto the message queue. If we never started the camera, presumably there's no problem, and calling PostProcessor::Stop() again in the destructor would force us to add some "am I stopped already?" test.

Looking at the stack traces, it looks like the main thread is destroying the LibcameraApp object, which in turn is destroying the PostProcessor object. The PostProcessor is stuck trying to clear up the condition variable presumably because its output thread is still waiting on it.

So how can we get to the end of LibcameraApp::~LibcameraApp(), which has called LibcameraApp::StopCamera() which in turn would have called PostProcessor::Stop(), which waits for and joins the output thread? That seems to be the mystery, right?

I'm trying to imagine if there's away we could be in the middle of the LibcameraApp::requestComplete() callback already when the camera is stopped. Could we end up calling PostProcessor::Process() again during or after PostProcessor::Stop() has happened? But still the output thread would have disappeared (or we'd be locked up in PostProcessor::Stop()), so that doesn't really make sense. Added to which the output thread never gets used in this example as there's no post processing going on...

davidplowman, 2022-02-25:
Another thought. Maybe the problem is really something completely different, for example the camera failing to start. In that case camera_started_ would remain false even though the PostProcessor was started. That would cause exactly this symptom. So in that case we'd either have to stop the post processor if the camera fails to start, or add the "stop if I haven't been" logic into PostProcessor::~PostProcessor().

davidplowman, 2022-02-25:
Or we could just call PostProcessor::Start() *after* camera_started_ is set. Looks to me like that would work too.

EricClaeys, 2022-02-25:
@davidplowman, @naushir,
if it matters, this hang has been occurring around once a day for the last week or so.  During the day it takes about 2 pictures per minute, and at night it takes about one 40-second picture per minute, for a total of around 2200 pictures per day.
The delay between day pictures is 30 seconds, and between night pictures is 10 seconds.  The only thing that changes per picture is the shutter and gain.

The CPU time on the few hangs I remember is 0 seconds per `ps` on my Pi 4B.  I don't know what a typical CPU time is for a normal libcamera-still.

naushir, 2022-02-25:
> Or we could just call PostProcessor::Start() _after_ camera_started_ is set. Looks to me like that would work too.

I suppose there is no harm in doing this even if it may not solve the underlying problem.

davidplowman, 2022-02-25:
`LibcameraApp::~LibcameraApp()` calls `LibcameraApp::StopCamera()`, so I don't think this is a problem.

davidplowman, 2022-02-25:
I've made that particular little change here: https://github.com/raspberrypi/libcamera-apps/pull/261

But as noted previously, if it fixes anything it will leave us with a different problem (the real one, at least). Or it may be a red herring, though at least it rules out one possibility.

EricClaeys, 2022-02-25:
@naushir, @davidplowman,
It hung again this morning.  If you let me know how to run it with gdb from my program, I will do that.  I don't see an option in gdb to have a program run automatically.   I use system() in the c++ program.

EricClaeys, 2022-02-25:
The stdout/stderr from this morning's hang was identical to what I get on success, other than the exposure time and timestamps.
Below is gdb output from this morning's hang.  A lot more threads than last time.
~~~
Attaching to process 2884
[New LWP 2887]
[New LWP 2888]
[New LWP 2889]
[New LWP 2893]
[New LWP 2894]
[New LWP 2895]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
futex_wait_cancelable (private=0, expected=0, futex_word=0xbed2fb68)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
88	../sysdeps/unix/sysv/linux/futex-internal.h: No such file or directory.
(gdb) thread apply all bt

Thread 7 (Thread 0xaf12b200 (LWP 2895)):
#0  futex_wait (private=<optimized out>, expected=265657, futex_word=0xb2ca14d8)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:61
#1  futex_wait_simple (private=<optimized out>, expected=265657, futex_word=0xb2ca14d8)
    at ../sysdeps/nptl/futex-internal.h:135
#2  __condvar_quiesce_and_switch_g1 (private=0, g1index=<synthetic pointer>, wseq=<optimized out>, 
    cond=0xb2ca14c8) at pthread_cond_common.c:416
#3  __pthread_cond_broadcast (cond=0xb2ca14c8) at pthread_cond_broadcast.c:73
#4  0xb598ecbc in std::condition_variable::notify_all() ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#5  0xb5c61a10 in libcamera::Thread::dispatchMessages(libcamera::Message::Type) ()
   from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#6  0xb5c58968 in libcamera::EventDispatcherPoll::processEvents() ()
   from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#7  0xb5c614d8 in libcamera::Thread::exec() () from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#8  0xb59949b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#9  0xb6e37494 in start_thread (arg=0xaf12b200) at pthread_create.c:486
#10 0xb57d1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 6 (Thread 0xaf92c200 (LWP 2894)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbed30184)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
--Type <RET> for more, q to quit, c to continue without paging--
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbed30158) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbed30158, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb598ec94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6e8f3cc in PostProcessor::outputThread() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#5  0xb59949b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0xb6e37494 in start_thread (arg=0xaf92c200) at pthread_create.c:486
#7  0xb57d1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 5 (Thread 0xb012d200 (LWP 2893)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbed30064)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbed30038) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbed30038, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb598ec94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6e79fb0 in LibcameraApp::previewThread() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#5  0xb59949b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0xb6e37494 in start_thread (arg=0xb012d200) at pthread_create.c:486
#7  0xb57d1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 4 (Thread 0xb1afe200 (LWP 2889)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xb2363b3c)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
[... 47 lines cut ...]
#13 0xb59949b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#14 0xb6e37494 in start_thread (arg=0xb2ca2200) at pthread_create.c:486
#15 0xb57d1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 0xb6f14540 (LWP 2884)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbed2fb68)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbed2fb40) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbed2fb40, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb598ec94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb5c5f884 in libcamera::Semaphore::acquire(unsigned int) ()
   from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#5  0xb5c5782c in libcamera::BoundMethodBase::activatePack(std::shared_ptr<libcamera::BoundMethodPackBase>, bool) () from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#6  0xb5cebb38 in libcamera::Camera::stop() () from /lib/arm-linux-gnueabihf/libcamera.so.0
#7  0xb6e7963c in LibcameraApp::StopCamera() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#8  0x0001dec4 in ?? ()
#9  0xb5711718 in __libc_start_main (main=0xbed303a4, argc=-1249624064, 
    argv=0xb5711718 <__libc_start_main+268>, init=<optimized out>, fini=0x2c04c, 
    rtld_fini=0xb6f044f0 <_dl_fini>, stack_end=0xbed303a4) at libc-start.c:308
#10 0x0001e568 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
~~~

davidplowman, 2022-02-25:
Is this running exactly the same code as the previous shorter stack traces that you posted?

EricClaeys, 2022-02-25:
This libcamera-still was called with `LIBCAMERA_LOG_LEVELS=RPI:0` per your suggestion, and with "2> /tmp/debug.txt" rather than "2>/dev/null".  I had to recompile the C++ program to enable those changes.
No other changes.

davidplowman, 2022-02-25:
Does it contain the change in that pull request (https://github.com/raspberrypi/libcamera-apps/pull/261)?

EricClaeys, 2022-02-25:
No.  I last did an apt update (?) a couple weeks ago and it installed the latest version.  I don't have a private copy.
If you'd like to me to use a newer version with PR #261, how do I get it and install it?

davidplowman, 2022-02-25:
No, I don't actually think it would make any difference in this case. Looking at it, it's actually a bit less "weird" in some respects. It seems to have got stuck trying to stop the camera so there are lots of other threads still sat around which, given that it hasn't stopped properly, is kind of normal. It seems like the pipeline handler is stuck, not clear to me why, though.

EricClaeys, 2022-02-26:
Another hang this afternoon; gdb output below.
I'm on vacation on the coast of Florida.  The camera is outside on the balcony, covered by a baggie kept in place by a rubber band.  The roughly 1 meter ribbon cable sticks out of the baggie and goes under a chair, into a plastic container with the lid partially open, but facing backwards, and connects to the Pi inside.  There has been some heavy fog so it's possible there's moisture in the baggie and/or in the plastic container with the Pi.  Is there any indication in the gdb output that this could be caused by a hardware issue?

~~~
Attaching to process 1026
[New LWP 1029]
[New LWP 1030]
[New LWP 1031]
[New LWP 1047]
[New LWP 1048]
[New LWP 1049]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
futex_wait_cancelable (private=0, expected=0, futex_word=0xbe851b68)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
88	../sysdeps/unix/sysv/linux/futex-internal.h: No such file or directory.
(gdb) thread apply all bt

Thread 7 (Thread 0xaf12b200 (LWP 1049)):
#0  futex_wait (private=<optimized out>, expected=265657, futex_word=0xb2c814d8)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:61
#1  futex_wait_simple (private=<optimized out>, expected=265657, futex_word=0xb2c814d8)
    at ../sysdeps/nptl/futex-internal.h:135
#2  __condvar_quiesce_and_switch_g1 (private=0, g1index=<synthetic pointer>, wseq=<optimized out>, 
    cond=0xb2c814c8) at pthread_cond_common.c:416
#3  __pthread_cond_broadcast (cond=0xb2c814c8) at pthread_cond_broadcast.c:73
#4  0xb596ecbc in std::condition_variable::notify_all() ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#5  0xb5c41a10 in libcamera::Thread::dispatchMessages(libcamera::Message::Type) ()
   from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#6  0xb5c38968 in libcamera::EventDispatcherPoll::processEvents() ()
   from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#7  0xb5c414d8 in libcamera::Thread::exec() () from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#8  0xb59749b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#9  0xb6e17494 in start_thread (arg=0xaf12b200) at pthread_create.c:486
#10 0xb57b1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 6 (Thread 0xaf92c200 (LWP 1048)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbe852184)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbe852158) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbe852158, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb596ec94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6e6f3cc in PostProcessor::outputThread() () from /lib/arm-linux-gnueabihf/libcamera_app.so
--Type <RET> for more, q to quit, c to continue without paging--
#5  0xb59749b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0xb6e17494 in start_thread (arg=0xaf92c200) at pthread_create.c:486
#7  0xb57b1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 5 (Thread 0xb012d200 (LWP 1047)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbe852064)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbe852038) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbe852038, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb596ec94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6e59fb0 in LibcameraApp::previewThread() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#5  0xb59749b0 in ?? () from /lib/arm-linux-gnueabihf/libstdc++.so.6
#6  0xb6e17494 in start_thread (arg=0xb012d200) at pthread_create.c:486
#7  0xb57b1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 4 (Thread 0xb1afe200 (LWP 1031)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xb2363b3c)
[... 49 lines cut ...]
#14 0xb6e17494 in start_thread (arg=0xb2c82200) at pthread_create.c:486
#15 0xb57b1568 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Thread 1 (Thread 0xb6ef4540 (LWP 1026)):
#0  futex_wait_cancelable (private=0, expected=0, futex_word=0xbe851b68)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
#1  __pthread_cond_wait_common (abstime=0x0, mutex=0x0, cond=0xbe851b40) at pthread_cond_wait.c:502
#2  __pthread_cond_wait (cond=0xbe851b40, mutex=0x0) at pthread_cond_wait.c:655
#3  0xb596ec94 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb5c3f884 in libcamera::Semaphore::acquire(unsigned int) ()
   from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#5  0xb5c3782c in libcamera::BoundMethodBase::activatePack(std::shared_ptr<libcamera::BoundMethodPackBase>, bool) () from /lib/arm-linux-gnueabihf/libcamera-base.so.0
#6  0xb5ccbb38 in libcamera::Camera::stop() () from /lib/arm-linux-gnueabihf/libcamera.so.0
#7  0xb6e5963c in LibcameraApp::StopCamera() () from /lib/arm-linux-gnueabihf/libcamera_app.so
#8  0x0001dec4 in ?? ()
#9  0xb56f1718 in __libc_start_main (main=0xbe8523a4, argc=-1249755136, 
    argv=0xb56f1718 <__libc_start_main+268>, init=<optimized out>, fini=0x2c04c, 
    rtld_fini=0xb6ee44f0 <_dl_fini>, stack_end=0xbe8523a4) at libc-start.c:308
#10 0x0001e568 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
~~~

davidplowman, 2022-03-01:
So that looks like the previous stack trace, I think. I'm starting to wonder if the original one was anomalous somehow.

I also noticed you said the following:

> The roughly 1 meter ribbon cable sticks out of the baggie and goes under a chair, into a plastic container with the lid partially 

This is the ribbon cable between the Pi and the camera board? These are not particularly reliable over longer distances, and I am really very suspicious of it. Is is possible for you to try something shorter? I myself would never be very confident of anything over about 30cm, though as noted, faults are intermittent so longer ones can work quite well... until randomly they don't.

You also mention moisture which I guess may be an issue too - it sounds a bit like some trials with more carefully controlled hardware and conditions might be needed.

EricClaeys, 2022-03-02:
Yes, the ribbon cable between the Pi and camera board. I replaced the 1 meter cable with one that's about half that length.  I also brought the camera inside since we're returning home tomorrow.

Did the gdb output provide enough information to know how to fix the issue?

davidplowman, 2022-03-02:
I think the first thing to do is to find out whether we have a problem when using a short (less than 30cm) cable, perhaps starting in an indoor environment. Currently I'm failing to reproduce the problem, in spite of running tens of thousands of back-to-back captures. I'm running the following script (you may want to edit it to make it more like your own use-case):
```
#!bin/bash
n=0
while true; do
    echo "RUN $n"
    libcamera-still -n -t 1000 --shutter 1000 --gain 1 -o test.jpg >& log.txt
    ((n=n+1))
done
```
If you still see the same problem, we might want to consider building debug versions of code to see if we can get more precise info from the stacktrace.

6by9, 2022-03-02:
Long cables and corruption brings https://github.com/raspberrypi/linux/pull/4873 to mind, where repeated frame start with no frame end (eg because it was corrupted on a long cable?) resulted in dropped buffers within Unicam.
No buffers could lead to something waiting on an event that will never happen.

EricClaeys, 2022-03-03:
After I brought the Pi indoors and used the 1/2 meter cable, it hung a few hours later.  I'm on the road for the next couple days so won't have it plugged in until I get home.

@davidplowman   If you let me know how to run libcamera-still under gdb and have it auto start, I'll change my script to do so.  I assume that'll give you much more information about how libcamera-still got to where it hangs?

EricClaeys, 2022-03-04:
@davidplowman.
* I'll try with the shortest cable I have.  If it makes any difference, I got a about 10 different-sized cables from Amazon.  I'll see if I can find the original cable.  The reason I got a longer cable is so the camera can be under a dome in one housing and the Pi in a separate box.  Both will be outside.
* I'll try when home.
* Yes.

EricClaeys, 2022-03-15:
@davidplowman When at home I ran the shell scripts for a while and had no problems.  Also no problems using my Allsky application.
A few days ago I updated the Pi and (I think) it updated the firmware and libcamera, but near the end of the update it gave a timeout.  When I booted it was getting a kernel fault during boot.  I replaced /boot with a new /boot, but now I can't get libcamera to work.  Raspistill works fine.  For libcamera I commented out `camera_auto_detect=1` in /boot/config.txt and added 
~~~
dtoverlay=imx477
start_x=1
~~~
to the first "[all]" section.

Is this the expected output on Buster with an IMX477?
~~~
pi:~ $ vcgencmd get_camera
supported=1 detected=0
~~~
Without the `start_x=1`, vcgencmd get_camera` returns `supported=0 detected=0`

This is the error I get.  **What entries should be in /boot/config.txt to get the HQ camera working on the current Buster?**
~~~
pi:~ $ libcamera-hello
Made X/EGL preview window
[0:01:08.152921569] [1327]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
[0:01:08.168319755] [1328]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[0:01:08.168422403] [1328]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:01:08.168474903] [1328] ERROR CameraSensor camera_sensor.cpp:551 'imx477 10-001a': Camera sensor does not support test pattern modes.
[0:01:08.188380736] [1328] ERROR RPI raspberrypi.cpp:1230 Unicam driver does not use the MediaController, please update your kernel!
[0:01:08.189064903] [1328] ERROR RPI raspberrypi.cpp:1129 Failed to register camera imx477 10-001a: -22
ERROR: *** no cameras available ***
~~~

davidplowman, 2022-03-15:
Hi, things seem to have got quite messed up there. In particular that `Unicam driver does not use the MediaController` message suggests you've somehow got a really old kernel. You probably need to re-try that update, or failing that maybe `sudo rpi-update` will put a usable kernel over the one you currently have?

Just to clarify a couple of things. When using libcamera, you don't need `start_x=1` in your /boot/config.txt. You shouldn't need `dtoverlay=imx477` if you have `camera_auto_detect=1`, though it's possible that if you've wound up with a really old kernel it might make a difference (though still not work).

At the end of your message you refer to Buster. Do you mean Buster or should that be Bullseye? I would avoid trying to use libcamera on Buster as there's no development or even maintenance going on there, that's all happening on Bullseye.

naushir, 2022-03-29:
@EricClaeys any updates on this issue?  Not sure if you are following https://github.com/raspberrypi/libcamera-apps/issues/271, but it is a similar issue to what you are seeing, and we suspect it is a possible fault with the sensor module that causes the stall.

naushir, 2022-03-30:
@EricClaeys thanks for the update.  I'm inclined to say your original problem was likely a hardware fault with your ov5647.  As such I suggest we close down this issue now and re-open if you think this may still be a software fault.

I have also pushed a change to libcamera that will through a log message if it detects that the sensor has stalled so that we can identify such problems more easily.

closed 2022-03-30 by naushir

naushir, 2022-03-31:
> @naushir Will it just output a message when stalled, or will it also exit the program?

It will just output the message on the console for now.  There is no way that I know if to feed the error back from the pipeline handler to the application through libcamera.  I will talk to the libcamera folks to see what may be possible here.

> 
> What does LIBCAMERA_LOG_LEVELS need to be set to in order to see the message?

This is an Error log message, so you do not need to set any log levels for it to appear on the console.

kbingham, 2022-03-31:
Sounds like that would be a signal on the camera that the PH could raise to let the application listen to events. Not sure what those events would be yet though - or whether this needs to be 'generic'. We might need to have one signal with a parameter, or one signal per event... certainly an interesting discussion topic for the list.

naushir, 2022-03-31:
Yes, I was thinking along the same lines as well.