raspberrypi/rpicam-apps #260: Save precapture in split videos captured via keypress or signal

tewarid, 2022-02-24:
I'm looking into using `--keypress` and/or `--signal` options to trigger video recording on demand, e.g.

```bash
libcamera-vid -o split%04d.h264 --inline --split --initial pause --keypress -t 0 --width 640 --height 480
```

I'm looking into adding a few seconds/frames of precapture in each video. Is this something I can do with an existing option?

davidplowman, 2022-02-25:
Hi, I'm not quite sure what your use case is, can you perhaps explain a bit more? I think I'm understanding that you want to start libcamera-vid and record several seconds, then you want it to pause the recording automatically, after which you want to be able to resume/pause the recording when there's a keypress (or signal). Is that correct?

tewarid, 2022-02-25:
Hi. Precapture is common in motion sensing applications where you want some video footage prior to the motion detection event e.g. https://motion-project.github.io/motion_config.html#pre_capture.

davidplowman, 2022-02-25:
Thanks for the explanation!

So I suppose there's always the obvious workaround, namely to start `libcamera-vid` in "record" mode but to send it a keypress/signal after a few seconds. Alternatively I could imagine changing the `--initial` option to accept a number (probably in milliseconds for consistency with the `--timeout` option) that would be a time for which it records initially before pausing automatically. I guess you might want ignore keypresses until the application has paused for the first time.

Just wondering if you're a C/C++ programmer? We'd be open to pull requests for a generally useful feature enhancement like that.

tewarid, 2022-02-25:
I do program in C++ occasionally - I am not yet familiar with libcamera and libcamera-apps, but could give it a try.

I'm not sure what the command line should look like, but have some thoughts on the implementation. I imagine using a precapture circular buffer that continuously stores camera payload. Its size is determined by the precapture time/frame count. Its content is written to file at the point start is signaled via keypress or signal. I think it does not make sense to have precapture unless you're using split file output.

davidplowman, 2022-02-25:
Ah I see. So every time you detect a motion event you want "a few seconds" prior to that? How are you detecting that, is that some independent motion sensor, and how do you know how long to record for? That sounds a bit more complicated.

Maybe there's a solution where you start libcamera-vid in circular buffer mode and leave it running. Then you wait for a motion "on" event and subsequently for the motion "off" event, at which point you terminate libcamera-vid. This gives you a video file, at which point you immediately restart libcamera-vid (waiting for the next motion event). Of course, your video length will presumably always be the length of the circular buffer, but maybe that's OK?

tewarid, 2022-02-25:
> Ah I see. So every time you detect a motion event you want "a few seconds" prior to that? How are you detecting that, is that some independent motion sensor, and how do you know how long to record for? That sounds a bit more complicated.
> 

Correct - start/stop trigger is done using an independent motion sensor.

> Maybe there's a solution where you start libcamera-vid in circular buffer mode and leave it running. Then you wait for a motion "on" event and subsequently for the motion "off" event, at which point [...]

I've looked into using circular buffer option, but need fine grained control over duration of precapture, and start/stop trigger.

naushir, 2022-02-26:
Perhaps this kind of functionality warrants a new app forked from libcamera-vid and using the existing circular buffer mechanism?

tewarid, 2022-02-26:
Forking libcamera-vid sounds like a good approach. I'll contribute back if circumstance allows. Thank you!

closed 2022-02-26 by tewarid