raspberrypi/rpicam-apps #262: [BUG]

jlpoolen, 2022-02-25:
**Describe the bug**
libcamera-apps build fails to build on Raspberry Pi Zero 2 W (Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux) with  /usr/local/src/libcamera-apps/preview/egl_preview.cpp:18:10: fatal error: libdrm/drm_fourcc.h: No such file or directory
       18 | #include <libdrm/drm_fourcc.h>

My operating system (downloaded pre-built image from Q Engineering similar to their new GitHub [site](https://github.com/Qengineering/RPi_64-bit_Zero-2-image) -- _thank you_, folks at Q Engineering!):

    pi@raspberrypi:/usr/local/src/libcamera-apps $ cat /etc/os-release
    PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
    NAME="Debian GNU/Linux"
    VERSION_ID="11"
    VERSION="11 (bullseye)"
    VERSION_CODENAME=bullseye
    ID=debian
    HOME_URL="https://www.debian.org/"
    SUPPORT_URL="https://www.debian.org/support"
    BUG_REPORT_URL="https://bugs.debian.org/"
    pi@raspberrypi:/usr/local/src/libcamera-apps $

Should pre-compile tests look for library/header files libdrm?  

For the benefit of other would-be compilers on the Pi Zero 2 running Bullseye: before I could commence the build, I had installed :

     exif
     libexif-dev 
     libodb-boost-2.4
     libboost-system1.74.0 
     libboost-thread-dev 
     libboost-program-options-dev

**Bug report**
[bug-report_20220225_0821.txt](https://github.com/raspberrypi/libcamera-apps/files/8142693/bug-report_20220225_0821.txt)

Add any other context about the problem here.
Build commenced around 7:53 p.m. and error-out at 10:58 p.m., hence 3 hours, with 44% completion

Here's the console:

    pi@raspberrypi:/usr/local/src/libcamera-apps $ make
    Scanning dependencies of target images
    [  2%] Building CXX object image/CMakeFiles/images.dir/bmp.cpp.o
    [  4%] Building CXX object image/CMakeFiles/images.dir/yuv.cpp.o
    [  6%] Building CXX object image/CMakeFiles/images.dir/jpeg.cpp.o
    [  8%] Building CXX object image/CMakeFiles/images.dir/png.cpp.o
    [ 10%] Building CXX object image/CMakeFiles/images.dir/dng.cpp.o
    [ 12%] Linking CXX shared library libimages.so
    [ 12%] Built target images
    Scanning dependencies of target post_processing_stages
    [ 14%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/post_processing_stage.cpp              .o
    [ 17%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/negate_stage.cpp.o
    [ 19%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/hdr_stage.cpp.o
    date
    [ 21%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/pwl.cpp.o
    [ 23%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/histogram.cpp.o
    [ 25%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/motion_detect_stage.cpp.o
    [ 27%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/sobel_cv_stage.cpp.o
    [ 29%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/face_detect_cv_stage.cpp.              o
    [ 31%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/annotate_cv_stage.cpp.o
    [ 34%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/plot_pose_cv_stage.cpp.o
    [ 36%] Building CXX object post_processing_stages/CMakeFiles/post_processing_stages.dir/object_detect_draw_cv_sta              ge.cpp.o
    [ 38%] Linking CXX shared library libpost_processing_stages.so
    [ 38%] Built target post_processing_stages
    Scanning dependencies of target VersionCpp
    Generating version string: 56721d65cfe1-intree 24-02-2022 (20:56:09)
    [ 38%] Built target VersionCpp
    Scanning dependencies of target preview
    [ 40%] Building CXX object preview/CMakeFiles/preview.dir/null_preview.cpp.o
    [ 42%] Building CXX object preview/CMakeFiles/preview.dir/preview.cpp.o
    [ 44%] Building CXX object preview/CMakeFiles/preview.dir/egl_preview.cpp.o
    /usr/local/src/libcamera-apps/preview/egl_preview.cpp:18:10: fatal error: libdrm/drm_fourcc.h: No such file or di              rectory
       18 | #include <libdrm/drm_fourcc.h>
          |          ^~~~~~~~~~~~~~~~~~~~~
    compilation terminated.
    make[2]: *** [preview/CMakeFiles/preview.dir/build.make:108: preview/CMakeFiles/preview.dir/egl_preview.cpp.o] Er              ror 1
    make[1]: *** [CMakeFiles/Makefile2:407: preview/CMakeFiles/preview.dir/all] Error 2
    make: *** [Makefile:171: all] Error 2
    pi@raspberrypi:/usr/local/src/libcamera-apps $ date
    Thu 24 Feb 20:58:21 PST 2022
    pi@raspberrypi:/usr/local/src/libcamera-apps $  git status
    On branch main
    Your branch is up to date with 'origin/main'.

    Untracked files:
      (use "git add <file>..." to include in what will be committed)
       CMakeCache.txt
       CMakeFiles/
       CPackConfig.cmake
       CPackSourceConfig.cmake
       Makefile
       apps/CMakeFiles/
       apps/Makefile
       apps/cmake_install.cmake
       cmake_install.cmake
       compile_commands.json
       core/CMakeFiles/
       core/Makefile
       core/cmake_install.cmake
       core/version.cpp
       encoder/CMakeFiles/
       encoder/Makefile
       encoder/cmake_install.cmake
       image/CMakeFiles/
       image/Makefile
       image/cmake_install.cmake
       image/libimages.so
       output/CMakeFiles/
       output/Makefile
       output/cmake_install.cmake
       post_processing_stages/CMakeFiles/
       post_processing_stages/Makefile
       post_processing_stages/cmake_install.cmake
       post_processing_stages/libpost_processing_stages.so
       preview/CMakeFiles/
       preview/Makefile
       preview/cmake_install.cmake
       utils/CMakeFiles/
       utils/Makefile
       utils/cmake_install.cmake

    nothing added to commit but untracked files present (use "git add" to track)
    pi@raspberrypi:/usr/local/src/libcamera-apps $ date
    Fri 25 Feb 08:15:37 PST 2022
    pi@raspberrypi:/usr/local/src/libcamera-apps $ pi@raspberrypi:/usr/local/src/libcamera-apps $ git log
    commit 56721d65cfe157056bb239a960c6fbd09f7cff0d (HEAD -> main, origin/main, origin/HEAD)
    Author: Devendra Tewari <devendra.tewari@gmail.com>
    Date:   Mon Feb 21 19:42:08 2022 -0300

        Include sstream header

        Fix for fatal error: implicit instantiation of undefined
        template 'std::basic_stringstream<char>'

davidplowman, 2022-02-25:
Sounds like maybe no one installed `libdrm-dev`? (as per https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-apps)

Alternatively you could use `-DENABLE_DRM=0` when you run `cmake`.

jlpoolen, 2022-02-25:
Oops.  I was unaware of the Build instructions as David cited in the comment preceding this.  I missed the link under "here" in the README.txt.

closed 2022-02-25 by jlpoolen