raspberrypi/rpicam-apps #266: [BUG] libcamera-still generating Segmentation fault - works fine with time delay "-t 1000" but not w/o

bayetan, 2022-02-27:
**Describe the bug**
"libcamera-still -o test.jpg" used to generate an output file results in a "Segmentation fault" and no output file.  Using the same command with "-t 1000 -o test.jpg" option does NOT produce the "Segmentation fault" and generate the output file.  So not having a time delay appears to cause the error.

**Bug report**

```

[bug-report.txt](https://github.com/raspberrypi/libcamera-apps/files/8148012/bug-report.txt)

davidplowman, 2022-02-28:
Hi, thanks for reporting this. It certainly seems a bit hard to understand, might you be able to run it under gdb and get a stack trace? Also, I don't see the console output in the bug report, could you maybe post that too? Thanks.

castorf, 2022-03-01:
I've seen this , too.  If I turn off the preview frame, there is no problem.
pi@pi64:~ $ cat cam.bt
gdb) run -o bar.jpg
Starting program: /usr/bin/libcamera-still -o bar.jpg
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Made X/EGL preview window
[44:31:50.849602781] [15273]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
[New Thread 0x7feb4ecec0 (LWP 15276)]
[44:31:50.867458115] [15276]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx477'
[44:31:50.867529232] [15276]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[44:31:50.867553585] [15276] ERROR CameraSensor camera_sensor.cpp:551 'imx477 10-001a': Camera sensor does not support test pattern modes.
[New Thread 0x7feac6aec0 (LWP 15277)]
[New Thread 0x7fea469ec0 (LWP 15278)]
[44:31:50.929100464] [15276]  INFO RPI raspberrypi.cpp:1326 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media2
[44:31:50.930055904] [15273]  INFO Camera camera.cpp:1028 configuring streams: (0) 2028x1520-YUV420
[44:31:50.930481750] [15276]  INFO RPI raspberrypi.cpp:747 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[New Thread 0x7fe8a98ec0 (LWP 15279)]
[New Thread 0x7fe3ffeec0 (LWP 15280)]
[New Thread 0x7fe37fdec0 (LWP 15281)]
[44:31:50.970899946] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[44:31:50.970980081] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[44:31:50.971005027] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[44:31:50.971028825] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[44:31:50.971051957] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[44:31:50.971075810] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[44:31:50.971114220] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[44:31:50.971136092] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[44:31:50.971163465] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[44:31:50.971184559] [15281]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
[Thread 0x7fe37fdec0 (LWP 15281) exited]
[Thread 0x7fe3ffeec0 (LWP 15280) exited]
[Thread 0x7fe8a98ec0 (LWP 15279) exited]
[44:31:57.006733682] [15273]  INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-YUV420 (1) 4056x3040-SBGGR12_CSI2P
[44:31:57.007022888] [15276]  INFO RPI raspberrypi.cpp:747 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[New Thread 0x7fe8a98ec0 (LWP 15282)]
[New Thread 0x7fe3ffeec0 (LWP 15283)]

Thread 8 "libcamera-still" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fe8a98ec0 (LWP 15282)]
0x0000007ff7f79a74 in LibcameraApp::previewThread() () from /lib/aarch64-linux-gnu/libcamera_app.so
(gdb) bt
#0  0x0000007ff7f79a74 in LibcameraApp::previewThread() () from /lib/aarch64-linux-gnu/libcamera_app.so
#1  0x0000007ff7db2cac in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#2  0x0000007ff7b26628 in start_thread (arg=0x7fe8a987c0) at pthread_create.c:477
#3  0x0000007ff7c2501c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
(gdb)

pi@pi64:~ $ dpkg -l | grep libcamera
ii  libcamera-apps                       0~git20220131+7ac5197-1          arm64        libcamera-apps
ii  libcamera-tools                      0~git20220201+e96d0201-1         arm64        complex camera support library (tools)
ii  libcamera0:arm64                     0~git20220201+e96d0201-1         arm64        complex camera support library

castorf, 2022-03-01:
Rebuilding at 7ac5197216eaef7d8ae3d48951f047ec87589060 I could reproduce the failure, and so I think it was fixed by

commit 2cad1a201fcd8c4d79cc42dfa86990eb03ef72d4
Author: David Plowman <david.plowman@raspberrypi.com>
Date:   Mon Feb 21 11:45:40 2022 +0000

davidplowman, 2022-03-01:
Thanks for the update. We'll close this if there are no further updates in the next few days.

closed 2022-03-03 by naushir