raspberrypi/rpicam-apps #269: Preview window CPU hog / "freeze"

fire-eggs, 2022-03-03:
**Describe the bug**

The preview window gradually increases its CPU consumption over time and may go into a long CPU hog state so as to appear to "freeze".

This was originally observed when leaving the preview window running for an extremely long time. However, the symptoms can be observed fairly easily with a 20 minute timeout.

Using this command:
`libcamera-still -t 1200000 -o test.jpg`

The symptoms are as follows:
1. Using Task Manager, libcamera-still starts with CPU around 2%. Over time, this gradually increases to reach 10-11% CPU.
2. At the timeout mark, the CPU usage increases to 25% and runs at that level for 35-40 seconds.
3. The file creation time for the captured image is 50 seconds after the expected timeout time. I.e. the command was started within a second or two after the minute mark, so the captured image should have been created within a couple of seconds of the expected timeout time.

[All these symptoms occur with a smaller timeout value as well, but are greatly decreased.]

**Bug report**

Same setup etc as in issue #251.

**Additional context**

Note this occurs even with the fix to issue #251 in place.

naushir, 2022-03-03:
I suspect this may be due to the info-text update on every frame.  For some reason this does seem to degrade performance as time goes by, but I'm not entirely sure why.

Would you be able to test this again with the ``--info-text ""`` command line argument and see if it solves the issue?

naushir, 2022-03-04:
Thanks for running that test and confirming info-test does not seem to be the issue here.  I'll look into this further as well, but would you be able to try with the ``--qt_window`` preview mode and see if you get the same result?

naushir, 2022-03-04:
I wonder if this is something entirely unrelated to preview - rather it's to do with the time the camera has been running.  See https://github.com/raspberrypi/libcamera-apps/issues/217 for a description of the problem.

A fix for this has been pushed to the libcamera repo, but the version reported by your bug-report output shows that the fix is not in the library on your system.  Are you in a position to try and manually build/install the latest version of libcamera to see if it fixes your issue here?  Instructions for this can be found [here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera).

I will arrange to get our libraries in apt updated as well to pull in this fix.

fire-eggs, 2022-03-04:
#217 seems like the likely answer here.

I've not been building libcamera to-date. This is not a showstopper issue, I'll keep an eye out for updates in `apt` and re-visit then.

naushir, 2022-03-07:
The libraries in apt should now be updated.  Please do an update and let us know if this issue still occurs.  If not, feel free to close down this issue.

fire-eggs, 2022-03-07:
The issue is solved. The CPU load for libcamera-still never went above 4%, even when the timelapse timeout was reached!

closed 2022-03-07 by fire-eggs