raspberrypi/rpicam-apps #270: [BUG] libcamera apps cannot detect camera(imx219) when running in a docker container using balenaOS

rahul-thakoor, 2022-03-03:
**Describe the bug**
Any utility in `libcamera-apps` is not able to detect the camera. All exit with
```
INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
No cameras available!
```

Here is the config.txt content:

```
camera_auto_detect=1
dtoverlay=vc4-kms-v3d,cma-512
gpu_mem=256
disable_touchscreen=1
max_framebuffers=2
avoid_warnings=1
disable_splash=1
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
```

I have tried several configs, e.g

```
camera_auto_detect=0
dtoverlay=vc4-kms-v3d
max_framebuffers=2
dtoverlay=imx219
```
**Bug report**
```
--------------------------------------------------------------------------------
libcamera-apps Bug Report
--------------------------------------------------------------------------------
Date: 03-03-2022 (17:33:51)
Command: /usr/bin/camera-bug-report -t 10 -o bug.txt -c libcamera-hello --list-cameras

--------------------------------------------------------------------------------
Hardware information
--------------------------------------------------------------------------------
** hostname **
271fb53dbf73

** cat /proc/cpuinfo **
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
[... 749 lines cut ...]
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

** vcgencmd version **
Jan 20 2022 13:56:48 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

** libcamera-hello --version **
libcamera-apps build: 7ac5197216ea 01-02-2022 (14:49:21)
libcamera build: v0.0.0+3406-e96d0201

--------------------------------------------------------------------------------
libcamera-hello --list-cameras
--------------------------------------------------------------------------------
[0:03:23.281825683] [294]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
No cameras available!

```
--------------------------------------------------------------------------------------

Any help is much appreciated :pray:

rahul-thakoor, 2022-03-03:
```
v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/media1

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media2
        /dev/media3

unicam (platform:fe801000.csi):
        /dev/video0
        /dev/video1
        /dev/media0

naushir, 2022-03-04:
Firstly, does your setup work correctly in a standard Raspberry Pi OS?

I'm not familiar with balenaOS, can you provide some further details?  Do the apps work when running outside a docker environment?

rahul-thakoor, 2022-03-04:
Hi, Yes, my setup works with Raspberry Pi OS `2022-01-28`

[BalenaOS]( https://www.balena.io/os/) is a minimal linux OS with services needed to run Docker on boards like the Raspberry Pi.

I am able to use le legacy camera stack in a container environment. `libcamera-apps` on the other hand do not work.

I can try to run the app using docker on _Raspberry Pi OS_.

Are there known issues in running the app without X11? I am using `libcamera-apps-lite` and also tried with `--no-preview` option.

Thanks

naushir, 2022-03-04:
> Are there known issues in running the app without X11? I am using `libcamera-apps-lite` and also tried with `--no-preview` option.

libcamera-apps run fine without X11.  I doubt your problems are related to preview either.  The error thrown says there are now cameras available, but from the logs you posted, a camera has clearly been probed.  This is most likely due to the BalenaOS/docker environment.  Perhaps there is a permissions issue trying to access the /dev/video device nodes?  I'm afraid I am not going to be able to help much since I am not familiar with this environment.

kbingham, 2022-03-04:
The required devices will need to be passed through into the container. This include /dev/v4l-subdev* /dev/video* and /dev/media*

Those devices will have to have the correct permissions or group to be able to access those devices too.

rahul-thakoor, 2022-03-04:
thanks @naushir @kbingham 

i am running a privileged container and the hostOS has

```
cat /etc/group | grep -i video
video:x:44:
```
I am also using 

```
    group_add:
      - video
 ```
 for the service.
 
 in the container, the root user is part of the video group.
 ```
 root@4deb13f05908:/usr/src/app# cat /etc/group | grep -i video
video:x:44:root
```

The container has access to all devices in `/dev`:

```
ls -l /dev/video*
crw-rw---- 1 root video 81, 12 Mar  4 11:47 /dev/video0
crw-rw---- 1 root video 81, 13 Mar  4 11:47 /dev/video1
crw-rw---- 1 root video 81,  5 Mar  4 11:47 /dev/video10
crw-rw---- 1 root video 81,  9 Mar  4 11:47 /dev/video11
crw-rw---- 1 root video 81, 10 Mar  4 11:47 /dev/video12
crw-rw---- 1 root video 81,  0 Mar  4 11:47 /dev/video13
crw-rw---- 1 root video 81,  1 Mar  4 11:47 /dev/video14
crw-rw---- 1 root video 81,  2 Mar  4 11:47 /dev/video15
crw-rw---- 1 root video 81,  3 Mar  4 11:47 /dev/video16
crw-rw---- 1 root video 81, 11 Mar  4 11:47 /dev/video18
crw-rw---- 1 root video 81,  4 Mar  4 11:47 /dev/video20
crw-rw---- 1 root video 81,  6 Mar  4 11:47 /dev/video21
crw-rw---- 1 root video 81,  7 Mar  4 11:47 /dev/video22
crw-rw---- 1 root video 81,  8 Mar  4 11:47 /dev/video23
```
```
ls -l /dev/v4l-subdev* 
crw-rw---- 1 root video 81, 14 Mar  4 11:47 /dev/v4l-subdev0
```

Not sure what is missing at this point 🤔

kbingham, 2022-03-04:
Can you run your libcamera hello command again but prefix with
LIBCAMERA_LOG_LEVELS=*:0 
And paste the output please?

kbingham, 2022-03-04:
I am actually suspecting it will be due to udev or lack thereof in the container.

Enforcing sysfs enumeration might identify or resolve it

rahul-thakoor, 2022-03-04:
@kbingham 

```
LIBCAMERA_LOG_LEVELS=*:0 libcamera-hello -v -n
Options:
    verbose: 1
    info_text:#%frame (%fps fps) exp %exp ag %ag dg %dg
    timeout: 5000
    width: 0
    height: 0
    output: 
    post_process_file: 
    rawfull: 0
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: (libcamera)
    lores-width: 0
    lores-height: 0
    mode: unspecified
    viewfinder-mode: unspecified
Running without preview window
Opening camera...
[1:01:25.599608812] [42] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so is signed
[1:01:25.599808230] [42] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so'
[1:01:25.600143902] [42] DEBUG IPAModule ipa_module.cpp:329 ipa_vimc.so: IPA module /usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so is signed
[1:01:25.600521054] [42] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/lib/arm-linux-gnueabihf/libcamera/ipa_vimc.so'
[1:01:25.600797673] [42]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
[1:01:25.601405039] [43] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[1:01:25.612516130] [43] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
[1:01:25.612613070] [43] DEBUG RPI raspberrypi.cpp:1114 Unable to acquire a Unicam instance
[1:01:25.612661180] [43] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'SimplePipelineHandler'
[1:01:25.612743528] [43] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerUVC'
[1:01:25.612788860] [43] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerVimc'
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
ERROR: *** no cameras available ***
```

`raspberrypi.cpp:1114 Unable to acquire a Unicam instance` looks like the cause?

also thanks for the debugging tip :)

rahul-thakoor, 2022-03-04:
I also have `udev` enabled in the container with the same result. are there any specific udev rules to be added?

rahul-thakoor, 2022-03-04:
btw, I just found this https://forums.raspberrypi.com/viewtopic.php?t=325755

kbingham, 2022-03-04:
`[1:01:25.612613070] [43] DEBUG RPI raspberrypi.cpp:1114 Unable to acquire a Unicam instance`

That looks like the starting point for debug. No unicam means no camera.

naushir, 2022-03-04:
Can you see /dev/media* within the docker container?  From you last set of logs, it seems maybe not.

This is my output for ``LIBCAMERA_LOG_LEVELS=*:0 libcamera-hello --list-cameras`` showing libcamera enumerating /dev/media* which seems to be missing from your logs:
```
pi@pi4~ $ LIBCAMERA_LOG_LEVELS=*:0 libcamera-hello --list-cameras
[6:10:59.840809600] [3201] DEBUG IPAModule ipa_module.cpp:329 ipa_rpi.so: IPA module /usr/local/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so is signed
[6:10:59.841173485] [3201] DEBUG IPAManager ipa_manager.cpp:240 Loaded IPA module '/usr/local/lib/arm-linux-gnueabihf/libcamera/ipa_rpi.so'
[6:10:59.841341114] [3201]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[6:10:59.841611389] [3202] DEBUG Camera camera_manager.cpp:106 Starting camera manager
[6:10:59.841843128] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "unicam" created from /dev/media0
[6:10:59.842110958] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media0: unicam
[6:10:59.842245309] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media3
[6:10:59.842476918] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media3: bcm2835-isp
[6:10:59.842587899] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-isp" created from /dev/media1
[6:10:59.842803656] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media1: bcm2835-isp
[6:10:59.842942655] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:224 New media device "bcm2835-codec" created from /dev/media2
[6:10:59.843305281] [3202] DEBUG DeviceEnumerator device_enumerator.cpp:252 Added device /dev/media2: bcm2835-codec
[6:10:59.843344762] [3202] DEBUG Camera camera_manager.cpp:149 Found registered pipeline handler 'PipelineHandlerRPi'
```

kbingham, 2022-03-04:
Ok - that really makes me think this is an issue with UDev, ... as the DeviceEnumerator is run by udev by default. If udev is available but doesn't return anything you can end up with ... no devices.

Check in: https://git.libcamera.org/libcamera/libcamera.git/tree/src/libcamera/device_enumerator.cpp#n141

```
std::unique_ptr<DeviceEnumerator> DeviceEnumerator::create()
{
	std::unique_ptr<DeviceEnumerator> enumerator;

#ifdef HAVE_LIBUDEV
	enumerator = std::make_unique<DeviceEnumeratorUdev>();
	if (!enumerator->init())
		return enumerator;
#endif

	/*
	 * Either udev is not available or udev initialization failed. Fall back
	 * on the sysfs enumerator.
	 */
	enumerator = std::make_unique<DeviceEnumeratorSysfs>();
	if (!enumerator->init())
		return enumerator;

	return nullptr;
}
```

Can you try removing the code between HAVE_LIBUDEV and rebuilding libcamera? (or otherwise forcing the enumerator to be constructed from the DeviceEnumeratorSysfs.

If this finds it, then we might want to make it easier to fallback to sysfs enumeration at runtime... or otherwise establish how to get libudev to operate in a container.

kbingham, 2022-03-04:
https://forums.docker.com/t/docker-not-passing-udev-events-to-container-even-after-using-net-host/92394/2

Seems to indicate you might need to pass /run/udev/control through as a volume to the container.

rahul-thakoor, 2022-03-04:
Hey @kbingham i managed to get it to work. it was indeed related to udev.

i ran `udevadm control --reload` in the container

```
root@060f98f:/usr/src/app# libcamera-hello 
Preview window unavailable
[0:26:33.094118765] [84]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3406-e96d0201
[0:26:33.140931186] [85]  WARN RPI raspberrypi.cpp:1211 Mismatch between Unicam and CamHelper for embedded data usage!
[0:26:33.142138824] [85]  INFO RPI raspberrypi.cpp:1326 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media1
[0:26:33.143402331] [84]  INFO Camera camera.cpp:1028 configuring streams: (0) 1640x1232-YUV420
[0:26:33.143898868] [85]  INFO RPI raspberrypi.cpp:747 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:26:33.167839650] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Saturation = 1.000000
[0:26:33.167991644] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Contrast = 1.000000
[0:26:33.168391129] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Brightness = 0.000000
[0:26:33.168476144] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AwbMode = 0
[0:26:33.168542290] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: Sharpness = 1.000000
[0:26:33.168610046] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: ExposureValue = 0.000000
[0:26:33.168764837] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeExposureMode = 0
[0:26:33.169005124] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:26:33.169079139] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: AeMeteringMode = 0
[0:26:33.169142693] [89]  INFO IPARPI raspberrypi.cpp:626 Request ctrl: NoiseReductionMode = 3
```

rahul-thakoor, 2022-03-04:
Here is a sample app to get it to work on balenaOS: https://github.com/balena-io-playground/libcamera-apps 
Tested on Pi4 with Pi camera v1.2(imx219)

closed 2022-03-04 by rahul-thakoor

bugsyb, 2022-03-10:
Seems like am running into same lack of knowledge, though applying all what seems to be required proven to be not enough still.
Host OS arm64 Raspberry PI OS debian:Bullseye 
Container Ubuntu:20.04
```
docker run -it --rm --name devfull --hostname devfull --network=host --privileged -v /dev/:/dev/ -v /run/udev:/run/udev -v /run/dbus/:/run/dbus 
```
In the container:
```
# udevadm control --reload 
# libcamera-hello --list-cameras -n -v
[11:32:55.075806365] [384]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
[11:32:55.147222672] [385]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!
[11:32:55.149524163] [385] ERROR RPI raspberrypi.cpp:1230 Unicam driver does not use the MediaController, please update your kernel!
[11:32:55.151257234] [385] ERROR RPI raspberrypi.cpp:1129 Failed to register camera imx219 10-0010: -22
Available cameras
-----------------
0 : USB 2.0 Camera: USB Camera [1920x1080] (/base/scb/pcie@7d500000/pci@1,0/usb@1,0-1.2:1.0-0c45:6366)
    Modes: 'MJPEG' : 320x240 640x480 1280x720 1920x1080 1920x1080 
           'YUYV' : 320x240 640x480 640x480 
Closing Libcamera application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed
```

End of strace look as:
```
openat(AT_FDCWD, "/usr/lib/aarch64/libdebuginfod.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libdebuginfod.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/dev/video0", O_RDWR) = 3
**openat(AT_FDCWD, "", O_RDONLY)          = -1 ENOENT (No such file or directory)**
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so", O_RDONLY) = 3
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so.sign", O_RDONLY) = 4
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so", O_RDONLY) = 3
openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so.sign", O_RDONLY) = 4
```

Interesting is `openat ""` - could this be the case?
All volumes are mapped and all devices under /dev seem to be accessible.

Would you have any suggestions? The referred solution just runs container in privileged mode - and it loosk like something else might be wrong here?

kbingham, 2022-03-10:
the error message tells you the fault 

[11:32:55.147222672] [385]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!
[11:32:55.149524163] [385] ERROR RPI raspberrypi.cpp:1230 Unicam driver does not use the MediaController, please update your kernel!
[11:32:55.151257234] [385] ERROR RPI raspberrypi.cpp:1129 Failed to register camera imx219 10-0010: -22

You need to make sure your on the latest kernel updates from raspberry pi

bugsyb, 2022-03-13:
Thank you @kbingham. I did hold on with pi kernel upgrades as there was recently a problem with libcamera apps which prohibited the two from working hence kept version from default installation (which worked). Not sure though why docker container would be complaining about it.

kbingham, 2022-03-14:
Does it work outside the container?

meawoppl, 2022-04-22:
For me at least (RasPi 4b + IMX477), I got to this page because I didn't have the `camera_auto_detect` flag set.

To do this you either have to set the flag in the `config.txt` **BEFORE THE FIRST BOOT**. Or use the balena bashboard to set it on the device. Navigate the menus: `Device Configuration >> Custom Configuration >> Add Custom Configuration`
Name: BALENA_HOST_CONFIG_camera_auto_detect
Value: 1

Or set it fleetwide in a menu above that with a similar flow.

Pepslee, 2023-12-22:
> > Seems like am running into same lack of knowledge, though applying all what seems to be required proven to be not enough still. Host OS arm64 Raspberry PI OS debian:Bullseye Container Ubuntu:20.04
> > ```
> > docker run -it --rm --name devfull --hostname devfull --network=host --privileged -v /dev/:/dev/ -v /run/udev:/run/udev -v /run/dbus/:/run/dbus 
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > In the container:
> > ```
> > # udevadm control --reload 
> > # libcamera-hello --list-cameras -n -v
> > [11:32:55.075806365] [384]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
> > [11:32:55.147222672] [385]  WARN RPI raspberrypi.cpp:1202 Mismatch between Unicam and CamHelper for embedded data usage!
> > [11:32:55.149524163] [385] ERROR RPI raspberrypi.cpp:1230 Unicam driver does not use the MediaController, please update your kernel!
> > [11:32:55.151257234] [385] ERROR RPI raspberrypi.cpp:1129 Failed to register camera imx219 10-0010: -22
> > Available cameras
> > -----------------
> > 0 : USB 2.0 Camera: USB Camera [1920x1080] (/base/scb/pcie@7d500000/pci@1,0/usb@1,0-1.2:1.0-0c45:6366)
> >     Modes: 'MJPEG' : 320x240 640x480 1280x720 1920x1080 1920x1080 
> >            'YUYV' : 320x240 640x480 640x480 
> > Closing Libcamera application(frames displayed 0, dropped 0)
> > Camera stopped!
> > Tearing down requests, buffers and configuration
> > Camera closed
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > End of strace look as:
> > ```
> > openat(AT_FDCWD, "/usr/lib/aarch64/libdebuginfod.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
> > openat(AT_FDCWD, "/usr/lib/libdebuginfod.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
> > openat(AT_FDCWD, "/dev/video0", O_RDWR) = 3
> > **openat(AT_FDCWD, "", O_RDONLY)          = -1 ENOENT (No such file or directory)**
> > openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
> > openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so", O_RDONLY) = 3
> > openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_rpi.so.sign", O_RDONLY) = 4
> > openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so", O_RDONLY) = 3
> > openat(AT_FDCWD, "/usr/lib/aarch64-linux-gnu/libcamera/ipa_vimc.so.sign", O_RDONLY) = 4
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Interesting is `openat ""` - could this be the case? All volumes are mapped and all devices under /dev seem to be accessible.
> > Would you have any suggestions? The referred solution just runs container in privileged mode - and it loosk like something else might be wrong here?
> 
> For everyone still looking, this actually works!!! For me ubuntu 22.04 as host and image.

Hi, can you provide the example of the Dockerfile ?

nbaldy, 2024-03-07:
> Hi, can you provide the example of the Dockerfile ?

I suspect your issue might be the boot config. I don't have anything in my dockerfile special to this to this; I install v4l-utils because I'll need it later, and the libcamera dependencies (script below). 

```
sudo apt install -y libcamera-dev libepoxy-dev libjpeg-dev libtiff5-dev libpng-dev python3 python3-dev
sudo apt install -y libcamera-tools g++

###########################################################
# libcamera
sudo apt install -y python3-pip git python3-jinja2
sudo apt install -y libboost-dev
sudo apt install -y libgnutls28-dev openssl libtiff5-dev pybind11-dev liblttng-ust-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
# sudo apt install -y meson cmake
sudo apt install -y cmake # ONLY cmake! Pip installs meson
sudo apt install -y python3-yaml python3-ply python3-pip libyaml-dev
sudo pip3 install --user meson # needs sudo to be on sudo pythonpath
PATH=$PATH:/home/$USER/.local/bin/
sudo apt install -y libglib2.0-dev libgstreamer-plugins-base1.0-dev

###########################################################
# rpi-apps
sudo apt install -y cmake libboost-program-options-dev libdrm-dev libexif-dev
sudo apt install -y meson ninja-build

```

Dockerfile:
```
FROM arm64v8/ros:humble-ros-base

RUN apt-get update && apt-get install -y v4l-utils

# Runtime volume will be created here for libcam caching
RUN mkdir -p /opt/installs/src
COPY install_libcam_deps.sh   /opt/installs/src/
RUN cd /opt/installs/src && ./install_libcam_deps.sh

# Runtime volume will be created here for build caching
WORKDIR /opt/colcon_ws/
```

Note that it's important, as said above, that you detect the correct cameras in config.txt as this loads the correct device tree overlays, as I understand it. If you're running raspbian as a host, you should be able to see the cameras on the host. Then, the connections mentioned above while running should be enough, I think. 

For instance, my run is: 
```
docker run -it -v ${PWD}/colcon_ws:/opt/colcon_ws -v ${PWD}/installs/:/opt/installs/src/ --privileged --network=host -v /dev/:/dev/ --env UDEV=1 --device /dev:/dev -v /run/udev:/run/udev:ro  libcam-ros /bin/bash
```

My run command is a combination of the advice above with my specific setup - unsure if every volume/udev instruction is necessary, but it works for me.

JulienRineau, 2024-12-06:
Hi,
I do have the same issue and couldn't resolve it unfortunately - libcamera cannot detect camera (ov5647) when running in a Docker container using Ubuntu (22.04).

When trying to access it from the host (Pi5 on Raspbian bookworm) it does work - I did captured some images on it

```bash
zero@pi:~ $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : ov5647 [2592x1944 10-bit GBRG] (/base/axi/pcie@120000/rp1/i2c@88000/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [46.34 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [32.81 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]
```

But nothing is detected inside the container

```bash
root@pi:/app/ros2_ws# libcamera-hello --list-cameras
No cameras available!
```

Below is my docker-compose file, as you can see I did mounted `/run/udev` as advised by @kbingham. I also mounted `/dev` , `/run/dbus`, `/sys` and `/proc` , added the environment variable `UDEV=1` and `SYS_ADMIN` (summary of everything I found online)

```bash
services:
  service_name:
    image: image_name
    platform: linux/arm64
    container_name: container_name
    network_mode: host
    group_add:
      - video
    environment:
      - ROS_DOMAIN_ID=0
      - ROS_LOCALHOST_ONLY=1
      - SUPERVISOR_USERNAME=admin
      - SUPERVISOR_PASSWORD=admin
      - UDEV=1
    volumes:
      - ~/.ssh:/root/.ssh:ro
      - /dev:/dev
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro
      - /mnt/ssd:/mnt/ssd
      - /run/udev:/run/udev
      - /run/dbus/:/run/dbus
      - /sys:/sys:ro
      - /proc:/proc:ro
    privileged: true
    devices:
      - /dev/i2c-1:/dev/i2c-1
      - /dev/i2c-2:/dev/i2c-2
      - /dev/i2c-3:/dev/i2c-3
      - /dev/vchiq:/dev/vchiq
      - /dev/video0:/dev/video0
    device_cgroup_rules:
      - 'c 81:* rmw'
    restart: on-failure
    healthcheck:
      test: ["CMD", "supervisorctl", "status"]
      interval: 30s
      timeout: 10s
      retries: 3
    cap_add:
      - SYS_TIME
    init: true

volumes:
  shared_data:
    driver: local
```

I tried to reload udev inside the container as mentioned by @rahul-thakoor but I get this error that I can’t resolve (I did run it successfully directly in the host and created a new container, but it didn’t solve the issue)

```bash
root@pi:/app/ros2_ws# udevadm control --reload 
Running in chroot, ignoring request.
```

Also as mentioned by @nbaldy I can detect and interact with the camera in the host son the `config.txt` should be ok.

Am I missing something here ?