raspberrypi/rpicam-apps #271: [BUG]

jg3, 2022-03-09:
**Describe the bug**
libcamera-still hangs periodically.  RPI 4B running Bullseye updated yesterday.  Feels similar @EricClaeys issue #259.
```
libcamera-apps build: 0dc5ea038c79 04-03-2022 (10:35:06)
libcamera build: v0.0.0+3424-e68e0f1e
```
Camera is an Arducam OV5647 ([link](https://www.amazon.com/gp/product/B0867C9SZP)) connected via a 30cm ribbon cable ([link](https://www.amazon.com/gp/product/B089LM5D1T/)).

**Bug report**
See attached bug.txt (taken when the problem was not happening)
Camera-error.log is debug-level logging trimmed to capture three iterations: first the last successful photo, then 2 minutes later the first failure, then the subsequent attempt.

[camera-error.log](https://github.com/raspberrypi/libcamera-apps/files/8215395/camera-error.log)
[bug.txt](https://github.com/raspberrypi/libcamera-apps/files/8215396/bug.txt)

**Additional context**
Building a time-lapse camera set-up.  Running `libcamera-still -c` out of cron every 2 minutes.  Sometimes runs for an hour, sometimes for several before hanging.

This RPI is doing nothing else (although last night while waiting for it to hang I added Docker and started working out a simple way to make the latest image available via http).

If I `killall libcamera-still` the process will resume taking pictures with the next iteration.

Here's config.txt ... I hope it isn't the cause of the problem, I'm very much still working through what the right settings should be to quickly and efficiently capture a single image:
```
# this is ~pi/config.txt
# every 2 minutes cron runs: libcamera-still -c
# causing it to the configurations in this file
verbose = 1
#rawfull = 1
datetime = 1
output = /home/pi/timelapse/
latest = timelapse/latest-image.jpg
nopreview = 1
#immediate = 1
```
Here's crontab -e for pi with the default comments elided:
```
# Take a picture every two minutes and save it in ~pi/timelapse
#*/2 * * * * libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
*/2 * * * * LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
* * * * * date >> /home/pi/timelapse/camera-latest.log 2>&1
* * * * * echo "-----------------------------------------------------------------------------" >> /home/pi/timelapse/camera-latest.log 2>&1
# every 10 minutes sync that to what is in the S3 bucket mounted on /var/s3/
*/10 * * * * rsync -avW --inplace --size-only --exclude=latest-image.jpg /home/pi/timelapse/* /var/s3/timelapse/
```

EricClaeys, 2022-03-09:
@jg3 What do you get if you `strace` the libcamera-still process when it's hung?
And what do you get if you run gdb with `thread apply all bt`.
Those will indicate if your problem is the same as mine.

jg3, 2022-03-10:
Thanks, @EricClaeys .. a bit of an adventure there, haven't used either of those tools before.  Please let me know if I've done either incorrectly.   It looks like **gdb** ran out of memory before it got done, hopefully there's enough there to diagnose.  If that's not good enough next time I can kill all the later threads that came after the one which hangs.

[gdb_thread-apply-all.txt](https://github.com/raspberrypi/libcamera-apps/files/8219397/gdb_thread-apply-all.txt)
[strace_libcamera-still.txt](https://github.com/raspberrypi/libcamera-apps/files/8219398/strace_libcamera-still.txt)

naushir, 2022-03-10:
@jg3, I see some errors in the kernel dmesg logs from your run that might point to the cause.  Could you enable additional kernel logging by doing the following:

```
sudo su
echo 3 > /sys/module/bcm2835_isp/parameters/debug
echo 3 > /sys/module/bcm2835_unicam/parameters/debug
```

and re-run your tests with the ``camera-bug-report`` tool?

Is there a possibility that your runs are overlapping somehow?  If you increase your cron time to, say, every 10 minutes, does it go away?

On a related note, ``libcamera-still`` does have a time-lapse mode built in, so you could consider using that instead of a cron job to take captures, see [here.](https://www.raspberrypi.com/documentation/accessories/camera.html#still-command-line-options)  However, we would still like to get to the bottom of your problem regardless.

jg3, 2022-03-10:
Hi @naushir, I have updated the debug paramaters and re-run the bug report twice, once while the system was hung and again after killing off the processes so it completes:

While hung:
[bug3-hung-report.txt](https://github.com/raspberrypi/libcamera-apps/files/8223686/bug3-hung-report.txt)

Again, after 
[bug3.txt](https://github.com/raspberrypi/libcamera-apps/files/8223684/bug3.txt)

It is absolutely the case that the runs are overlapping.  Here's what I see in camera-error.log pasted above: at some point when cron calls libcamera-still it seems (to my scanning of the debug log) to be running normally when ...
```
Camera started!
[8:49:09.077999882] [10729] DEBUG RPI raspberrypi.cpp:1060 queueRequestDevice: New request.
[8:49:09.079128167] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 0 for Unicam Image
[8:49:09.079280573] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 0
[8:49:09.079366128] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 0
[8:49:09.110459502] [10729] DEBUG RPI raspberrypi.cpp:1772 Stream Unicam Image buffer dequeue, buffer id 1, timestamp: 31749051664000
[8:49:09.110574445] [10729] DEBUG RPI raspberrypi.cpp:2064 Signalling signalIspPrepare: Bayer buffer id: 1
[8:49:09.110951275] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 1
[8:49:09.112254225] [10729] DEBUG RPI raspberrypi.cpp:1689 Input re-queue to ISP, buffer id 1, timestamp: 31749051664000
[8:49:09.112311891] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer -1 for ISP Input
[8:49:09.169814241] [10729] DEBUG RPI raspberrypi.cpp:1772 Stream Unicam Image buffer dequeue, buffer id 0, timestamp: 31749110893000
[8:49:09.170182034] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 2
[8:49:09.229437737] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 3
[8:49:09.245711963] [10729] DEBUG RPI raspberrypi.cpp:1800 Stream ISP Input buffer complete, buffer id 1, timestamp: 31749051664000
[8:49:09.245816129] [10729] DEBUG RPI raspberrypi.cpp:1828 Stream ISP Output0 buffer complete, buffer id 1, timestamp: 31749051664000
[8:49:09.245872351] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 1 for ISP Output0
[8:49:09.245981183] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 0 for ISP Output0
[8:49:09.246336494] [10729] DEBUG RPI raspberrypi.cpp:1828 Stream ISP Output1 buffer complete, buffer id 0, timestamp: 31749051664000
[8:49:09.246396549] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 0 for ISP Output1
[8:49:09.246529863] [10729] DEBUG RPI raspberrypi.cpp:1828 Stream ISP Stats buffer complete, buffer id 0, timestamp: 31749051664000
[8:49:09.246753490] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 0 for ISP Stats
[8:49:09.246881507] [10729] DEBUG RPI raspberrypi.cpp:1971 Dropping frame at the request of the IPA (1 left)
[8:49:09.246964395] [10729] DEBUG RPI raspberrypi.cpp:2064 Signalling signalIspPrepare: Bayer buffer id: 0
[8:49:09.248535935] [10729] DEBUG RPI raspberrypi.cpp:1689 Input re-queue to ISP, buffer id 0, timestamp: 31749110893000
[8:49:09.248618064] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer -1 for ISP Input
[8:49:09.288670348] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 4
[8:49:09.347891904] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 5
[8:49:09.382246264] [10729] DEBUG RPI raspberrypi.cpp:1800 Stream ISP Input buffer complete, buffer id 0, timestamp: 31749110893000
[8:49:09.382349078] [10729] DEBUG RPI raspberrypi.cpp:1828 Stream ISP Output0 buffer complete, buffer id 1, timestamp: 31749110893000
[8:49:09.382437817] [10729] DEBUG RPI raspberrypi.cpp:1828 Stream ISP Output1 buffer complete, buffer id 0, timestamp: 31749110893000
[8:49:09.382491909] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 0 for ISP Output1
[8:49:09.382629149] [10729] DEBUG RPI raspberrypi.cpp:1828 Stream ISP Stats buffer complete, buffer id 0, timestamp: 31749110893000
[8:49:09.382844091] [10729] DEBUG RPISTREAM rpi_stream.cpp:241 Queuing buffer 0 for ISP Stats
[8:49:09.382968423] [10729] DEBUG RPI raspberrypi.cpp:1971 Dropping frame at the request of the IPA (0 left)
[8:49:09.407113904] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 6
[8:49:09.466338700] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 7
[8:49:09.525604107] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 8
[8:49:09.584795681] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 9
[8:49:09.644091606] [10729] DEBUG RPI raspberrypi.cpp:1461 frame start 10
```
... and then it continues incrementing frame start indefinitely.   In 2 minutes when cron launches libcamera-still again, it hangs up and this continues.   Also, I don't remember seeing two "frame start 0" lines in cases except where the hang occurs.

naushir, 2022-03-11:
Unfortunately I do not see anything obvious from the last set of logs.  Did you have any results from when you used a 10 minute delay?

Just to clarify, when you say the runs overlap, do you mean that one instance of ``libcamera-still`` is still running when a second instance is started?  In these cases, the second instance should immediately fail, and the first instance remains un-interrupted.  I've just tried this on my setup, and this is indeed the case, so it probably suggests this is not the problem you are seeing.

jg3, 2022-03-12:
As far as I can tell changing the cron timer from 2 to 10 minutes didn't change anything.   When I checked a few hours later the pictures were not generating and there were a ton of libcamera-still processes:
```
[13:36:11][pi@timelapsepi:~/latest-image]$ ps -ef | grep camera
pi          2369    2367  0 10:00 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2372    2369  0 10:00 ?        00:00:47 libcamera-still --config /home/pi/config.txt
pi          2447    2444  0 10:10 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2449    2447  0 10:10 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          2532    2529  0 10:20 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2534    2532  0 10:20 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          2623    2621  0 10:30 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2626    2623  0 10:30 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          2695    2692  0 10:40 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2698    2695  0 10:40 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          2768    2765  0 10:50 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2770    2768  0 10:50 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          2861    2858  0 11:00 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2864    2861  0 11:00 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          2933    2930  0 11:10 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          2935    2933  0 11:10 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          3014    3010  0 11:20 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          3015    3014  0 11:20 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          3101    3099  0 11:30 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          3104    3101  0 11:30 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          3175    3172  0 11:40 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          3177    3175  0 11:40 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          3248    3246  0 11:50 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          3251    3248  0 11:50 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          3340    3337  0 12:00 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
pi          3343    3340  0 12:00 ?        00:00:00 libcamera-still --config /home/pi/config.txt
pi          3413    3410  0 12:10 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0  libcamera-still --config /home/pi/config.txt >> /home/pi/timelapse/camera-latest.log 2>&1
...
```
When I looked a little while ago `ps -ef | grep -v grep | grep -v /bin/sh |  grep libcamera-still | wc -l` gave me **219**!

Does that help clarify what I mean by overlapping?

naushir, 2022-03-12:
Yes, I see the overlap now.  This is mostly speculative, but could you update to the latest 5.15 kernel and give it a try?  There has been a fix for buffer handling in the ISP driver put in - recently https://github.com/raspberrypi/linux/pull/4921.  I am not confident it will help as libcamera-apps do not provoke the error from what I can tell.

Could you also add V4L2 logging to your command line and provide an output:
```
LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still ...
```

EricClaeys, 2022-03-12:
The Allsky program kills any existing libcamera-still processes before it starts a new one, so we avoid having multiple instances running at once.

jg3, 2022-03-12:
I am happy to try the 5.15 kernel .. is that achieveable through [rpi-update](https://www.raspberrypi.com/documentation/computers/os.html#using-rpi-update)?  

For whatever reason it has not hung today since a reboot at midnight, running every 10 minutes.

I have added the V4L2 logging.

jg3, 2022-03-13:
I found that one of the runs of `libcamera-still` hung overnight at teh 3:50 AM run.  So, I killed off all the later processes and ran strace:
`strace --output="strace_libcamera-still_25364.txt" --attach=25364`

the output:
futex(0x7ff7630b8c, FUTEX_WAIT_PRIVATE, 0, NULL <detached ...>

Here is a log file of three runs/attempts as before.. the last run that was apparently successful, the run that hung, and the following run that attempts, with increased verbosity V4L2:
[trim-camera.log](https://github.com/raspberrypi/libcamera-apps/files/8239914/trim-camera.log)

naushir, 2022-03-14:
@jg3, thanks for the update.  Unfortunately, I do not see anything obvious that is going wrong in your logs.  I have started a run on my setup here to see if I can reproduce the lockup.

Out of interest, do you have a v2 camera that you can use instead of the ov5647?  It would be interesting to know if you get the same lockup.

jg3, 2022-03-14:
I have a "Raspberry Pi camera Rev 1.3" that I forgot about until just now.  I tested it once but never really used it.  I'll give that a try tonight.

naushir, 2022-03-16:
I've been running my test for over 48-hours now, and have not managed to get it to hang.  @jg3, have you seen a crash with the other camera yet?

jg3, 2022-03-16:
I have not seen it hang with the other camera yet. I'm hopeful this isn't just a bad hardware problem, or that it would have been more evident if it were.   I have another unopened Arducam ov5647 ... I just have to decide if I hook it up for a test or send it back and get a "real" v2 camera. 🤔

Sharsie, 2022-03-16:
I might have encountered a similar problem, not sure if related, but I thought I would mention it, just in case. Note that I didn't try with the ISP driver update in kernel 5.15, but I'll try that once I get to the camera again.

What happened to me was that I ran libcamera-vid through udp and at the same time I tried to capture an image with libcamera-still. The still hang even after killing the libcamera-vid and the process never finished, so I guess it was never notified that the hardware was free to be interacted with. The camera is Waveshare RPi IR-CUT Camera (B) and it was 100% reproducible.
I think the same happens with any concurrent libcamera commands, but would have to retest.

EricClaeys, 2022-03-16:
In the Allsky application, we do a `pgrep libcamera-still` and kill any outstanding process before running a new one.  That eliminates the possibility of running concurrent camera commands.

naushir, 2022-03-16:
@EricClaeys the concurrent camera useage is easy to spot in the logs, so I don't think that would be your problem either.

Sharsie, 2022-03-16:
@EricClaeys understood, I handled this pretty much in the same way, I was just surprised that the command did not fail straight away and became stuck forever, that's why it seemed to me it could be connected, but probably is not. Sorry to hijack.

EricClaeys, 2022-03-17:
@Sharsie Understand, and agree - I also would have expected the second instance to have provided a meaningful message, then exited with an error code, rather than getting stuck.

naushir, 2022-03-17:
> I have not seen it hang with the other camera yet. I'm hopeful this isn't just a bad hardware problem, or that it would have been more evident if it were.

One possible way I could see this happening is if the sensor stops producing Frame Start/End events, the pipeline handler will simply stall waiting for something to do, with no indication that the sensor is in a bad state.  I'll look to add some instrumentation into the libcamera library to test for this and throw an error.  Recovery from this is essentially do a restart as you have been doing.

naushir, 2022-03-18:
@jg3 did you have any updates with your tests using the other sensor module?

jg3, 2022-03-18:
Hi @naushir -- it took a few _days_ to get it to hang but it eventually did.   I don't know how important this is but it's an ov5647 also, it seems?

I should also note that today was a little strange environmentally.  It wasn't man-handled, but the room it was positioned in got painted and it may have been moved.  Nice to get fresh paint but I was bummed to interrupt my timelapse of a blooming orchid.  Anyway, it's a suspicious correlation given that this has been taking one picture a minute for the past 3 days and only has this one hang to show for it.

```
[14:16:03.031723740] [15291]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
[14:16:03.054201161] [15292] ERROR CameraSensor camera_sensor.cpp:551 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[14:16:03.080728849] [15292]  INFO RPI raspberrypi.cpp:1317 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP
 device /dev/media1
```

naushir, 2022-03-19:
> [14:16:03.031723740] [15291]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3424-e68e0f1e
> [14:16:03.054201161] [15292] ERROR CameraSensor camera_sensor.cpp:551 'ov5647 10-0036': Camera sensor [...]

This log message is not really an error - it shows up on every run  so can be safely ignored.  Do you still happen to have the full log output of when it crashed?

jg3, 2022-03-23:
Hi @naushir ... I have re-enabled the extra logging `LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0` while I wait for it to occur again.

naushir, 2022-03-23:
@jg3 That would be useful to debug this issue.  If you are able to, re-building the libcamera libraries with the patches in https://patchwork.libcamera.org/project/libcamera/list/?series=2983 would also be quite helpful.

Sharsie, 2022-03-23:
I can confirm I started experiencing the same issue with two ov5647 on two rpi zeros (unrelated to what I previously posted). Bit difficult to debug as the cameras don't have internet connection and it's bit of a drive to get to them, but if you could summarize the directions to get the patches in and build the thing, I can take a shot at it probably during the weekend. So far they always froze in about 3-5 hours and it happened twice (I was there twice:)

naushir, 2022-03-24:
The patch series I mentioned above has been superseded with https://patchwork.libcamera.org/project/libcamera/list/?series=2992.

If you want to try it out, make sure you are setup to build libcamera libraries (instructions [here](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps).  To apply the patches to your tree, go to the above link, click on one of the 2 patches, and select "series" on the top right of the page to download the patch diff.  To apply the diff use ``git am <patch_file_downloaded>`` in your libcamera tree and rebuild.

Sharsie, 2022-03-27:
@naushir I managed to build and install libcamera and libcamera-apps, I wasn't sure only about the comment in

```
ninja -C build   # use -j 2 on Raspberry Pi 3 or earlier devices
```

Should this be used on rpi zero as well? I built it with `-j 2` flag as I assume it falls into the category, but just to make sure.

As per suggestions in the document, I also rebuilt libcamera-apps from scratch using
```
cmake .. -DENABLE_COMPILE_FLAGS_FOR_TARGET=armv8-neon -DENABLE_DRM=1 -DENABLE_X11=0 -DENABLE_QT=0 -DENABLE_OPENCV=0 -DENABLE_TFLITE=0
```
and again
```
make -j4  # use -j1 on Raspberry Pi 3 or earlier devices
```
I tried both -j1 and -j4, but after `sudo make install` i get `Illegal instruction` from any libcamera-x command

Googling further, it brought me to https://github.com/raspberrypi/documentation/issues/1902 which suggests `DENABLE_COMPILE_FLAGS_FOR_TARGET=none` which I did and that seems to have worked, would be worth documenting I guess

So now I'm going to run my service using `LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0` and logging stdout and stderr. If there is anything else that could help pinpoint the issue, let me know please.

Sharsie, 2022-03-28:
Here is a gist https://gist.github.com/Sharsie/7d0245da8d6e8f373dd9dbf02d78c6ed
Found it running for an hour, so anything after line 719 is probably garbage as a result of the hang

naushir, 2022-03-28:
>```
> [3:41:16.477707790] [3029]  WARN V4L2 v4l2_videodevice.cpp:1895 /dev/video0[11:cap]: Dequeue timer of 2130 ms has expired!
>[3:41:16.478143787] [3029] ERROR RPI raspberrypi.cpp:1769 Unicam has timed out!
>Please check that your camera sensor connector is attached securely.
>Alternatively, try another cable and/or sensor.
>```

This does suggest that perhaps there is a HW fault with the camera.  However, note that this code is not fully exercised, so might point to a different unknown problem.  @Sharsie can you provide the exact command line you used to during this run?  If you are able to, please try with a different cable and/or sensor to see if you still get the problem.

Sharsie, 2022-03-28:
Okay, I will test IMX219, but from the previous tests I never noticed a hang, I'll let it run for a whilte and see if it happens.
I can say that both cameras with OV5647 I had (both are different Waveshare models and both on different rpi Zeros) had this exact same problem, so it's most likely not a problem with the specific HW, but its implementation (which might be on Waveshare's side though, not blaming libcamera:)

The exact command I am running is 
```
LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still -q 88 --nopreview --immediate -o "..." --width 2592 --height 1944
```

Sharsie, 2022-03-29:
Considering I always had a hang in several hours and it's been running with IMX219 for about 16 hours now without any issues, I'd guess it's not going to stall at this point. However I'll keep it running for another day or two.

closed 2022-03-30 by naushir

jg3, 2022-03-31:
Hi @naushir -- thanks for the attention you've put on this.  I took a few days off so I haven't been around.    I put the patches you described in place last night.  I have swapped the camera to a new arducam OV5647 identical to the one I started with and I saw the same issue again overnight.

```
[08:27:42][pi@timelapsepi:~/buttons]$ ps -ef | grep -v grep | grep camera | wc -l
838
[08:28:08][pi@timelapsepi:~/buttons]$ crontab -e
crontab: installing new crontab
[08:28:35][pi@timelapsepi:~/buttons]$ ps -ef | grep -v grep | grep camera | head
pi         36929   36926  0 01:30 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still --config /home/pi/timelapse/config.txt >> /var/log/timelapse-camera/timelapse-camera.log 2>&1
pi         36932   36929  0 01:30 ?        00:01:27 libcamera-still --config /home/pi/timelapse/config.txt
pi         37184   37176  0 01:30 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still --config /home/pi/timelapse/config.txt >> /var/log/timelapse-camera/timelapse-camera.log 2>&1
pi         37191   37184  0 01:30 ?        00:00:00 libcamera-still --config /home/pi/timelapse/config.txt
pi         37382   37380  0 01:32 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still --config /home/pi/timelapse/config.txt >> /var/log/timelapse-camera/timelapse-camera.log 2>&1
pi         37384   37382  0 01:32 ?        00:00:00 libcamera-still --config /home/pi/timelapse/config.txt
pi         37619   37616  0 01:33 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still --config /home/pi/timelapse/config.txt >> /var/log/timelapse-camera/timelapse-camera.log 2>&1
pi         37620   37619  0 01:33 ?        00:00:00 libcamera-still --config /home/pi/timelapse/config.txt
pi         37855   37854  0 01:34 ?        00:00:00 /bin/sh -c LIBCAMERA_LOG_LEVELS=RPI:0,V4L2:0 libcamera-still --config /home/pi/timelapse/config.txt >> /var/log/timelapse-camera/timelapse-camera.log 2>&1
pi         37857   37855  0 01:34 ?        00:00:00 libcamera-still --config /home/pi/timelapse/config.txt
```

So, I have trimmed the log to start at 1:28 and end at 1:31 while cron triggered that command every minute. (is it strange that the 1:31 run isn't in this process list?)
[timelapse-camera_jg20220331_trim.log](https://github.com/raspberrypi/libcamera-apps/files/8392100/timelapse-camera_jg20220331_trim.log)

Hopefully these logs with the new code and lots of debug info are helpful.