raspberrypi/rpicam-apps #272: A 5 seconds image took 30 seconds to get it done?

universe241981, 2022-03-12:
Would like to use the libcamera-still to take long exposure image and save in raw (dng). But the application took about 30 seconds to get the 5 seconds exposure image done. A 32 seconds exposure image took more than 2 minutes to finish it.
Is this a nature of the libcamera-still?

naushir, 2022-03-13:
You need to use the ``--immediate`` option, otherwise the 3A algorithms will need a number of frames to converge before a capture occurs.  See [here](https://www.raspberrypi.com/documentation/accessories/camera.html#very-long-exposures) for more details.

universe241981, 2022-03-14:
Ah yes. My bad. I didn't specify the awbgains value. That's took a lot of time for the camera to finish it job. Thanks for the tips.