raspberrypi/rpicam-apps #273: Kernel crash marked as gstreamer, but happens only if v4l2h264enc is used.

bugsyb, 2022-03-13:
This is intermittent issue for which test conditions were repeated number of times and sometimes it happens every reboot/power cycle, sometimes it doesn't happen at all. No other parallel apps or anything else is running on the system.

When it happens, crash happens after 2-5 runs with USB camera connected since reboot (much more often than with no usb cam).

```
PARAM=" --width=1920 --heigh=1080 --codec yuv420 --framerate 15 --gain 500"
while true; do timeout 15 libcamera-vid -t 0 $PARAM -o -     | gst-launch-1.0 -vvvv fdsrc fd=0     ! rawvideoparse use-sink-caps=false width=1920 height=1080 framerate=15/1 format=i420     ! v4l2convert ! v4l2h264enc extra-controls="controls,video_bitrate_mode=0,video_bitrate=1000000,repeat_sequence_header=1"     ! 'video/x-h264,profile=high,level=(string)4.2' ! h264parse ! rtph264pay     ! udpsink host=desktop port=8004; done
```

Doesn't cause an issue:
```
while true; do timeout 15 libcamera-vid -t 0 $PARAM -o -     | gst-launch-1.0 -vvvv fdsrc fd=0     ! rawvideoparse use-sink-caps=false width=1920 height=1080 framerate=15/1 format=i420     ! fakesink ; sleep 1;done
```

If USB cam is used in between tries (not in parallel) of picam via libcamera, then issue sometimes happen already on cycle of: reboot, libcamera for 15 sec, usbcam, libcamer starts with crash instantly,
``` 
gst-launch-1.0 v4l2src device=/dev/video0 ! 'image/jpeg,framerate=30/1,width=1920,height=1080'! v4l2jpegdec ! videoconvert ! v4l2h264enc extra-controls="controls,video_bitrate=2500000,video_bitrate_mode=0" !  'video/x-h264,level=(string)4' ! h264parse ! rtph264pay config-interval=1  ! udpsink host=dp55l port=8004
```

```
v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/media1

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/media0

unicam (platform:fe801000.csi):
	/dev/video4
	/dev/video5
	/dev/media2

USB 2.0 Camera: USB Camera (usb-0000:01:00.0-1.4):
	/dev/video0
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/media3
```

Environment:
```
Linux piwatcher 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux

libcamera-vid --version
libcamera-apps build: 2a38ae93f143 21-10-2021 (14:53:32)
libcamera build: v0.0.0+3156-f4070274

gst-launch-1.0 version 1.18.4
GStreamer 1.18.4
http://packages.qa.debian.org/gstreamer1.0
```

Kernel/libcamera hasn't been updated yet, due to:
https://forums.raspberrypi.com/viewtopic.php?p=1962367#p1962742
Which is probably fixed now, but preferred to stick with so far working version until today discovery.

Kernel crash (below one did happen when USB cam was disconnected and connected back to USB2.0 port, but same crash happens also without cam being connected/disconnected. Crash happens when running libcamera with gst. 
If libcamera is being run again after crash, system hangs. If gst is used, it works fine.
```
Mar 13 23:47:08 pihost kernel: usb 1-1.4: new high-speed USB device number 3 using xhci_hcd
Mar 13 23:47:08 pihost kernel: usb 1-1.4: New USB device found, idVendor=0c45, idProduct=6366, bcdDevice= 1.00
[... 128 lines cut ...]
Mar 14 00:57:31 pihost kernel:  update_process_times+0x68/0xe0
Mar 14 00:57:31 pihost kernel:  tick_sched_handle.isra.19+0x48/0x60
Mar 14 00:57:31 pihost kernel:  tick_sched_timer+0x60/0xb8
Mar 14 00:57:31 pihost kernel:  __hrtimer_run_queues+0x11c/0x470
Mar 14 00:57:31 pihost kernel:  hrtimer_interrupt+0xfc/0x258
Mar 14 00:57:31 pihost kernel:  arch_timer_handler_phys+0x38/0x48
Mar 14 00:57:31 pihost kernel:  handle_percpu_devid_irq+0xa8/0x2a0
Mar 14 00:57:31 pihost kernel:  generic_handle_irq+0x38/0x50
Mar 14 00:57:31 pihost kernel:  __handle_domain_irq+0x9c/0x110
Mar 14 00:57:31 pihost kernel:  gic_handle_irq+0xb0/0xf0
Mar 14 00:57:31 pihost kernel:  el1_irq+0xcc/0x180
Mar 14 00:57:31 pihost kernel:  queued_spin_lock_slowpath+0x1f0/0x400
Mar 14 00:57:31 pihost kernel:  _raw_spin_lock+0x74/0x78
Mar 14 00:57:31 pihost kernel:  __mutex_lock.isra.13+0x88/0x540
Mar 14 00:57:31 pihost kernel:  __mutex_lock_slowpath+0x1c/0x28
Mar 14 00:57:31 pihost kernel:  mutex_lock+0x40/0x58
Mar 14 00:57:31 pihost kernel:  _vb2_fop_release+0x38/0xa8 [videobuf2_v4l2]
Mar 14 00:57:31 pihost kernel:  vb2_fop_release+0x34/0x60 [videobuf2_v4l2]
Mar 14 00:57:31 pihost kernel:  v4l2_release+0xc8/0x108 [videodev]
Mar 14 00:57:31 pihost kernel:  __fput+0x80/0x238
Mar 14 00:57:31 pihost kernel:  ____fput+0x18/0x28
Mar 14 00:57:31 pihost kernel:  task_work_run+0x90/0x150
Mar 14 00:57:31 pihost kernel:  do_notify_resume+0x22c/0x6d8
Mar 14 00:57:31 pihost kernel:  work_pending+0xc/0x43c
```

naushir, 2022-03-14:
> When it happens, crash happens after 2-5 runs with USB camera connected since reboot (much more often than with no usb cam).

Can i confirm that you are using a USB webcam with libcamera-apps and not one of the Raspberry Pi cameras? libcamera-apps is not intended to be used with anything but a CSI2 camera setup.

naushir, 2022-03-18:
I'll close this issue down now as it seems related to USB webcams that we do not support.  If this is not correct, please feel free to reopen the issue.

bugsyb, 2022-03-19:
@naushir, please kindly re-open - explanation below - I've been slow in responding as been traveling for day job.

Issue is with libcamera-vid/maybe kernel and exposed when using libcamera couple of times, where in between USB camera is used.
It smells like some pointer issue?

```
Mar 13 23:47:16 pihost kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
```

The crash does not happen when using USB cam (directly with gst), but only when using libcamera which effectively uses picam and pipes to gst.

naushir, 2022-03-21:
@bugsyb both your libcamera and kernel versions seem quite a bit out of date.  Are you able to use a new sdcard with a flash of the latest image, and ensure all your libraries and kernel are up-to-date.  There have been may fixes that have gone in over the last few months, and it is possible that the issue you might be encountering has already been fixed.

naushir, 2022-03-30:
@bugsyb did you manage to update your libraries and try this again?

bugsyb, 2022-03-30:
@naushir, apologies not yet - had other activities which slowed the hobby side a bit.

naushir, 2022-04-20:
I'll close this issue down now as I think the latest libraries do not show this bug.  If this is wrong, please do re-open.

closed 2022-04-20 by naushir